realsenseai/librealsense #293: R200 on Ubuntu 16.04 showing only RGB, no depth or IR

ScottJohnson2718, 2016-09-22:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06  | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  |  4.4.0-38-generic  |
| Build System |  Makefile | 

I am trying to get my new R200 camera to work on my Dell XPS L701X laptop with Ubuntu 16.04 and eventually ROS.
When I run the bin\cpp-capture utility, there is only an RGB image.  The IR images and the depth image are blank.  Usually there is a console warning "rs.warn : Subdevice 0 bad magic number 0x0" but in issue 49 it was stated this isn't enough to warrant an error.  

I am a software developer so I checked that the patch in the scipts/patch-uvcvideo-16.04.simple.sh step worked correctly.  I went through each change to the kernel files and verified that they were there.  I also verified that the new kernel was built and copied to its destination location.

I tried my camera last night on Windows 10 using the Intel Realsense SDK.  It worked fine with the same cable and using the same USB3 port on my laptop.

I found the Hardware Compatibility section in the Readme. 
https://github.com/IntelRealSense/librealsense#hardware-requirements
My USB3 controller is NEC uPD720200 USB 3.0 Host Controller and
PCI Bridges are all Intel 5 Series/3400 Series Chipset PCI Express

I can't tell if I meet the hardware requirements or not or if not, if that is fatal.  

Any help is appreciated.  If not the hardware requirements then what else can I check?

teknotus, 2016-09-22:
What does dmesg show when you plug in the camera?

ScottJohnson2718, 2016-09-22:
[ 8640.790837] wlp4s0: Limiting TX power to 14 dBm as advertised by 1c:aa:07:43:a5:af
[ 8656.123101] usb 4-1: USB disconnect, device number 3
[ 8661.871487] usb 4-1: new SuperSpeed USB device number 4 using xhci_hcd
[ 8661.893626] usb 4-1: New USB device found, idVendor=8086, idProduct=0a80
[ 8661.893634] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8661.893639] usb 4-1: Product: Intel RealSense 3D Camera R200
[ 8661.893644] usb 4-1: Manufacturer: Intel Corp
[ 8661.893648] usb 4-1: SerialNumber: SN_2441007647
[ 8661.896152] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 8661.900642] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 8661.903307] uvcvideo: Unable to create debugfs 4-4 directory.
[ 8661.903689] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:1c.3/0000:06:00.0/usb4/4-1/4-1:1.2/input/input24
[ 8661.905249] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 8661.908243] uvcvideo: Unable to create debugfs 4-4 directory.
[ 8661.908608] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:1c.3/0000:06:00.0/usb4/4-1/4-1:1.4/input/input25

teknotus, 2016-09-22:
Usually there is a complaint in the log about unknown formats if the patch
didn't apply.

Can you get the list of formats for each /dev/videoX ?

*v4l2*-*ctl* -- *list*-*formats*-*ext *-d /dev/video...

ScottJohnson2718, 2016-09-23:
scott@scott-XPS-L701X:~/librealsense/bin$ v4l2-ctl --list-formats-ext -d /dev/video0
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 640x400
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 384x216
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 176x144
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 160x128
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 768x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.040s (25.000 fps)
			Interval: Discrete 0.050s (20.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
			Interval: Discrete 0.200s (5.000 fps)
		Size: Discrete 1024x768
[... 203 lines cut ...]
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'RW10'
	Name        : Raw data 10-bit (RW10)
		Size: Discrete 2400x1081
			Interval: Discrete 0.033s (30.000 fps)

ScottJohnson2718, 2016-09-23:
Theory : the R200 is working fine but the bin/cpp-catpture program is not rendering the depth and IR buffers because they are pixel formats other than the very standard RGB8.  
So I installed the NVidia drivers and tried again.  After some headaches I am up and running again but I get the same result.  Still, there is no step in the cpp-capture program that swizzles the pixels manually so it must be up to underlying opengl library to do it.  
This is the avenue I'm currently pursuing.  I tried using vlc and mplayer to render from the /dev/videoX devices.  There are way too many parameters to play with and they crash the entire machine when they are unhappy.  Still trying.

Here is what the capture application thinks the pixel formats are:

scott@scott-XPS-L701X:~/librealsense/bin$ ./cpp-capture
Capturing DEPTH at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
format : Z16
Capturing COLOR at 640 x 480, fov = 54.4 x 41.8, distortion = MODIFIED_BROWN_CONRADY
format : RGB8
Capturing INFRARED at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
format : Y8
Capturing INFRARED2 at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
format : Y8

ev-mp, 2016-09-25:
According to all logs the kernel patch is in place and all pixel formats are correctly recognized.
The only thing that is different for your setup is the laptop embedded camera under dev/video0. 
In most cases a blank/white screen for Depth/Infrared streams is a sign for a unrecognized pixel format, but it seem not to be the case here.
Can you please try running non-gui demos, such as cpp-tutorial-1-depth and cpp-headless and share the results ?
The first demo should display Depth using ASCII characters in console, while the second stores snapshots for the streams in png format on hard disk.
These demos do not use openGl so it can help to rule out NVidia driver issues.

ScottJohnson2718, 2016-09-25:
The cpp-tutorial-1-depth does print characters to the screen.  So the depth is able to stream.  So I changed the cpp-capture to render the actual depth image instead of the histogram of the depth buffer.  I also cut/pasted the depth stream setup line from the cpp-tutorial-1-depth application.  If I only enable the depth stream (and no color or IR) then I get a reasonable image at 640x480 @30fps using the depth as luminance.  That's a good sign.
If I reenable the color stream at 640x480 @30fps then the depth image goes away.  This is confirmed in the cpp-config-ui tutorial.  I tried to see just an IR image but that didn't work.

You asked for the result of cpp-headless and its output matches what I see in the capture when I have all three streams enabled.  That is the color image is good, and the depth and IR are black.

So the problem has become why can't I get a depth and a color stream at the same time and why IR is always black.  Also, it wasn't clear to me what the cpp-capture application was supposed to be showing me until I read the code.  Still a long way off from having a fully functioning camera.

Any idea why I can't get color and depth at the same time or IR at all?  I keep looking at dmesg for errors and I don't see any.

ev-mp, 2016-09-25:
It is worth verifying the power supply that the laptop provides via USB port:
- Connect the laptop to AC power supply.
- Disconnect the USB port next to the one used by R200 camera, so the USB controller will have only one consumer.
- Connect and run the camera via self-powered USB3 hub.

ScottJohnson2718, 2016-09-26:
I found a USB3 hub (https://www.amazon.com/AmazonBasics-Port-2-5A-power-adapter/dp/B00DQFGH80)  and tried it.  There was no change.  It was a good idea though.  I disconnected all other USB devices including the keyboard and mouse.  I can do that on a laptop.

I started looking at the Ubuntu site for debugging web cams.  It mentioned a tool called guvcview so I installed it.  Oddly enough, I can run 
$ guvcview -d /dev/video2
and I get a bizarre 312x938 green badly interlaced image that is roughly a depth map.  This corresponds to the first reported pixel format for the device.
In another terminal while the above is still running
$ guvcview -d /dev/video3
and I get a nice clear 640x480 color image at 30fps.  

So I *can* get both channels at the same time with the same application running in two processes.  
Just not with the example applications in the same process.  I'm way over my head here and it works enough to bait me into thinking I'm close.

ScottJohnson2718, 2016-09-27:
I started over on the Dell XPS-L701 laptop with Ubuntu 14.04.  The results were the same.  I grabbed an Alienware laptop and installed Ubuntu 14.04 on it.  Bingo!  The sample applications work nicely.  It is such a relief to have a working configuration.  Thanks to everyone that replied.

alberto2000, 2017-01-04:
Any idea what actually caused the issue / what the problem was? Can I expect the camera to work with current Ubuntu on any other computer?

tdnet12434, 2017-04-19:
I face exactly the same problem. So i try to install Upboard kernel from [here](https://01.org/developerjourney/installing-ubutnu-1404-lts-intel-realsense-robotic-development-kit). It fixed this problem. So i think maybe this is kernel problem?

zhangduoshou, 2017-09-22:
maybe the IR  emitter u default set is 0

jlgarcia75, 2017-10-03:
Hello,

If the issue is software related, we are no longer updating this software as we have moved to our latest generation, the D400 series. If you are in your warranty, a replacement can be provided.

Jesus G.
Intel Customer Support

closed 2017-10-03 by jlgarcia75