realsenseai/librealsense #299: Can't stream IR in multicam example?

bloyl, 2016-09-27:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06 | 
| Operating System & Version |  ubuntu 16.04.1 |
| Kernel Version (Linux Only)  |  4.4.0-38-generic #57-Ubuntu SMP Tue Sep 6 15:42:33 UTC 2016 |
| Build System |  Makefile | 

I've been encountering problems enabling the infrared stream on a r200 in a multicamera setup. I can replicate the issue by enabling the IR stream in the cpp-multicam.cpp example and using only a single camera.

I think the kernel mod is all setup.

uname -a

> Linux megHPI-acq 4.4.0-38-generic #57-Ubuntu SMP Tue Sep 6 15:42:33 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

When I plug in the camera dmesg returns

> [  108.524755] usb 6-1: new SuperSpeed USB device number 2 using xhci_hcd
> [  108.545499] usb 6-1: New USB device found, idVendor=8086, idProduct=0a80
> [  108.545505] usb 6-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [  108.545508] usb 6-1: Product: Intel RealSense 3D Camera R200
> [  108.545510] usb 6-1: Manufacturer: Intel Corp
> [  108.545512] usb 6-1: SerialNumber: SN_2361002473
> [  108.549171] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
> [  108.553659] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
> [  108.556746] uvcvideo: Unable to create debugfs 6-2 directory.
> [  108.556912] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:03.0/0000:04:00.0/usb6/6-1/6-1:1.2/input/input16
> [  108.558160] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
> [  108.561371] uvcvideo: Unable to create debugfs 6-2 directory.
> [  108.561532] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:03.0/0000:04:00.0/usb6/6-1/6-1:1.4/input/input17
> 

running the unmodified cpp-multicam works with nothing showing up in the dmesg

if I add the following line to cpp-multicam.cpp, it compiles fine
`dev->enable_stream(rs::stream::infrared, rs::preset::best_quality);`

but when I run it I get...

> bloyl@megHPI-acq:~/Projects/librealsense/src$ ./bin/cpp-multicam 
> Starting Intel RealSense R200... RealSense error calling rs_start_source(device:0x243acc0, source:VIDEO):
>     VIDIOC_STREAMON error 5, Input/output error

and the following shows up in dmesg

> [  108.561532] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:03.0/0000:04:00.0/usb6/6-1/6-1:1.4/input/input17
> [  445.369889] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
> [  445.472384] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
> [  445.573406] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
> [  445.673801] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
> [  445.774044] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
> [  445.874416] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
> [  445.974664] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
> [  446.074922] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
> [  446.175285] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
> [  446.275676] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
> [  446.376077] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
> 

Any idea whats going on?

ddiakopoulos, 2016-09-27:
Try also enabling infrared2: `dev->enable_stream(rs::stream::infrared2, rs::preset::best_quality);`

bloyl, 2016-09-27:
Thanks worked like a charm.

closed 2016-09-27 by bloyl