realsenseai/librealsense #303: Setting f200_laser_power on started device results  in hang

tasptz, 2016-10-03:
When I try to set **f200_laser_power** it sometimes causes my program to hang, as it does not return from the **set_option** call. Is this a bug or is it not intended/allowed to set this option on a **started device**?

| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |  3.17.0.0 | 
| Operating System & Version |  Windows 8.1 64 bit |
| Build System |  VS2015 |

dorodnic, 2016-10-03:
Hi tasptz,
It sounds very much like a known bug on Windows 8.1. You can verify by repeating your tests on Windows 10. Alternatively, check the call-stack of the thread stuck on `set_option` - on Windows 8.1 calls to `IKsControl::KsProperty` would sometimes dead-lock inside the driver. 
You can try installing [KB3075872](https://support.microsoft.com/en-us/kb/3075872) and [KB2919355](https://support.microsoft.com/en-us/kb/2919355), designed specifically to solve this problem. 
The only potential software work-around we came up with is calling `IKsControl::KsProperty` on a separate thread and then killing it if it is taking too long. It is complicated and we run the risk of leaking resources, so we decided not to implement this in librealsense.

tasptz, 2016-10-04:
Hi dorodnic,
I had the second update already installed - so the first one seems to have solved the problem. Thanks.

closed 2016-10-04 by tasptz