realsenseai/librealsense #304: Add Qt support for OSX/macOS

kubark42, 2016-10-03:
Qt is a wonderfully cross-platform build environment. Instead of building only for Linux, it would be useful to build for both Linux and macOS. This could lead to a reduction in supported code complexity, as XCode project support could be deprecated.

dorodnic, 2016-10-03:
I fully agree that Qt is wonderfully convenient. 
However, as the project matures we are gradually moving away from the hand-crafted Qt, XCode and Visual Studio projects and encourage users to generate the build script they prefer using CMake. 
While CMake support is not perfect yet (notably it was not working well on OSX) it's definitely getting there.

kubark42, 2016-10-03:
Cool to know it's moving toward a cross-platform solution. Thanks!

P.S. Not that it should influence your project, but I always cringe a little when I see CMake projects. I've had much better success with qmake, especially when trying to diagnose build failures from third-party projects. However, I understand that Qt is an unacceptable dependency for a library which doesn't need a UI.

closed 2016-10-03 by kubark42