realsenseai/librealsense #310: Understanding librealsense Releases

srware, 2016-10-07:
Looking for the right librealsense release on github is the most confusing thing I have seen in weeks.

There seem to be a number of releases running in parallel from both the development branch and master branch with slightly different names (like for ZR300) but there is also a 1.10.0 version being released before 1.9.7 and a couple of times there have been multiple releases of the same version but to a later commit.

I'm proposing coming up with a simpler release mechanism or at least having documentation to explain what all of the releases are and explaining which release people want for a given situation on the repo home page.

Thoughts?

dorodnic, 2016-10-09:
First, here is a summary of what we have right now: 
The master branch represents the known best configuration for the community, at the time of writing this v1.11.0. If you want to play it extra safe you can use the **stable** tag of the legacy branch. 
There are several release trees being maintained in parallel for different customers. Each customer has its own rigid schedule and doesn't want to be affected by changes not relevant to his product. 
However, these trees inevitably merged back so that over time we end up with a single library suitable for all usages. 

We can most certainly add this info to the documentation, if you think it will help the users. 
Any other suggestions?

srware, 2016-10-10:
Thank you for the explanation.

I definitely think it would be good to document the branch and release situation so that users can make an informed decision on which version to use or what will be best for their use case.

A couple more suggestions:

Keep main releases (from master branch) consistent in terms of naming convention with logical semantic versioning. For example, calling one release "v1.9.7" and the next release "v1.10.0 for the ZR300" makes it sound like v1.10.0 is only for the ZR300.

If there are intermediate releases based on the development branch (or another branch) make sure this is clearly highlighted in the name and decide whether semantic versioning should apply to that release.

I think by default users will always grab the latest release based on semantic version so these should always be community releases from master (or pre-release). Anything released specific to a customer should be easily distinguishable from mainline releases if they are not suitable for general use.

Thanks again for your response.

closed 2016-10-18 by dorodnic