realsenseai/librealsense #311: UVC driver patching process failed for R200

fw-timku, 2016-10-09:
| Camera Model | R200| 
| Operating System & Version | Ubuntu 16.04 LTS  |
| Kernel Version (Linux Only)  | 4.4.0-040400-generic |

Hello all,

I am trying to install the librealsense library to acquire images from the Intel RealSense R200 camera. However, I faced similar issue (https://github.com/intel-ros/realsense/issues/12#issuecomment-198079546) when trying to patch the UVC driver. 
The last few output from the patching of './scripts/patch-uvcvideo-4.4.sh v4.4-wily' is as shown:

patching file debian.master/config/i386/config.flavour.lowlatency
patching file debian.master/config/powerpc/config.common.powerpc
patching file debian.master/config/ppc64el/config.common.ppc64el
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
patching file include/uapi/linux/videodev2.h
cp: will not overwrite just-created './.config' with '/usr/src/linux-headers-4.4.0-31-generic/.config'
cp: will not overwrite just-created './.config' with '/usr/src/linux-headers-4.4.0-38-generic/.config'
kuxz@kuxz-UX32LN:~/librealsense$

Now I need to manually execute the copying step and then run the rest of the sections of the script.
The problem is how do I go about manually executing the copying step of the script (https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-uvcvideo-4.4.sh#L50) ??

Thanks!

l3s777, 2016-10-11:
hey! I am using the same version in ubuntu but still got problems with the patch.
when I run:
$ dmesg | grep uvcvideo
I get this:
[  211.394477] uvcvideo: module verification failed: signature and/or required key missing - tainting kernel
[  211.394744] usbcore: registered new interface driver uvcvideo
[  231.152774] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)

any help will be welcome!

ev-mp, 2016-10-13:
@fw-timku , 
Please follow instructions in https://github.com/IntelRealSense/librealsense/issues/312#issuecomment-253310029 to modify the patch for selection of the latest kernel headers, then rerun it.

@l3s777,  'tainting kernel' warning is a notification and it doesn't impede driver's functionality. Actually it is a positive sign of a patched module being loaded into the kernel.
The following info will enable to validate the correctness of the patch , please run and post the output:
v4l2-ctl --list-formats-ext -d /dev/video0
v4l2-ctl --list-formats-ext -d /dev/video1
v4l2-ctl --list-formats-ext -d /dev/video2

l3s777, 2016-10-14:
hi! thanks @ev-mp for the help
These are the responses:
les$ v4l2-ctl --list-formats-ext -d /dev/video0
Failed to open /dev/video0: No such file or directory

les$  v4l2-ctl --list-formats-ext -d /dev/video1
Failed to open /dev/video1: No such file or directory

les$ v4l2-ctl --list-formats-ext -d /dev/video2
Failed to open /dev/video2: No such file or directory

but if I first run:  sudo modprobe uvcvideo
I have these responses:
les$ v4l2-ctl --list-formats-ext -d /dev/video0
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 320x962
			Interval: Discrete 0.017s (60.000 fps)
(... until Index : 4)

les$  v4l2-ctl --list-formats-ext -d /dev/video1
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 314x938
			Interval: Discrete 0.017s (60.000 fps)
$  v4l2-ctl --list-formats-ext -d /dev/video2
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)

	Index       : 1

les$  v4l2-ctl --list-formats-ext -d /dev/video2
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)

thanks for the help.

ev-mp, 2016-10-14:
You actually omitted the relevant part -  pixel format ans its name - that is specified per index.
Please repost the above making sure that the data in bold is preserved for all devices and indexes.
...
	Index       : 1
	Type        : Video Capture
	**Pixel Format: 'Z16 '
	Name        : Depth data 16-bit (Z16)**
		Size: Discrete 628x469
			Interval: Discrete 0.011s (90.000 fps)
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 628x361
			Interval: Discrete 0.011s (9...

closed 2016-11-19 by dorodnic