realsenseai/librealsense #312: ubuntu 14.04 SR300 driver installation issue, distorted/flickering depth image and no infrared image

changhai0551, 2016-10-10:
| Required Info  |   |
|---|---|
| Camera Model |  SR300 | 
| Firmware Version |   | How do I get the firmware version?
| Operating System & Version |   |Ubuntu 14.04.5 LTS \n \l
| Kernel Version (Linux Only)  |   |4.4.0-040400-generic
| Build System |  Makefile | 

Nvidia driver version:
NVRM version: NVIDIA UNIX x86_64 Kernel Module  367.48  Sat Sep  3 18:21:08 PDT 2016
GCC version:  gcc version 4.9.4 (Ubuntu 4.9.4-2ubuntu1~14.04.1)

Hello, everyone
I am installing the drivers for SR300, but when I run **./bin/cpp-capture**, then the depth image flickers and is distorted. There is also no infrared image visible, just like the #266 describes.

But I choose the setp:
**Updated 4.4 Stable Kernel**
Run the following script to install necessary dependencies (GCC 4.9 compiler and openssl) and update kernel to v4.4-wily
./scripts/install_dependencies-4.4.sh
Run the following script to patch uvcvideo.ko
./scripts/patch-uvcvideo-4.4.sh v4.4-wily (note the argument provided to this version of the script)
This script involves shallow cloning the Linux source repository (~100mb), and may take a while

but when finished ./scripts/patch-uvcvideo-4.4.sh v4.4-wily, it outputs that
 cp:Will not using "/usr/src/linux-headers-4.4.0-31-generic/.config" to cover the just created ". /.config" 
cp:Will not using "/usr/src/linux-headers-4.4.0-38-generic/.config"  to cover the just created ". /.config" 

**Check installation by examining the last 50 lines of the dmesg log:**
sudo dmesg | tail -n 50, it outputs

[10009.641288]  [<ffffffff8107dc3c>] warn_slowpath_fmt+0x5c/0x80
[10009.641296]  [<ffffffffc08854fd>] ? uvc_ioctl_enum_fmt+0x9d/0xc0 [uvcvideo]
[10009.641306]  [<ffffffffc0cbf6f6>] v4l_enum_fmt+0xd86/0x10f0 [videodev]
[10009.641316]  [<ffffffffc0cc0261>] __video_do_ioctl+0x291/0x310 [videodev]
[10009.641321]  [<ffffffff81210304>] ? terminate_walk+0x64/0xd0
[10009.641326]  [<ffffffff81214462>] ? path_openat+0x492/0x1280
[10009.641336]  [<ffffffffc0cbfd9e>] video_usercopy+0x33e/0x550 [videodev]
[10009.641345]  [<ffffffffc0cbffd0>] ? video_ioctl2+0x20/0x20 [videodev]
[10009.641355]  [<ffffffffc0cbffc5>] video_ioctl2+0x15/0x20 [videodev]
[10009.641363]  [<ffffffffc0cbb6a3>] v4l2_ioctl+0xd3/0xe0 [videodev]
[10009.641370]  [<ffffffff81219668>] do_vfs_ioctl+0x298/0x480
[10009.641376]  [<ffffffff812153c4>] ? putname+0x54/0x60
[10009.641382]  [<ffffffff8120545f>] ? do_sys_open+0x1af/0x230
[10009.641390]  [<ffffffff812198c9>] SyS_ioctl+0x79/0x90
[10009.641398]  [<ffffffff817fd9b6>] entry_SYSCALL_64_fastpath+0x16/0x75
[10009.641402] ---[ end trace 24dd434cad4e8b7a ]---
[10009.641406] ------------[ cut here ]------------
[10009.641416] WARNING: CPU: 0 PID: 2491 at /home/kernel/COD/linux/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd86/0x10f0 [videodev]()
[10009.641418] Unknown pixelformat 0x00000000
[10009.641420] Modules linked in: snd_usb_audio snd_usbmidi_lib btrfs xor raid6_pq ufs qnx4 hfsplus hfs minix ntfs msdos jfs xfs libcrc32c nvidia_uvm(POE) cmac bnep rfcomm binfmt_misc nls_iso8859_1 nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) snd_hda_codec_hdmi snd_hda_codec_generic arc4 iwlmvm mac80211 uvcvideo intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp videobuf2_vmalloc btusb videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev kvm btrtl btbcm btintel i915 media bluetooth iwlwifi irqbypass hp_wmi snd_hda_intel snd_hda_codec snd_hda_core cfg80211 snd_hwdep crct10dif_pclmul sparse_keymap crc32_pclmul drm_kms_helper hp_wireless wmi aesni_intel drm snd_pcm aes_x86_64 lrw hp_accel gf128mul glue_helper ablk_helper cryptd i2c_algo_bit fb_sys_fops syscopyarea lis3lv02d rtsx_pci_ms memstick snd_seq_midi joydev input_leds serio_raw input_polldev snd_seq_midi_event sysfillrect snd_rawmidi snd_seq parport_pc snd_seq_device sysimgblt ppdev processor_thermal_device snd_timer intel_soc_dts_iosf int3403_thermal int340x_thermal_zone int3400_thermal i2c_hid acpi_thermal_rel mei_me intel_lpss_acpi intel_lpss shpchp mac_hid video acpi_pad mei lp pinctrl_sunrisepoint pinctrl_intel parport snd soundcore hid_generic usbhid hid rtsx_pci_sdmmc psmouse ahci r8169 rtsx_pci mii libahci fjes
[10009.641535] CPU: 0 PID: 2491 Comm: AudioThread Tainted: P        W  OE   4.4.0-040400-generic #201601101930
[10009.641538] Hardware name: HP HP Pavilion Notebook/820D, BIOS F.09 07/19/2016
[10009.641540]  0000000000000000 000000009dd19b66 ffff880245143bf0 ffffffff813c8d94
[10009.641545]  ffff880245143c38 ffff880245143c28 ffffffff8107dba2 ffff880245143da0
[10009.641575]  0000000000000000 0000000000000000 ffff88021c4a8000 0000000000000040
[10009.641610] Call Trace:
[10009.641624]  [<ffffffff813c8d94>] dump_stack+0x44/0x60
[10009.641636]  [<ffffffff8107dba2>] warn_slowpath_common+0x82/0xc0
[10009.641649]  [<ffffffff8107dc3c>] warn_slowpath_fmt+0x5c/0x80
[10009.641668]  [<ffffffffc08854fd>] ? uvc_ioctl_enum_fmt+0x9d/0xc0 [uvcvideo]
[10009.641686]  [<ffffffffc0cbf6f6>] v4l_enum_fmt+0xd86/0x10f0 [videodev]
[10009.641705]  [<ffffffffc0cc0261>] __video_do_ioctl+0x291/0x310 [videodev]
[10009.641719]  [<ffffffff81210304>] ? terminate_walk+0x64/0xd0
[10009.641733]  [<ffffffff81214462>] ? path_openat+0x492/0x1280
[10009.641750]  [<ffffffffc0cbfd9e>] video_usercopy+0x33e/0x550 [videodev]
[10009.641765]  [<ffffffffc0cbffd0>] ? video_ioctl2+0x20/0x20 [videodev]
[10009.641779]  [<ffffffffc0cbffc5>] video_ioctl2+0x15/0x20 [videodev]
[10009.641793]  [<ffffffffc0cbb6a3>] v4l2_ioctl+0xd3/0xe0 [videodev]
[10009.641806]  [<ffffffff81219668>] do_vfs_ioctl+0x298/0x480
[10009.641817]  [<ffffffff812153c4>] ? putname+0x54/0x60
[10009.641825]  [<ffffffff8120545f>] ? do_sys_open+0x1af/0x230
[10009.641836]  [<ffffffff812198c9>] SyS_ioctl+0x79/0x90
[10009.641847]  [<ffffffff817fd9b6>] entry_SYSCALL_64_fastpath+0x16/0x75
[10009.641855] ---[ end trace 24dd434cad4e8b7b ]---
[10357.281979] uvcvideo: Non-zero status (-71) in video completion handler.
[10357.287097] uvcvideo: Non-zero status (-71) in video completion handler.
[10648.649570] usb 1-1: USB disconnect, device number 10
[10648.649578] usb 1-1.2: USB disconnect, device number 11
[10648.847355] usb 2-1: USB disconnect, device number 6
[10648.847364] usb 2-1.1: USB disconnect, device number 7

dottantgal, 2016-10-11:
I have the same issues. An help could be very appreciated

ev-mp, 2016-10-12:
The file "patch-uvcvideo-4.4.sh" has to be updated as it fails to resolve the target version correctly in case of multiple 4.4.0-x kernel headers available ("/usr/src/linux-headers-4.4.0-31... and ..4.4.0.-38 in your case).

In the meanwhile, you can replace the following line in the patch
**CONFIG_LOCATION=/usr/src/linux-headers-$RAW_TAG*-generic/**
with the following (assuming that you'd want to use the latest version available):
**CONFIG_LOCATION=$(ls -d /usr/src/linux-headers-$RAW_TAG*-generic/ | tail -n1)**

l3s777, 2016-10-13:
I applied the patch but when I restarted the pc, it still does not find the module.
the error is:
rs::error was thrown when calling rs_create_context-api_version:11100-:      uvcvideo kernel module is not loaded

ev-mp, 2016-10-13:
The uvcdriver module is loaded on demand, when a first uvc device is attached.
You can run 'lsusb' and look for '8086:0aa5' string, which isSR300's VID/PID.
In case your device is not listed, re-plug/switch usb port to re-enumerate the device and to load the driver.

In case it still fails- collect and share the outputs of 'lsusb' and 'dmesg' for further inspection

l3s777, 2016-10-14:
hi!
when I turn on the computer and run "lsusb"
I have: Bus 004 Device 002: ID 8086:0a80 Intel Corp.
but still...  video4linux seems not installing, so I have the same error when I want to run any application for R200, this error message:

rs::error was thrown when calling rs_create_context-api_version:11100-: uvcvideo kernel module is not loaded

And only when I run this command: sudo modprobe uvcvideo
It seems to work

ev-mp, 2016-10-14:
Apparently the installation process did not work well on your system, but the info you provided is insufficient to assess it. 
Please elaborate on both the installation and runtime steps performed:

Did you follow the Linux installation chapter in the manual, and applied udev rules as specified ?
What is the 'dmesg' output when the [modified ](https://github.com/IntelRealSense/librealsense/issues/312#issuecomment-253310029)'./scripts/patch-uvcvideo-4.4.sh v4.4-wily' script is finished?
What is the 'dmesg' output when you plug in R200 camera?
What is the output of 'lsusb -t' with R200 attached?
What do you mean by 'seems to work' ? Are you able to see all the advertised video outputs correctly? Does it apply for R200, SR300 or both ?

changhai0551, 2016-10-16:
Done as followed:
My problem is that
when finished ./scripts/patch-uvcvideo-4.4.sh v4.4-wily, it outputs that
cp:Will not using "/usr/src/linux-headers-4.4.0-31-generic/.config" to cover the just created ". /.config"
cp:Will not using "/usr/src/linux-headers-4.4.0-38-generic/.config" to cover the just created ". /.config"
which can't make  my  uvcvideo  installed correctly.
so I change the file /scripts/patch-uvcvideo-4.4.sh, just like **ev-mp** says:
replacing the line **CONFIG_LOCATION=/usr/src/linux-headers-$RAW_TAG*-generic/**
with the following (assuming that my version is  4.4.0-040400-generic):
**CONFIG_LOCATION=$(ls -d /usr/src/linux-headers-4.4.0-0404*-generic)**
then rerun the patch ./scripts/patch-uvcvideo-4.4.sh v4.4-wily and finish the following steps as Video4Linux backend describes.
Finally, the image displays correctly.
Done

closed 2016-10-16 by changhai0551