realsenseai/librealsense #317: librealsense without DCM

tibaes, 2016-10-14:
Hi there,

I am in doubt about the DCM. Is it possible to run librealsense without the DCM installed on Windows?
I need to run the SR300 on Windows 8.1, however, only DCM for Windows 10 is available.

| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.17.0.0 | 
| Operating System & Version | Windows10, Windows 8.1 |
| Build System |  VS2013, Makefile |

ddiakopoulos, 2016-10-14:
librealsense is distinctly different from the RealSense SDK and the RealSense DCM. 

librealsense does not need the DCM.

closed 2016-10-14 by ddiakopoulos

dorodnic, 2016-10-15:
To expand on Dimitri's answer and alleviate some of the confusion:
librealsense is a developer kit, open-sourcing the various algorithms and protocols of the RealSense devices. DCM is an official Intel product. It comes with support, recovery and more. 

While librealsense will work with SR300 on Windows 8.1, there is a reason for DCM to require Win10: 
SR300 relies on Win10-specific Microsoft infrastructures for better face detection and background segmentation. librealsense does not include these features at all, that's why it can run on Windows 8.1.

tibaes, 2016-10-17:
I was in doubt because DCM install some virtual devices on Windows 10.
I tested with a Windows 8.1, patched, without DCM and it worked.
Thank you for clearing this to me!