realsenseai/librealsense #320: XDM file generation

joshberry, 2016-10-18:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | latest | 
| Operating System & Version |  Windows 10 |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013 | 

Does librealsense provide the capability to generate an XDM file? If not, are there any libraries that can be used in combination with librealsense to accomplish this? We are hoping to use librealsense in an app that can provide XDM files to another application that's using the RealSense SDK (which understands XDM).

Any guidance or suggestions would be appreciated... Thanks!

dorodnic, 2016-10-18:
Hi,
librealsense does not include any record and playback capabilities at this moment. 
Some work on recording and playback functionality on top of librealsense has been done in [RealSense Linux SDK](https://github.com/IntelRealSense/realsense_sdk), however I fear it is not quite what you are looking for (to the best of my knowledge it uses a different format and currently only available on Linux right now). 
While simple recording and playback is not necessarily out of scope of librealsense, XDM is a proprietary format so it would be complicated to support it.

ddiakopoulos, 2016-10-22:
@joshberry XDM is a fully open-source format co-specified by Intel and Google. The Intel RealSense GitHub organization also hosts some software to help developers read, write, and transcode XDM files for C++ developers here: https://github.com/IntelRealSense/XDM_CPlusPlusParser

closed 2016-10-22 by ddiakopoulos