realsenseai/librealsense #321: Color stream does not work

atenpas, 2016-10-18:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.4.0-38-generic  |
| Build System |  Makefile | 

The color stream of our Intel Realsense SR300 has suddenly stopped to work correctly. The following screenshot shows the output of  `./cpp-tutorial-2-streams`.
[image]

Restarting the computer and plugging the camera does not help with this issue. And the colors were correct yesterday.

dorodnic, 2016-10-18:
Just as a theory - did you connect the device to Windows 10 PC in between? Also, could you please report the FW version from cpp-enumerate-devices?

atenpas, 2016-10-19:
The device has never been connected to a Windows computer. Always to the same Ubuntu computer.

jens25, 2017-05-16:
Hey Guys, 
Is there any fix for this issue? Because I've got he same issue with my SR300.
I checked out the latest librealsense version and tried to grab the rgb stream via cpp-capture and the ros realsense_camera package, but i got this noisy green picture shown in the issue above. If I try to grab the rgb stream with the samples, provided with the realsense SDK on windows, everything works fine. Any suggestions, what i can do, to fix this?
| Info        |               |
| ------------- |:-------------:|
| Camera     | SR300 |
| Kernel      | 4.4.0-77 |
| Firmware: | 3.21.0.0  |

RichieLee93, 2017-09-23:
Hi Guys,
I met the exact same problem as you all. Have anyone solved this problem?

jens25, 2017-09-26:
We send back the camera and got a new one. That was the only way, we could fix this problem.

jlgarcia75, 2017-10-17:
Hello,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

TheElec, 2017-11-24:
@jlgarcia75 I seem to have the same issues, is there any fix available yet? I'd rather not send it back considering I had the camera shipped overseas.

jlgarcia75, 2017-11-27:
@TheElec,

Unfortunately, this seems to be a camera issue as we cannot reproduce it.  The user above resolved it by getting a new camera.

Regards,
Jesus G.
Intel Customer Support

TheElec, 2017-11-27:
@jlgarcia75 I managed to fix it by downgrading with the intel_rs_dcm_sr300_3.0.24.59748.exe installer, and then going back to the latest firmware. It seems to be an issue with upgrading from factory firmware all the way to latest.

jlgarcia75, 2017-11-27:
@TheElec,

Wow, that is great to know! We have never encountered this before. Thank you for letting us know how to fix this. I'm sure this can help a lot of people.

Regards,
Jesus G.
Intel Customer Support

emigmo, 2017-12-25:
the color stream is the green noisy image, any solution?(don't get a new one)
[image]

ev-mp, 2017-12-25:
Hi @emigmo ,
The SR300 is configured into development mode. Is it possible that you've got an engineering/pre-sale sample.
To fix it you need to:
- Build and run realsense terminal tool (rs-terminal or cpp-terminal, depending on librealsense version)
- Connect the SR300 device
- Run the following command in the terminal (copy&paste):
14 00 ab cd 0b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
- replug SR300 and check with the viewer whether this artifact was gone.

ev-mp, 2017-12-28:
Hi @emigmo ,
Did you manage to resolve the issue, or do you need additional assistance  ?

closed 2018-02-04 by dorodnic