realsenseai/librealsense #324: RSSDK Recorded Stream Format

jmaye, 2016-10-20:
Hi,

Sorry for annoying you with this issue on this github repository, but I don't know where it should go.

I was able to decode a rssdk stream recorded with the Windows SDK on Linux. I can get H.264 frames for color and so-called LZO frames for IR/depth. However, the LZO format used there is not standard. I found no library able to decompress it successfully. Is it something custom for Intel? Any hint?

Thanks in advance.

jmaye, 2016-10-20:
Problem semi-solved:) I could decompress the LZO depth using Intel IPP library. Apparently, it does not follow the standard. It would be nice to know in which way.

digitologist, 2016-11-01:
This is the best place to find the file format info, from the RSSDK documentation:
https://software.intel.com/sites/landingpage/realsense/camera-sdk/v1.1/documentation/html/doc_appendix_recorded_file_forma.html

jmaye, 2016-11-02:
Thanks for the link, I've seen a similar one.

The problem here is really the lzo format though. It seems to be non-standard. Only IPP tools are able to decompress it.

closed 2016-11-19 by dorodnic