realsenseai/librealsense #326: api version mismatch causing ros crash?

jacksonkr, 2016-10-22:
Camera Model | R200 
Operating System & Version |   Ubuntu 14.04 amd64
Kernel Version (Linux Only)  |   3.19.0-73-generic
Build System |  Makefile 

```
$ roslaunch realsense_camera r200_nodelet_rgbd.launch 
... logging to /home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/roslaunch-jackson-MacBookPro-14037.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.100:59541/

SUMMARY
========

PARAMETERS
 * /camera/camera/base_frame_id: camera_link
 * /camera/camera/camera_type: R200
 * /camera/camera/color_fps: 30
 * /camera/camera/color_frame_id: camera_rgb_frame
 * /camera/camera/color_height: 480
 * /camera/camera/color_optical_frame_id: camera_rgb_optica...
 * /camera/camera/color_width: 640
 * /camera/camera/depth_fps: 30
 * /camera/camera/depth_frame_id: camera_depth_frame
 * /camera/camera/depth_height: 360
 * /camera/camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/camera/depth_width: 480
 * /camera/camera/enable_color: True
 * /camera/camera/enable_depth: True
 * /camera/camera/enable_pointcloud: False
 * /camera/camera/enable_tf: True
 * /camera/camera/ir2_frame_id: camera_ir2_frame
 * /camera/camera/ir_frame_id: camera_ir_frame
 * /camera/camera/mode: manual
 * /camera/camera/serial_no: 
 * /camera/camera/usb_port_id: 
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/disparity_depth/max_range: 4.0
 * /camera/disparity_depth/min_range: 0.5
 * /camera/disparity_registered_sw/max_range: 4.0
 * /camera/disparity_registered_sw/min_range: 0.5
 * /camera/points_xyzrgb_sw_registered/queue_size: 100
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /camera/
    camera (nodelet/nodelet)
    camera_nodelet_manager (nodelet/nodelet)
    depth_metric (nodelet/nodelet)
    depth_metric_rect (nodelet/nodelet)
    depth_points (nodelet/nodelet)
    depth_rectify_depth (nodelet/nodelet)
    depth_registered_sw_metric_rect (nodelet/nodelet)
    disparity_depth (nodelet/nodelet)
    disparity_registered_sw (nodelet/nodelet)
    ir_rectify_ir (nodelet/nodelet)
    points_xyzrgb_sw_registered (nodelet/nodelet)
    register_depth_rgb (nodelet/nodelet)
    rgb_debayer (nodelet/nodelet)
    rgb_rectify_color (nodelet/nodelet)
    rgb_rectify_mono (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
[... 28 lines cut ...]
log file: /home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-camera_nodelet_manager-1*.log
[camera/depth_rectify_depth-7] process has died [pid 14082, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load image_proc/rectify camera_nodelet_manager --no-bond image_mono:=depth/image_raw image_rect:=depth/image_rect_raw __name:=depth_rectify_depth __log:=/home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-depth_rectify_depth-7.log].
log file: /home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-depth_rectify_depth-7*.log
[camera/depth_metric-9] process has died [pid 14106, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/convert_metric camera_nodelet_manager --no-bond image_raw:=depth/image_raw image:=depth/image __name:=depth_metric __log:=/home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-depth_metric-9.log].
log file: /home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-depth_metric-9*.log
[camera/depth_points-10] process has died [pid 14119, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyz camera_nodelet_manager --no-bond image_rect:=depth/image_rect_raw points:=depth/points __name:=depth_points __log:=/home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-depth_points-10.log].
log file: /home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-depth_points-10*.log
[camera/register_depth_rgb-11] process has died [pid 14124, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/register camera_nodelet_manager --no-bond rgb/camera_info:=rgb/camera_info depth/camera_info:=depth/camera_info depth/image_rect:=depth/image_rect_raw depth_registered/image_rect:=depth_registered/sw_registered/image_rect_raw __name:=register_depth_rgb __log:=/home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-register_depth_rgb-11.log].
log file: /home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-register_depth_rgb-11*.log
[camera/points_xyzrgb_sw_registered-12] process has died [pid 14137, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/point_cloud_xyzrgb camera_nodelet_manager --no-bond rgb/image_rect_color:=rgb/image_rect_color rgb/camera_info:=rgb/camera_info depth_registered/image_rect:=depth_registered/sw_registered/image_rect_raw depth_registered/points:=depth_registered/points __name:=points_xyzrgb_sw_registered __log:=/home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-points_xyzrgb_sw_registered-12.log].
log file: /home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-points_xyzrgb_sw_registered-12*.log
[camera/disparity_depth-14] process has died [pid 14158, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/disparity camera_nodelet_manager --no-bond left/image_rect:=depth/image_rect_raw right:=projector left/disparity:=depth/disparity __name:=disparity_depth __log:=/home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-disparity_depth-14.log].
log file: /home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-disparity_depth-14*.log
[camera/disparity_registered_sw-15] process has died [pid 14173, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load depth_image_proc/disparity camera_nodelet_manager --no-bond left/image_rect:=depth_registered/sw_registered/image_rect_raw right:=projector left/disparity:=depth_registered/disparity __name:=disparity_registered_sw __log:=/home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-disparity_registered_sw-15.log].
log file: /home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-disparity_registered_sw-15*.log
[camera/camera-2] process has died [pid 14056, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load realsense_camera/R200Nodelet camera_nodelet_manager camera/depth/image_raw:=depth/image_raw camera/color/image_raw:=rgb/image_raw camera/ir/image_raw:=ir/image_raw camera/ir2/image_raw:=ir2/image_raw camera/depth/points:=depth/points camera/depth/camera_info:=depth/camera_info camera/color/camera_info:=rgb/camera_info camera/ir/camera_info:=ir/camera_info camera/ir2/camera_info:=ir2/camera_info __name:=camera __log:=/home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-camera-2.log].
log file: /home/jackson/.ros/log/bbf68ea8-9881-11e6-81d3-00904c0df43e/camera-camera-2*.log
```

Extra (related?):
```
$ cpp-capture 
RealSense error calling rs_create_context(api_version:11100):
    api version mismatch
```

jacksonkr, 2016-10-23:
So the `api version mismatch` was definitely related. I did a number of things and something eventually worked. I'm not sure what it was that worked but I'm pretty sure it was kernel upgrade.

```
$ cd librealsense/build
$ sudo make uninstall

$ sudo find / -name librealsense* # then I deleted everything that was found
```

```
sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade
lsb_release -a
```

I went from kernel 3.19 to 4.4 which changed a couple of the install steps. Finally I can run cpp-capture again!

jacksonkr, 2016-10-23:
Sure enough, even after everything was working, if you run `sudo apt-get install ros-[distro]-librealsense` then you'll get the api mismatch version.

Here's the trick:
realsense ros requires that package in order to run.
once the ros-librealsense is installed, the dammage is done. Uninstalling the package doesn't appear to help. I'm assuming that in the catkin_make some of the ros-librealsense files are scattered to the system.

Any ideas how to get past this?

mdhorn, 2016-10-24:
@jacksonkr 
Please keep ROS related Issue on the [ROS realsense camera GitHub](https://github.com/intel-ros/realsense).

There should be ONE and only one version of librealsense install on your system. Librealsense does a check to ensure that any binary which uses the library was compiled with that exact version of the library.

If you are using ROS, you should stick with the ROS Debian packages as the realsense-camera Debian package is guaranteed to be compiled with the version of librealsense from the Debian package.

If for some reason you need to build the realsense-camera ROS package, it should be compiled against the Debian librealsense installed in /opt/ros/<distro>.  If you need to "hack on" librealsense and build it from code, then you have to uninstall both the realsense-camera and librealsense ROS Debian packages, build librealsense from source first, then build realsense-camera from source.

The librealsense team here is support the library cross-platform, and is not tracking any issues released to ROS. If you're still having ROS released issues, let's keep the tickets on the ROS GitHub. https://github.com/intel-ros/realsense

Thanks.

closed 2016-10-24 by mdhorn