realsenseai/librealsense #327: New SR300 IR/Depth information

Voidminded, 2016-10-22:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | Intel RealSense SR300, Firmware: 3.17.0.0.  | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |
| Build System |  Makefile | 

Hi,

I've bought a new SR300 dev kit, on windows it works fine but in linux using this driver I can't get the IR and depth image, (tried all the samples, also the ROS version, all working with 2 other F200 and R200 sensors that I have but not with the SR300). 
In ROS, the output IR image is constant 0.0 for all the pixels.
The SR300 came with 3.15.0.0 firmware, upgrading to following firmware did not solve the issue:
Intel RealSense SR300, Firmware: 3.17.0.0.
Can you please lead me to correct direction (if it is a data conversion issue (which probably is) or because of driver issue).

Thanks

Voidminded, 2016-10-22:
It was "uvcvideo" issue, fixed !

JudyMRSD, 2016-11-06:
Hi Voidminded, I am having the exactly same issue. Do you mind give me some more details on how you solved it? Thank you very much!

Voidminded, 2016-11-07:
If you follow up the "Video4Linux backend" part of installation document (https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) it will fix the issue, just make sure there is no error accruing during these steps.
BTW if you tell me which version of Ubuntu/Firmware you are using I might be able to help more precisely.

JudyMRSD, 2016-11-09:
Thank you for your quick response. Following are details for my computer:
Ubuntu: 14.04
ROS: indigo
Kernel: 4.4.0-040400-generic

I tried the video4linux part, but the depth data has the same issue.

Thank you very much!

hironoriyh, 2016-12-09:
I have exactrly the same issue as  @JudyMRSD mentioned. I have a same setup for my Ubuntu version and using VF0810 new SR 300. I was able to read correctly the previous version though. Probably update messed things up?

JudyMRSD, 2016-12-09:
@hironoriyh I didn't figure it out so I switched to another sensor : )

closed 2017-01-08 by dorodnic