realsenseai/librealsense #330: Bad color stream buffer after reinitialization.

catree, 2016-10-25:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.15.0.0 | 
| Operating System & Version | Fedora 24 |
| Kernel Version (Linux Only)  | 4.7.9-200.fc24.x86_64 |
| Build System | QtCreator | 

Not sure if it is an issue or if I missed something.

The following code produces a bad color image buffer:
```
  rs::context ctx;
  rs::device & dev = *ctx.get_device(0);

  int width = 640, height = 480;
  dev.enable_stream(rs::stream::color, width, height, rs::format::bgr8, 60);

  // Start our device
  dev.start();

  // Stop our device
  dev.stop();

  width = 1920, height = 1080;
  dev.enable_stream(rs::stream::color, width, height, rs::format::bgr8, 30);

  cv::Mat color_image_hd(height, width, CV_8UC3);

  // Start our device
  dev.start();

  while (true) {
    dev.wait_for_frames();

    memcpy(color_image_hd.data, (uchar *) dev.get_frame_data(rs::stream::color), width*height*3);

    cv::imshow("Color image HD", color_image_hd);
    char c = cv::waitKey(30);
    if (c == 27) {
      break;
    }
  }
```

Example image:
[image]

Please note that the following code is ok (consuming the color image):
```
  rs::context ctx;
  rs::device & dev = *ctx.get_device(0);

  int width = 640, height = 480;
  dev.enable_stream(rs::stream::color, width, height, rs::format::bgr8, 60);

  // Start our device
  dev.start();

  cv::Mat color_image(height, width, CV_8UC3);
  while (true) {
    dev.wait_for_frames();

    memcpy(color_image.data, (uchar *) dev.get_frame_data(rs::stream::color), width*height*3);

    cv::imshow("Color image", color_image);
    char c = cv::waitKey(30);
    if (c == 27) {
      break;
    }
  }

  // Stop our device
  dev.stop();

  width = 1920, height = 1080;
  dev.enable_stream(rs::stream::color, width, height, rs::format::bgr8, 30);

  cv::Mat color_image_hd(height, width, CV_8UC3);

  // Start our device
  dev.start();

  while (true) {
    dev.wait_for_frames();

    memcpy(color_image_hd.data, (uchar *) dev.get_frame_data(rs::stream::color), width*height*3);

    cv::imshow("Color image HD", color_image_hd);
    char c = cv::waitKey(30);
    if (c == 27) {
      break;
    }
  }
```

But this code seems also ok (depth stream without consuming the stream buffer):
```
  rs::context ctx;
  rs::device & dev = *ctx.get_device(0);

  int width = 640, height = 480;
  dev.enable_stream(rs::stream::depth, width, height, rs::format::z16, 60);

  // Start our device
  dev.start();

  // Stop our device
  dev.stop();

  width = 640, height = 240;
  try {
    dev.enable_stream(rs::stream::depth, width, height, rs::format::z16, 60);
  } catch (std::exception &e) {
    std::cerr << "Exception: " << e.what() << std::endl;
  }

  cv::Mat depth_image(height, width, CV_16S);
  cv::Mat depth_image_display(height, width, CV_8U);

  // Start our device
  dev.start();

  while (true) {
    dev.wait_for_frames();

    memcpy(depth_image.data, (uchar *) dev.get_frame_data(rs::stream::depth), width*height*2);
    depth_image.convertTo(depth_image_display, CV_8U, 0.5);

    cv::imshow("Depth", depth_image_display);
    char c = cv::waitKey(30);
    if (c == 27) {
      break;
    }
  }
```

catree, 2016-11-10:
Is anyone willing to test the code to confirm / disconfirm the bug with the SR300?

Only OpenCV is needed as a dependency.

dorodnic, 2016-11-10:
I tried to reproduce with very similar code (just without OpenCV) couple of days ago and couldn't see the issue. I would recommend doing couple more experiments to try to identify where it is coming from. This may be camera hardware failure, USB3 controller incompatibility, or some kind of driver issue (we work with Ubuntu, but SR300 was officially verified to work only with Windows 10)

catree, 2016-11-15:
I tried with Ubuntu 16.04 on the same computer and got the same issue.

Maybe it is an hardware related problem.

jlgarcia75, 2017-10-17:
Hello @catree ,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

catree, 2017-10-18:
@jlgarcia75 I will retry with the latest version of librealsense and will report here.

catree, 2017-10-20:
@jlgarcia75 

I have retried with:

- librealsense legacy branch 009f011
- Ubuntu 16.04 4.4.0-97-generic
- Firmware version: 3.15.0.0
- libusb-1.0:
```
||/ Nom                  Version         Architecture    Description
+++-====================-===============-===============-=============================================
ii  libusb-1.0-0:amd64   2:1.0.20-1      amd64           userspace USB programming library
ii  libusb-1.0-0-dev:amd 2:1.0.20-1      amd64           userspace USB programming library development
ii  libusb-1.0-doc       2:1.0.20-1      all             documentation for userspace USB programming
```
The camera is the Creative SR300 first version similar to this one:
[image]

I still have the issue:
[image]

The code is (modified from `cpp-tutorial-2-streams.cpp`):
```
// First include the librealsense C++ header file
#include <librealsense/rs.hpp>
#include <cstdio>

// Also include GLFW to allow for graphical display
#include <GLFW/glfw3.h>

int main() try
{
    // Create a context object. This object owns the handles to all connected realsense devices.
    rs::context ctx;
    printf("There are %d connected RealSense devices.\n", ctx.get_device_count());
    if(ctx.get_device_count() == 0) return EXIT_FAILURE;

    // This tutorial will access only a single device, but it is trivial to extend to multiple devices
    rs::device * dev = ctx.get_device(0);
    printf("\nUsing device 0, an %s\n", dev->get_name());
    printf("    Serial number: %s\n", dev->get_serial());
    printf("    Firmware version: %s\n", dev->get_firmware_version());

    dev->enable_stream(rs::stream::color, 640, 480, rs::format::rgb8, 30);

    dev->start();
    if(dev->is_streaming()) dev->stop();
    
    dev->enable_stream(rs::stream::color, 960, 540, rs::format::rgb8, 30);
    dev->start();

    // Open a GLFW window to display our output
    glfwInit();
    GLFWwindow * win = glfwCreateWindow(960, 540, "test bug", nullptr, nullptr);
    glfwMakeContextCurrent(win);
    while(!glfwWindowShouldClose(win))
    {
        // Wait for new frame data
        glfwPollEvents();
        dev->wait_for_frames();

        glClear(GL_COLOR_BUFFER_BIT);
        glPixelZoom(1, -1);

        // Display color image as RGB triples
        glRasterPos2f(-1, 1);
        glDrawPixels(960, 540, GL_RGB, GL_UNSIGNED_BYTE, dev->get_frame_data(rs::stream::color));

        glfwSwapBuffers(win);
    }

    return EXIT_SUCCESS;
}
catch(const rs::error & e)
{
    // Method calls against librealsense objects may throw exceptions of type rs::error
    printf("rs::error was thrown when calling %s(%s):\n", e.get_failed_function().c_str(), e.get_failed_args().c_str());
    printf("    %s\n", e.what());
    return EXIT_FAILURE;
}
```
The procedure that leads to the issue is basically:

- enable 640x480 color stream
- start the device
- stop the device
- enable 960x540 color stream
- start the device
- consume and display the frames

If you are not able to reproduce the issue and as the issue is more or less an edge case (start -> stop -> start without consuming the frames after the first start), you can close the issue.

jlgarcia75, 2017-11-01:
HJi @catree,

I was able to repro your issue. It happens whenever you change resolutions and can be replicated with your code using any resolution in the place of 640x480 and 960x540. If you keep the same resolutions the issue does not occur. 

Have you retried with the latest version of librealsense? The legacy branch is no longer supported.

Regards,
Jesus
Intel Customer Support

closed 2017-11-08 by jlgarcia75