realsenseai/librealsense #333: When to build from source?

jacksonkr, 2016-10-27:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |  ? | 
| Operating System & Version | ubuntu 14.04  |
| Kernel Version (Linux Only)  |  3.10.40-grinch-21.3.4 |
| Build System |  Makefile | 

What do I do if my kernel is not supported with the debian package? Is that a good reason to build from the github source?

dorodnic, 2016-11-07:
Your Linux kernel has to support the custom pixel formats exposed by the device. You can either move to a kernel version where the relevant camera support has been already upstreamed and included (for R200 it is 4.4), you can apply provided kernel patch for several kernels we support (that's what the Debian package is doing under the hood) or you can try to patch the kernel manually. 
The lowest version that the [Installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) supports is the 3.19.xx kernel, but the library is not being validated on it and hence it's not recommended.

jacksonkr, 2016-11-07:
So I'm installing on a Jetson TK1 so the range of supported kernels is narrowed. After googling a while I found this guy Jim who runs jetsonhacks.com make, he made a video on [how to install the realsense_camera on a Jetson TK1](http://www.jetsonhacks.com/2016/06/23/realsense-camera-ros-install-jetson-tk1/).

closed 2016-11-07 by jacksonkr