realsenseai/librealsense #335: R200 parameter settings for outdoor using on c++ ( Linux )  ?

ChanghyeonKim93, 2016-10-28:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 | 
| Firmware Version |   1.0.72.06| 
| Operating System & Version |  Linux Ubuntu 14.04 LTS , more recent version|
| Kernel Version (Linux Only)  | 4.2.0  |
| Build System | Makefile | 

Hi, I'm a user of intel realsense R200. 
Thanks for answering of https://github.com/IntelRealSense/librealsense/issues/208 , I can set the realsense parameters by running "cpp-config-ui.cpp".
But, parameter settings just are valid only when "cpp-config-ui.cpp" 's GUI. So I cannot use the set parameters on my own c++ code applications . 
I analysed the source code of the "cpp-config-ui.cpp", but I cannot find the way to set realsense params on my code. 
Could you tell me the way to use the parameter settings on my code? or what part of the librealsense github do I need to see so as to solve the problem? 

Thanks.

mdhorn, 2016-10-28:
Looks like you answered yourself in #208 already.
But take a look at the documentation:
https://github.com/IntelRealSense/librealsense#documentation
All of the APIs are in rs.h or rs.hpp and can get formatted nicely with doxygen following the directions in the above link.

closed 2016-10-28 by mdhorn

ChanghyeonKim93, 2016-10-30:
Thanks a lot!