realsenseai/librealsense #337: scripts/sign-file.c:23:30: fatal error: openssl/opensslv.h: No such file or directory

RickBlacker, 2016-10-28:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Operating System & Version |   Ubuntu 16.04 |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013, XCode or Makefile | 
I'm rebuilding my Ubuntu system. I have done done sudo-apt get update / upgrade.  Now running through the LibRealSense aspects of my system rebuild.  When running  **./scripts/patch-uvcvideo-16.04.simple.sh.**  I'm getting this error... 
**scripts/sign-file.c:23:30: fatal error: openssl/opensslv.h: No such file or directory**

Anyone else see this error?

RickBlacker, 2016-10-28:
I did some searching, looks like I needed to install 
sudo apt-get install libssl-dev

Went back and re-ran the patch and things seemed to go ok.

RickBlacker, 2016-10-28:
I did see this after running 
sudo modprobe uvcvideo 
and then
sudo dmesg | tail -n 50

are these issues I need to be concerned about?

**[ 3446.945377] IPv6: ADDRCONF(NETDEV_UP): wlp57s0: link is not ready**
[ 4285.788482] media: Linux media interface: v0.10
[ 4285.794801] Linux video capture interface: v2.00
**[ 4285.802611] uvcvideo: module verification failed: signature and/or required key missing - tainting kernel**
[ 4285.802959] usbcore: registered new interface driver uvcvideo
[ 4285.802961] USB Video Class driver (1.1.1)

lotfielhafi, 2017-01-28:
I confirm I have the same error on Ubuntu 16.04 with the missing openssl/opensslv.h.

ev-mp, 2017-09-13:
Openssl library is required to build uvcvideo kernel module. The current script `patch-realsense-ubuntu-xenial.sh` verifies and installs the missing dependencies (including openssl) if necessary.
The specific **tainting kernel** log message is informative only and does not affect the patched module functionality. It can be safely disregarded.

closed 2017-09-13 by ev-mp

verdy-p, 2022-05-31:
When building the kernel you may also need to install libelf-dev (there's a warning emitted in the compilation, but a missing required will be missing if this preinstallation is not there). This is still true today with kernel 5.4.* stable branch (and all its many patches), even if there are newer major kernel versions.