realsenseai/librealsense #339: V4L2_PIX_FMT_SRGGB10P Undeclared

CWallbridge, 2016-11-01:
| Required Info  |   |
|---|---|
| Camera Model | F200 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 14.04.5 LTS - Virtual Machine using VMware Fusion  |
| Kernel Version (Linux Only)  |  3.19.0-73-generic  |
| Build System |  Makefile | 

While I am aware that VM's are not supported this issue would appear to be not caused by the VM. The installation fails when using either:
`sudo apt install ros-indigo-realsense-camera`
or the script described in the installation.md having cloned the git repository:
`./scripts/patch-uvcvideo-ubuntu-mainline.sh`

The installation fails with the following message:
```
/home/<user>/uvc_patch/linux-lts-vivid-3.19.0/drivers/media/usb/uvc/uvc_driver.c:163:11: error: ‘V4L2_PIX_FMT_SRGGB10P’ undeclared here (not in a function)
   .fcc  = V4L2_PIX_FMT_SRGGB10P,
           ^
```

mdhorn, 2016-11-01:
@CWallbridge As far as the ROS Debian package, I'm surprised that you have a 3.x kernel with 14.04.5 -- my guess is that VM Fusion put a hold on the kernel to support the extension it required. The ROS Debian package requires you enable kernel sources:
http://wiki.ros.org/librealsense#Installation_Prerequisites
`wget -O enable_kernel_sources.sh http://bit.ly/en_krnl_src`
`bash ./enable_kernel_sources.sh`

CWallbridge, 2016-11-02:
@mdhorn Thanks for your response. I have tried running that script and I am still getting the same problem. Is this likely caused by the mismatch of kernel and Ubuntu version then?

mdhorn, 2016-11-03:
@CWallbridge After running the enable_kernel_sources.sh script, you still get an error when trying to install the ROS Debian package?
`sudo apt install ros-indigo-realsense-camera`

I know we had tested with the stocker 14.04.1 kernel, which should be 3.19.x.
If you are using an F200, you probably want to update to a newer kernel.
Can you enable the use of the Xenial kernel in the VM which should be the default for 14.04.5?
`sudo apt-get install linux-image-generic-lts-xenial`

CWallbridge, 2016-11-03:
@mdhorn I have actually updated to 4.4 using the script mentioned in the install guide. This does seem to have fixed the problem. I forgot to update though as I am still having issues getting the actual image to come through, the camera is actually being picked up and some data is coming through. This does seem to be the issue warned about with virtual machines, the usb interface is being overwritten by the virtual machine's. I am now waiting to get a machine that I can install a separate Ubuntu partition onto. Thanks for your help.

leonidk, 2017-02-24:
This is an issue with the new patches and the 3.x family of kernels. Either we explicitly say we don't support the old kernels or we should fix the patches.

oordan, 2017-09-13:
librealsense supports kernel 4.4 and newer.

closed 2017-09-13 by oordan