realsenseai/librealsense #341: Versioning

srware, 2016-11-03:
Sorry to bring this issue up again but it's making my life maintaining the Yocto meta layer difficult.

Releases/tags in my opinion cannot be a moving target. Release v1.11.0 started at commit 74ff66da50210e6b9edc3157411bad95c209740f and has now changed to 6626123bf36ff4e6399081a6972d547ef6498bef so it has 2 parents without even a minor version bump. This has been the case for v1.10.0 and previous releases also and isn't maintainable in my view.

I'm up for having a 1.11.x branch which receives minor bug fixes but periodic releases from that branch should have a minor version bump also.

Interested in your opinions on this?

dorodnic, 2016-11-03:
The practice of retagging existing releases was mostly useful for integration with ROS. Since at the moment parta of the ROS debian package is maintained inside librealsense master branch, whenever release is scheduled, the package gets updated shortly after without any changes to functionality. While minor bug fixes were sometimes addressed between retags, they were mostly included by mistake. I personally prefer bumping the minor version.
To eliminate this practice I'm working with @mdhorn on splitting ROS debian into a seperate branch with distinct releases.
Thanks for raising this up, your concern is completely justified.

dorodnic, 2016-11-19:
rosdebian branch is now in use for ROS releases and existing tags will no longer be adjusted.

closed 2016-11-19 by dorodnic