realsenseai/librealsense #343: Strange failures upon connecting to powered USB3 port

vadali, 2016-11-06:
Hey,
When connecting the camera to a powered usb3 port, I am getting the following errors in dmesg (long and repetitive..). 

I followed the installation guide, patched and installed. Sometimes cpp-capture (and the rest) works for a short while, and after some time they die with a nasty "VIDIOC_DQBUF error 19, No such device"

What do you think? Is this related? Any clues?

> dmesg                                                                                
> [ 3186.690504] usb 3-1: new high-speed USB device number 17 using xhci_hcd
> [ 3186.823075] usb 3-1: New USB device found, idVendor=05e3, idProduct=0610
> [ 3186.823079] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [ 3186.823080] usb 3-1: Product: USB2.0 Hub
> [ 3186.823082] usb 3-1: Manufacturer: GenesysLogic
> [ 3186.823911] hub 3-1:1.0: USB hub found
> [ 3186.824438] hub 3-1:1.0: 2 ports detected
> [ 3186.930670] usb 4-1: new SuperSpeed USB device number 53 using xhci_hcd
> [ 3186.949073] usb 4-1: New USB device found, idVendor=05e3, idProduct=0616
> [ 3186.949076] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [ 3186.949078] usb 4-1: Product: USB3.0 Hub
> [ 3186.949079] usb 4-1: Manufacturer: GenesysLogic
> [ 3186.951041] hub 4-1:1.0: USB hub found
> [ 3186.951331] hub 4-1:1.0: 2 ports detected
> [ 3187.742415] usb 3-1.2: new full-speed USB device number 18 using xhci_hcd
> [ 3187.987281] usb 3-1.2: New USB device found, idVendor=041e, idProduct=4099
> [ 3187.987285] usb 3-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 3187.987287] usb 3-1.2: Product: Creative VF0800
> [ 3187.987289] usb 3-1.2: Manufacturer: Creative Technology Ltd
> [ 3187.987290] usb 3-1.2: SerialNumber: 201501143044
> [ 3188.123830] input: Creative Technology Ltd Creative VF0800 as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1.2/3-1.2:1.3/0003:041E:4099.0007/input/input128
> [ 3188.178633] hid-generic 0003:041E:4099.0007: input,hiddev0,hidraw0: USB HID v1.11 Device [Creative Technology Ltd Creative VF0800] on usb-0000:00:14.0-1.2/input3
> [ 3190.026296] usb 4-1.1: new SuperSpeed USB device number 54 using xhci_hcd
> [ 3190.042981] usb 4-1.1: New USB device found, idVendor=8086, idProduct=0a66
> [ 3190.042985] usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 3190.042987] usb 4-1.1: Product: Intel(R) RealSense(TM) 3D Camera (Front F200)
> [ 3190.042988] usb 4-1.1: Manufacturer: Intel(R) RealSense(TM) 3D Camera (Front F200)
> [ 3190.042990] usb 4-1.1: SerialNumber: 047140082010
> [ 3190.044209] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) 3D Camera (Front F200) (8086:0a66)
> [ 3190.045949] input: Intel(R) RealSense(TM) 3D Camer as /devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1.1/4-1.1:1.0/input/input129
> [ 3190.046384] uvcvideo: Found UVC 1.10 device Intel(R) RealSense(TM) 3D Camera (Front F200) (8086:0a66)
> [ 3190.048013] uvcvideo: Unable to create debugfs 4-54 directory.
> [ 3190.048122] input: Intel(R) RealSense(TM) 3D Camer as /devices/pci0000:00/0000:00:14.0/usb4/4-1/4-1.1/4-1.1:1.2/input/input130
> [ 3190.062215] ------------[ cut here ]------------
> [ 3190.062240] WARNING: CPU: 6 PID: 2282 at /build/linux-kOWph6/linux-4.4.0/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd86/0x10f0 [videodev]()
> [ 3190.062241] Unknown pixelformat 0x5a564e49
> [ 3190.062243] Modules linked in: uvcvideo(OE) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media hid_generic usbhid snd_usb_audio hid snd_usbmidi_lib drbg ansi_cprng ctr ccm rfcomm ec_sys bnep arc4 binfmt_misc x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm snd_hda_codec_via snd_hda_codec_hdmi snd_hda_codec_generic irqbypass iwlmvm snd_hda_intel crct10dif_pclmul snd_hda_codec crc32_pclmul btusb snd_hda_core btrtl mac80211 snd_hwdep btbcm btintel snd_pcm bluetooth snd_seq_midi snd_seq_midi_event aesni_intel snd_rawmidi aes_x86_64 lrw gf128mul glue_helper ablk_helper snd_seq cryptd iwlwifi mei_me snd_seq_device snd_timer rtsx_pci_ms cfg80211 joydev input_leds snd memstick serio_raw soundcore mei lpc_ich shpchp tpm_infineon mac_hid parport_pc ppdev
> [ 3190.062295]  lp parport autofs4 btrfs xor raid6_pq dm_mirror dm_region_hash dm_log i915 rtsx_pci_sdmmc i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops e1000e drm psmouse ahci rtsx_pci libahci ptp pps_core wmi fjes video [last unloaded: videobuf2_memops]
> [ 3190.062312] CPU: 6 PID: 2282 Comm: AudioThread Tainted: G        W  OE   4.4.0-45-generic #66-Ubuntu
> [ 3190.062314] Hardware name: System76, Inc. Galago UltraPro/W740SU                          , BIOS 4.6.5 12/12/2013
> [ 3190.062315]  0000000000000286 00000000b9d147ed ffff8803d22afbf0 ffffffff813f1fe3
> [ 3190.062318]  ffff8803d22afc38 ffffffffc0808a78 ffff8803d22afc28 ffffffff81081212
> [ 3190.062321]  ffff8803d22afda0 0000000000000000 0000000000000001 ffff88040aca73c0
> [ 3190.062323] Call Trace:
> [ 3190.062330]  [<ffffffff813f1fe3>] dump_stack+0x63/0x90
> [ 3190.062335]  [<ffffffff81081212>] warn_slowpath_common+0x82/0xc0
> [ 3190.062337]  [<ffffffff810812ac>] warn_slowpath_fmt+0x5c/0x80
> [ 3190.062342]  [<ffffffffc083772d>] ? uvc_ioctl_enum_fmt+0x9d/0xc0 [uvcvideo]
> [ 3190.062348]  [<ffffffffc07f36f6>] v4l_enum_fmt+0xd86/0x10f0 [videodev]
> [ 3190.062354]  [<ffffffffc07f42c1>] __video_do_ioctl+0x291/0x310 [videodev]
> [ 3190.062357]  [<ffffffff81217d04>] ? terminate_walk+0x64/0xd0
> [ 3190.062360]  [<ffffffff810ba57c>] ? set_next_entity+0x9c/0xb0
> [ 3190.062366]  [<ffffffffc07f3d96>] video_usercopy+0x336/0x5b0 [videodev]
> [ 3190.062371]  [<ffffffffc07f4030>] ? video_ioctl2+0x20/0x20 [videodev]
> [ 3190.062377]  [<ffffffffc07f4025>] video_ioctl2+0x15/0x20 [videodev]
> [ 3190.062382]  [<ffffffffc07ef6a3>] v4l2_ioctl+0xd3/0xe0 [videodev]
> [ 3190.062384]  [<ffffffff8122124f>] do_vfs_ioctl+0x29f/0x490
> [ 3190.062387]  [<ffffffff8121cf84>] ? putname+0x54/0x60
> [ 3190.062389]  [<ffffffff8120c8df>] ? do_sys_open+0x1bf/0x2a0
> [ 3190.062391]  [<ffffffff812214b9>] SyS_ioctl+0x79/0x90
> [ 3190.062394]  [<ffffffff818318b2>] entry_SYSCALL_64_fastpath+0x16/0x71
> [ 3190.062396] ---[ end trace 0b4fc342399969d2 ]---
[... 302 lines cut ...]
> [ 3190.065032]  [<ffffffff81081212>] warn_slowpath_common+0x82/0xc0
> [ 3190.065035]  [<ffffffff810812ac>] warn_slowpath_fmt+0x5c/0x80
> [ 3190.065038]  [<ffffffffc083772d>] ? uvc_ioctl_enum_fmt+0x9d/0xc0 [uvcvideo]
> [ 3190.065044]  [<ffffffffc07f36f6>] v4l_enum_fmt+0xd86/0x10f0 [videodev]
> [ 3190.065051]  [<ffffffffc07f42c1>] __video_do_ioctl+0x291/0x310 [videodev]
> [ 3190.065053]  [<ffffffff81217d04>] ? terminate_walk+0x64/0xd0
> [ 3190.065056]  [<ffffffff810ba57c>] ? set_next_entity+0x9c/0xb0
> [ 3190.065061]  [<ffffffffc07f3d96>] video_usercopy+0x336/0x5b0 [videodev]
> [ 3190.065067]  [<ffffffffc07f4030>] ? video_ioctl2+0x20/0x20 [videodev]
> [ 3190.065073]  [<ffffffffc07f4025>] video_ioctl2+0x15/0x20 [videodev]
> [ 3190.065079]  [<ffffffffc07ef6a3>] v4l2_ioctl+0xd3/0xe0 [videodev]
> [ 3190.065081]  [<ffffffff8122124f>] do_vfs_ioctl+0x29f/0x490
> [ 3190.065084]  [<ffffffff8121cf84>] ? putname+0x54/0x60
> [ 3190.065086]  [<ffffffff8120c8df>] ? do_sys_open+0x1bf/0x2a0
> [ 3190.065089]  [<ffffffff812214b9>] SyS_ioctl+0x79/0x90
> [ 3190.065092]  [<ffffffff818318b2>] entry_SYSCALL_64_fastpath+0x16/0x71
> [ 3190.065093] ---[ end trace 0b4fc342399969dd ]---

Thanks!

Camera Model: F200 
Firmware version: 2.60.0.0
Operating System & Version: Ubuntu 16.04.1 LTS
Kernel Version (Linux Only): 4.4.0-45-generic
Build System: Makefile

vadali, 2016-11-06:
Could be related to #232

dorodnic, 2016-11-07:
What kernel version are you using?

vadali, 2016-11-07:
As written at the end of the first post  ;) 

Camera Model: F200
Firmware version: 2.60.0.0
Operating System & Version: Ubuntu 16.04.1 LTS
Kernel Version (Linux Only): 4.4.0-45-generic
Build System: Makefile

vadali, 2016-11-23:
nothing at all? this seems to be a recurring problem in linux (as seen in other issues here..)

ev-mp, 2016-11-23:
According to the log  the manually-inserted uvcmodule does have the patch applied.
Are you using "patch-uvcvideo-16.04.simple.sh" for Ubuntu 16 ?

Please rerun the installation chapter, especially the Linux v4l2 backend part and verify the progress.
The critical part to look into is:
...
make: Entering directory '/home/mydir/'
  CC [M]  /home/mydir/drivers/media/usb/uvc//uvc_driver.o
  CC [M]  /home/mydir/drivers/media/usb/uvc//uvc_queue.o
  CC [M]  /home/mydir/drivers/media/usb/uvc//uvc_v4l2.o
  CC [M]  /home/mydir/drivers/media/usb/uvc//uvc_video.o
  CC [M]  /home/mydir/drivers/media/usb/uvc//uvc_ctrl.o
  CC [M]  /home/mydir/drivers/media/usb/uvc//uvc_status.o
  CC [M]  /home/mydir/drivers/media/usb/uvc//uvc_isight.o
  CC [M]  /home/mydir/drivers/media/usb/uvc//uvc_debugfs.o
  CC [M]  /home/mydir/drivers/media/usb/uvc//uvc_entity.o
  LD [M]  /home/mydir/drivers/media/usb/uvc//uvcvideo.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/mydir/drivers/media/usb/uvc//uvcvideo.mod.o
  LD [M]  /home/mydir/drivers/media/usb/uvc//uvcvideo.ko
make: Leaving directory '/home/local_admin/git/v4l-dvb'
Unloading existing uvcvideo driver...
...

You may also need to check dmesg report to see that the module is inserted correctly after the patch is completed

vadali, 2016-11-23:
Thanks for getting back to me.

Yes, I ran "patch-uvcvideo-16.04.simple.sh", and actually, to make sure that it is being installed correctly, I added my own log lines that I later saw in dmesg. This way I can be sure that it is indeed installed and loaded.. 

Ill give it all another shot and report back..

vadali, 2016-11-23:
I redid everything, 

> make: Entering directory '/home/shlomi/realsense/librealsense/ubuntu-xenial'
>   CC [M]  /home/shlomi/realsense/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_driver.o
>   CC [M]  /home/shlomi/realsense/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_queue.o
>   CC [M]  /home/shlomi/realsense/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_v4l2.o
>   CC [M]  /home/shlomi/realsense/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_video.o
>   CC [M]  /home/shlomi/realsense/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_ctrl.o
>   CC [M]  /home/shlomi/realsense/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_status.o
>   CC [M]  /home/shlomi/realsense/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_isight.o
>   CC [M]  /home/shlomi/realsense/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_debugfs.o
>   CC [M]  /home/shlomi/realsense/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_entity.o
>   LD [M]  /home/shlomi/realsense/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.o
>   Building modules, stage 2.
>   MODPOST 1 modules
>   CC      /home/shlomi/realsense/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.mod.o
>   LD [M]  /home/shlomi/realsense/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko
> make: Leaving directory '/home/shlomi/realsense/librealsense/ubuntu-xenial'
> Unloading existing uvcvideo driver...

Checked the files are identical

> shell> md5sum /lib/modules/`uname -r`/kernel/drivers/media/usb/uvc/uvcvideo.ko
a27f9b29c1b9e6ae37b08e58f67cd481 
>  shell> md5sum /home/shlomi/realsense/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko
a27f9b29c1b9e6ae37b08e58f67cd481

Uninstalled and reinstalled:

> shell> sudo dmesg --clear 
> shell> sudo modprobe -r uvcvideo                                      
> shell> dmesg      
> [78560.893901] usbcore: deregistering interface driver uvcvideo
> shell> sudo modprobe  uvcvideo
> shell> dmesg
> [78841.868371] media: Linux media interface: v0.10
> [78841.873233] Linux video capture interface: v2.00
> [78841.878704] usbcore: registered new interface driver uvcvideo
> [78841.878706] USB Video Class driver (1.1.1)

Connected the device to a powered USB3 port:

> dmesg
> ...
> [78641.782094] ------------[ cut here ]------------
? [78641.782098] WARNING: CPU: 2 PID: 4659 at /build/linux-xHzv4a/linux-4.4.0/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd86/0x10f0 [videodev]()
> [78641.782100] Unknown pixelformat 0x494c4552

And..... got the same problem.

What else could I do?

ev-mp, 2016-11-28:
Only the following F200 pixel formats are required for F200 to operate correctly:
0x49564e49-... <=> INVI  - Infrared
0x495a4e49-... <=>INZI   - Depth + Ir
0x5a564e49-... <=> INVZ - Depth
These three formats are covered by Ubuntu 16 script which is  called patch-uvcvideo-16.04.simple.sh
The specific format that you mentioned, 0x494c4552, is not supported and therefore is not in the patch.

Check that the three above formats are resolved, i.e. not on dmesg's "black list".
In addition, run a positive verification with v4l-utils:

'$v4l2-ctl --list-formats-ext -d /dev/video0' and
'$v4l2-ctl --list-formats-ext -d /dev/video1'
Check field "Name":
 - for a patched format you'd see 
"Name        : INZI ... "
- for unresolved format you'd receive 
"Name        : 495a4e49-1a66-a242-9065-d01814a"

ev-mp, 2017-08-20:
Hi @vadali, I'm closing it now as it seem to be either resolved or abandoned.
Feel free to reopen it if you're still experiencing issues

closed 2017-08-20 by ev-mp