realsenseai/librealsense #344: How to support 16.04 kernel 4.7

hlc111, 2016-11-07:
| Required Info  | How to support 16.04 kernel 4.7   |
| Camera Model | R200| 
| Firmware Version | last up-to-date   | 
| Operating System & Version | 16.04   |
| Kernel Version (Linux Only)  | 4.7  |
| Build System |  Makefile | 
Hi, 

Currently i use Intel 6th Generation with 14.04, R200 and OpenCL, all works fine.
I try to use Intel 7th Generation  test the performance improve.
But it seems OpenCL driver require Linux 4.7 kernel , could you kindly let me know how to patch uvcvideo for kernel  4.7 ?

Thanks!

hlc111, 2016-11-08:
Because OpenCL driver need patch and build kernel 4.7.

I just add realsense patch like below seems works.
cd linux-4.7

# Apply UVC formats patch for RealSense devices
patch -p1 < ../scripts/realsense-camera-formats_ubuntu16.patch

Thanks!

dorodnic, 2016-11-08:
We are currently not working with kernel 4.7 yet, but this is valuable feedback, thank you

closed 2016-11-08 by dorodnic