realsenseai/librealsense #353: LR200 Pixel Deprojection

danem, 2016-11-09:
| Required Info  |   |
|---|---|
| Camera Model | LR200 | 
| Firmware Version |  2.0.71.18 | 
| Operating System & Version | Windows 10   |
| Build System |  VS2014 | 

I'm trying to map pixels from screen space to point space. Specifically, when a user clicks on the screen, I'd like to retrieve the corresponding point. When I do this, the program crashes [here](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense/rsutil.h#L35) because of the distortion type. I've noted in the documentation [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/projection.md#distortion-models) that deprojection from screen space isn't always possible, but does anyone know how I would work around this in my case? 

Reference Code:
```
float depth = depthImage[y * depthIntrinsics.width + x] * depthScale;
rs::float3 colorPt = colorIntrinsics.deproject(pixel, depth);
rs::float3 depthPt = _color_to_depth.transform(colorPt);
```

Thanks

Bardo91, 2017-03-30:
Hi @danem, I think I am facing the same problem. I managed to map from depth pixels to color pixels (to colorize the cloud). But now I would like to map from color pixels to depth pixels, so I can get the 3d coordinate os a color pixel. 
I read the documentation in "https://github.com/IntelRealSense/librealsense/blob/master/doc/projection.md" but doesn clarified me too much. 
Have you found a solution?

danem, 2017-03-30:
@Bardo91 yes, it's been a little while and I don't have to code in front of me, but what I found to work was some combination of rectified_color / color_aligned_to_depth + depth / depth_aligned_to_rectified_color streams instead of using just color + depth. Once I made that change the projection / deprojection  code worked as expected. 

Let me know if that works for you.

Bardo91, 2017-04-03:
@danem I am trying to understand the different streams of data as defined in 

```
 enum class stream : int32_t
    {
        depth                           ,  /**< Native stream of depth data produced by RealSense device */
        color                           ,  /**< Native stream of color data captured by RealSense device */
        .......
        rectified_color                 ,  /**< Synthetic stream containing undistorted color data with no extrinsic rotation from the depth stream */
        color_aligned_to_depth          ,  /**< Synthetic stream containing color data but sharing intrinsic of depth stream */
       .....
        depth_aligned_to_color          ,  /**< Synthetic stream containing depth data but sharing intrinsic of color stream */
        depth_aligned_to_rectified_color,  /**< Synthetic stream containing depth data but sharing intrinsic of rectified color stream */
        ....
    };
```

There is an extrinsic calibration between the cameras that is the extrinsic rotation and tranlation, that can be obtained using `mRsDevice->get_extrinsics(rs::stream::color, rs::stream::XXXXX);`. Also, for each stream it can be obtained the intrinsic calibration using the similar method: `mRsDevice->get_stream_intrinsics(rs::stream::XXXX);`. The intrinsic calibration is the class that has the methods for projecting and deprojecting, but these methods needs the value of the depths.

I am not sure if the aligned stream means that pixels in one image correspond directly to pixels in the other images (for example color_aligned_to_depth pixels are aligned with pixels in depth image) or not.

I end did some math  for getting this correspondence between the depth pixels and the color pixels (correct me if I am wrong): 

[image]

But this is happening to me:
[image]

The feature points are shifted in the cloud toward the right side. I don't mind to do it using the aligned streams as you said but I would like to understand them better.

Bardo91, 2017-04-04:
After playing around I found what you said. Here is the result:

[image]

I only enabled the color and depth stream (as it it not posible to use the other due to being not native streams)

mRsDevice->enable_stream(rs::stream::depth, 640, 480, rs::format::z16, 60);
mRsDevice->enable_stream(rs::stream::color, 640, 480, rs::format::rgb8, 60);

I get the intrinsics of the depth.

mRsDepthIntrinsic = new rs::intrinsics();
auto tempDepthIntrinsic = mRsDevice->get_stream_intrinsics(rs::stream::depth);
memcpy(mRsDepthIntrinsic, &tempDepthIntrinsic, sizeof(rs::intrinsics));

Then for each frame I get both depth and color

cv::cvtColor(cv::Mat(mRsColorIntrinsic->height, mRsColorIntrinsic->width, CV_8UC3, (uchar*)mRsDevice->get_frame_data(rs::stream::color_aligned_to_depth)), mLastRGB, CV_RGB2BGR);		
mLastDepth = cv::Mat(mRsDepthIntrinsic->height, mRsDepthIntrinsic->width, CV_16U, (uchar*) mRsDevice->get_frame_data(rs::stream::depth));

and finally pixels in color and depth are completely aligned so for each (X,Y) in color image you can directly take the (X,Y) on the depth and deproject using the intrinsic matrix of the depth.

Maybe this issue can be closed.

p0wdrdotcom, 2017-09-22:
Actually I think that this is quite simple.

1. Select the pixel(s) you are interested in from the raw colour stream (rs::stream::color)
2. Those pixels {x,y} will map directly to the synthetic depth stream rs::stream::depth_aligned_to_color
3. from there you can gather the depth and get a 3d point of the pixel from the depth intrinsics

consider the following code... where x1 and y1 are pixel coordinates from the color stream

```depth_intrinsics = dev->get_stream_intrinsics(rs::stream::depth);
const uint16_t * depth_image = (const uint16_t *)dev->get_frame_data(rs::stream::depth_aligned_to_color);
float scale = dev->get_depth_scale();
uint16_t p1_depth_value = depth_image[y1 * depth_intrinsics.width + x1];
float p1_depth_in_meters = p1_depth_value * scale;
rs::float2 p1_depth_pixel = {(float)x1, (float)y1};
rs::float3 p1_depth_point = depth_intrinsics.deproject(p1_depth_pixel, p1_depth_in_meters);
```
I believe this to give you a way to get a 3d point (p1_depth_point above) of a pixel coordinate from the color stream.
If you use the color stream to segment an object. You can get its 3D points via this means.

I haven't tested this extensively but it appears to work where I'm measuring the size of a test object.

closed 2017-09-25 by oordan