realsenseai/librealsense #354: depth_aligned_to_color crashes when debugging

danem, 2016-11-11:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 2.0.71.18  | 
| Operating System & Version | Windows 10  |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2014 |

When I try to get frame data using the `depth_aligned_to_color` stream while debugging, the program crashes [here](https://github.com/IntelRealSense/librealsense/blob/0d75f5c893ff0ce7862dea77d0d442c82960a789/src/image.cpp#L558) with the message `Debug Assertion Failed! Expression: Invalid Comparator` . This affects cpp-alignimages example as well when built with debugging enabled.

chexenia, 2017-04-27:
Any progress on fixing it?

danem, 2017-04-27:
I found that changing the lambda from lines 556 to 561 fixed my issue:

    [out_z, z_pixels](int z_pixel_index, int other_pixel_index) {
        const unsigned short a = out_z[other_pixel_index];
        const unsigned short b = z_pixels[z_pixel_index];
        out_z[other_pixel_index] = a ? std::min(a,b) : b;
    })

Not sure why std::min couldn't generate correct implementation, but this seemed to resolve my problems.

souch55, 2017-06-19:
There is type miss match between "other_pixel_index" and "z_pixel_index", "other_pixel_index" should be of type const uint16_t * ,unlike "z_pixel_index" ,for std::min() to compare the objects.I hope following correction in the source code will fix this bug.

closed 2017-10-26 by jlgarcia75