realsenseai/librealsense #362: librealsense detects F200 but not SR300

andrecarlucci, 2016-11-18:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 16.04.1 LTS  |
| Kernel Version (Linux Only)  |  4.4.0-47-generic |
| Build System | Makefile | 

Hi guys,

I installed everything following the tutorial and I could run the examples just fine using my F200 camera.
When I switch to the SR300, I get a "No device detected. Is it plugged in?"

The SR300 works on librealsense for Windows (Windows 10 build 14965) and also with the full SDK.

What kind of extra information can I provide you to assist me in this issue?

Thanks a lot,

Andre

dorodnic, 2016-11-18:
Hi, as the first step could you please check or share the output of lsusb and dmesg after you connect the camera?

andrecarlucci, 2016-11-19:
Looking at the logs I can see that it's not a librealsense problem, Ubuntu is not recognizing the SR300, only the F200. Windows recognizes both, and Virtual Box too.

Any ideas why?

# F200

[image]

---
lsusb 
```
carlucci@ubuntuvm:~$ lsusb
Bus 002 Device 006: ID 8086:0a66 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 015: ID 041e:4099 Creative Technology, Ltd 
Bus 001 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

dmesg | grep usb 
```
carlucci@ubuntuvm:~$ dmesg | grep usb
[    0.358816] usbcore: registered new interface driver usbfs
[    0.358834] usbcore: registered new interface driver hub
[    0.358852] usbcore: registered new device driver usb
[    1.157776] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    1.157778] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.157779] usb usb1: Product: xHCI Host Controller
[    1.157781] usb usb1: Manufacturer: Linux 4.4.0-47-generic xhci-hcd
[    1.157782] usb usb1: SerialNumber: 0000:00:0c.0
[    1.158779] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003
[    1.158781] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.158782] usb usb2: Product: xHCI Host Controller
[    1.158783] usb usb2: Manufacturer: Linux 4.4.0-47-generic xhci-hcd
[    1.158784] usb usb2: SerialNumber: 0000:00:0c.0
[    1.467310] usb 1-1: new full-speed USB device number 2 using xhci_hcd
[    1.624757] usb 1-1: New USB device found, idVendor=80ee, idProduct=0021
[    1.624760] usb 1-1: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[    1.624762] usb 1-1: Product: USB Tablet
[    1.624763] usb 1-1: Manufacturer: VirtualBox
[    1.624873] usb 1-1: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
[    1.950569] usb 1-2: new full-speed USB device number 3 using xhci_hcd
[    2.349557] usb 1-2: New USB device found, idVendor=041e, idProduct=4099
[    2.349572] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.349576] usb 1-2: Product: Creative VF0800
[    2.349581] usb 1-2: Manufacturer: Creative Technology Ltd
[    2.349585] usb 1-2: SerialNumber: 000000030043
[    2.470540] usb 1-3: new high-speed USB device number 4 using xhci_hcd
[    2.694946] usb 1-3: device descriptor read/64, error 18
[    3.022993] usb 1-3: device descriptor read/64, error 18
[    3.346987] usb 1-3: new high-speed USB device number 5 using xhci_hcd
[    3.571606] usb 1-3: device descriptor read/64, error 18
[    3.898961] usb 1-3: device descriptor read/64, error 18
[    4.222797] usb 1-3: new high-speed USB device number 6 using xhci_hcd
[    4.239529] usb 1-3: Invalid ep0 maxpacket: 9
[    4.462520] usb 1-3: new high-speed USB device number 7 using xhci_hcd
[    4.479119] usb 1-3: Invalid ep0 maxpacket: 9
[    4.479324] usb usb1-port3: unable to enumerate USB device
[    4.572054] usbcore: registered new interface driver usbhid
[    4.572056] usbhid: USB HID core driver
[    4.574372] input: VirtualBox USB Tablet as /devices/pci0000:00/0000:00:0c.0/
usb1/1-1/1-1:1.0/0003:80EE:0021.0001/input/input6
[    4.574602] hid-generic 0003:80EE:0021.0001: input,hidraw0: USB HID v1.10 Mou
se [VirtualBox USB Tablet] on usb-0000:00:0c.0-1/input0
[    4.577796] input: Creative Technology Ltd Creative VF0800 as /devices/pci000
0:00/0000:00:0c.0/usb1/1-2/1-2:1.3/0003:041E:4099.0002/input/input7
[    4.631959] hid-generic 0003:041E:4099.0002: input,hiddev0,hidraw1: USB HID v
1.11 Device [Creative Technology Ltd Creative VF0800] on usb-0000:00:0c.0-2/input3
[   13.498165] usbcore: registered new interface driver snd-usb-audio
[  141.051609] usbcore: registered new interface driver uvcvideo
[... 143 lines cut ...]
[    1.103328] usb usb2: Manufacturer: Linux 4.4.0-47-generic xhci-hcd
[    1.103329] usb usb2: SerialNumber: 0000:00:0c.0
[    1.411215] usb 1-1: new full-speed USB device number 2 using xhci_hcd
[    1.557790] usb 1-1: New USB device found, idVendor=80ee, idProduct=0021
[    1.557793] usb 1-1: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[    1.557795] usb 1-1: Product: USB Tablet
[    1.557796] usb 1-1: Manufacturer: VirtualBox
[    1.557902] usb 1-1: ep 0x81 - rounding interval to 64 microframes, ep desc says 80 microframes
[    1.891061] usb 1-2: new high-speed USB device number 3 using xhci_hcd
[    2.115154] usb 1-2: device descriptor read/64, error 18
[    2.439407] usb 1-2: device descriptor read/64, error 18
[    2.763796] usb 1-2: new high-speed USB device number 4 using xhci_hcd
[    2.984054] usb 1-2: device descriptor read/64, error 18
[    3.312348] usb 1-2: device descriptor read/64, error 18
[    3.635382] usb 1-2: new high-speed USB device number 5 using xhci_hcd
[    3.652078] usb 1-2: Invalid ep0 maxpacket: 9
[    3.871362] usb 1-2: new high-speed USB device number 6 using xhci_hcd
[    3.887694] usb 1-2: Invalid ep0 maxpacket: 9
[    3.888016] usb usb1-port2: unable to enumerate USB device
[    3.896174] usbcore: registered new interface driver usbhid
[    3.896176] usbhid: USB HID core driver
[    3.897634] input: VirtualBox USB Tablet as /devices/pci0000:00/0000:00:0c.0/usb1/1-1/1-1:1.0/0003:80EE:0021.0001/input/input6
[    3.897990] hid-generic 0003:80EE:0021.0001: input,hidraw0: USB HID v1.10 Mouse [VirtualBox USB Tablet] on usb-0000:00:0c.0-1/input0
[  257.686162] usbcore: registered new interface driver uvcvideo
```

dorodnic, 2016-11-19:
There are some known issues when working with realsense inside a VM. Unfortunately for now some of these seem impossible to resolve, but we do work in this direction.
For now, please retry with a physical box, you can try with Ubuntu live-stick if you don't want full Linux installation.

closed 2016-11-19 by dorodnic