realsenseai/librealsense #365: Using Python ctypes to access camera

drasmuss, 2016-11-18:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |  2.1.27.2853 | 
| Operating System & Version | Windows 10  |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2015 | 

I am trying to use `ctypes` to access the `realsense` DLL (compiled in VS2015).  Here's a minimal script showing what I'm doing:
``` python
    import ctypes

    RS_API_MAJOR_VERSION = 1
    RS_API_MINOR_VERSION = 11
    RS_API_PATCH_VERSION = 0
    RS_API_VERSION = (RS_API_MAJOR_VERSION * 10000 +
                      RS_API_MINOR_VERSION * 100 +
                      RS_API_PATCH_VERSION)

    libc = ctypes.CDLL("realsense-d.dll")
    libc.rs_create_context(self.RS_API_VERSION)
    ...
```

This approach used to work (as of at least Sept 23 2016), and I was able to stream data from the camera.  But there have been some firmware and API updates since then, and now the `libc.rs_create_context(self.RS_API_VERSION)` line fails with `exception: access violation reading 0xFFFFFFFFFFFFFFFF`.

Any ideas what might have changed that is causing this to fail?  Or just general suggestions for how to get this working again?

drasmuss, 2016-11-29:
Can confirm that #378 fixes the error for me

closed 2016-11-29 by dorodnic

toinsson, 2016-12-13:
I have made a python bindings that is based on ctypes, but lack windows users, maybe you would like to have a look @drasmuss ? 
https://github.com/toinsson/pyrealsense

PR #378 (2016-11-29): Stop relying on the validity of error object passed to create_context
This should resolve #365