realsenseai/librealsense #366: Need clarification for timestamp_handler usage

mkhansen-intel, 2016-11-18:
Can someone explain what the purpose of the timestamp_handler callback is for the enable_motion_tracking API? Obviously it is requiring a callback function pointer, but what is the purpose of that callback?

In the rs.h file it says this:

```
/**
* enable and configure motion-tracking data handlers
* \param[in] on_motion_event    user-defined routine to be invoked when a motion data arrives
* \param[in] motion_handler     a user data point to be passed to the motion event callback
* \param[in] on_timestamp_event user-defined routine to be invoked on timestamp
* \param[in] timestamp_handler  a user data point to be passed to the motion event callback
* \param[out] error             if non-null, receives any error that occurs during this call, otherwise, errors are ignored
*/
void rs_enable_motion_tracking(rs_device * device,
    rs_motion_callback_ptr on_motion_event, void * motion_handler,
    rs_timestamp_callback_ptr on_timestamp_event, void * timestamp_handler,
rs_error ** error);
```

dorodnic, 2016-11-19:
Yes, `rs_enable_motion_tracking` is responsible for activating the motion module on-board the device. One of the services it provides is producing shared and high-resolution timestamps for all component hooked-up to it. Usually, librealsense takes care of that and just drops the timestamps on the relevant frames. 
However, let's say you have an external device (like a compass) and you wish to synchronize it precisely with image and motion. You could connect your device to a one of two 3.5v GPIOs (exposed on the side of the camera case). Every time you signal it, you get a timestamp callback with a frame number, source ID and a timestamp. 
This would allow you to synchronize your compass events (presumably coming to you though I2C or some other method) with realsense data. 

Personally, I have used this feature for testing - hook up an Arduino to an LED and a ZR300 GPIO. Now every time you toggle the LED you know exactly when it actually showed up on stream and when it was supposed to show up (more or less up to +-2ms)

closed 2016-11-19 by dorodnic

mkhansen-intel, 2016-11-21:
So if we are not using a external device that we want to sync to the frame, can we pass a null pointer as the callback? Also, can you please edit the comment in the header file to make this more clear? Thanks.

mkhansen-intel, 2016-11-22:
@dorodnic can you answer my question? If we aren't using an external device, can we pass a null ptr?

dorodnic, 2016-11-23:
There is an overload of `enable_motion_tracking` accepting just motion-callback. 
Will do our best to update documentation on that function.