realsenseai/librealsense #371: Video streaming not working on ubilinux

michael5486, 2016-11-21:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.10  | 
| Operating System & Version | ubilinux 3.0  |
| Kernel Version (Linux Only)  | 4.4.0-ubi3-amd64  |
| Build System |  Makefile | 

Been trying to get a stable setup for this RealSense camera. Started on Ubuntu, but it kept causing the computer to freeze when plugging the camera in. Migrated to Ubilinux and so far have had good luck. However, I've been encountering some issues with displaying the streams...specifically something with the LLVM

## Actual Issue

```
Using device 0, an Intel RealSense R200
Serial number: 2381009652
Firmware version: 1.0.72.10
LLVM ERROR: Cannot select: intrinsic %llvm.x86.sse41.pblendvb
```

## What does work

So far cpp-enumerate-devices, cpp-tutorial-depth and cpp-headless work, so I know the camera works and can communicate with Ubilinux successfully. Has anyone encountered an issue like this before? I'm thinking it's an issue with missing graphics drivers/dependencies but have not figured it out yet. I am currently trying to see if I can create my own streams using opencv, but getting these examples to work would be incredibly helpful.

michael5486, 2016-11-22:
In an attempt to diagnose the issue, I began commenting out every line in cpp-capture.cpp. Commenting out this code gets rid of the error, but breaks the display function since the frames are no longer updated.

```
//buffers[0].show(dev, align_color_to_depth ? rs::stream::color_aligned_to_depth : (color_rectification_enabled ? rs::stream::rectified_color : rs::stream::color), 0, 0, tile_w, tile_h);
//buffers[1].show(dev, align_depth_to_color ? (color_rectification_enabled ? rs::stream::depth_aligned_to_rectified_color : rs::stream::depth_aligned_to_color) : rs::stream::depth, w / cols, 0, tile_w, tile_h);
//buffers[2].show(dev, rs::stream::infrared, 0, h / rows, tile_w, tile_h);
//buffers[3].show(dev, rs::stream::infrared2, w / cols, h / rows, tile_w, tile_h);
```

Still working on a fix.

jlgarcia75, 2017-10-07:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

Best regards,
Jesus
Intel Customer Support

closed 2017-10-16 by jlgarcia75