realsenseai/librealsense #379: Error on startup with R200: no matching media type for  pixel format 5a313620

joelpryde, 2016-12-01:
I am getting an error whenever I start any of the examples and try to get depth data from my R200 with librealsense (color data works fine).  The error seems to occur because it cannot match the selected mode's width and height (or other settings) with the modes provided by the device.

This is my selected mode:
[0] = {mode={subdevice=1 native_dims={x=628 y=469 } pf={fourcc=1513174560 plane_count=1 bytes_per_pixel=2 ...} ...} ...}

That resolution doesn't match with any of the R200's valid resolutions:
https://github.com/IntelRealSense/librealsense/blob/master/doc/supported_video_formats.pdf

Note: the depth example projects from the RealSense SDK run fine on this machine.

| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.10 | 
| Operating System & Version |  Windows 8 |
| Build System |  VS2013 | 

Thanks,
Joel

ddiakopoulos, 2016-12-01:
Sometimes this is indicative of Windows Media Foundation issues. The native dims are consistent with what you'd actually get if you asked for 640x480 (due to cropping/rectification/some pixels used for per-frame data xfer). Can I ask a silly question and make sure you're not using _actual_ Windows 8? (definitely not supported, Windows 8.1 added in tons of WMF fixes).

elmewo, 2017-07-24:
I've got the same issue with Win 10. 
I get the error for any of the presets as well as for manually setting the format to
`rsCamera->enable_stream(rs::stream::depth, 640, 480, rs::format::z16, 30, rs::output_buffer_format::continous );
    rsCamera->enable_stream( rs::stream::color, 640, 480, rs::format::yuyv, 30, rs::output_buffer_format::continous );`

Any hints to solve the problem?

VanessaLange, 2017-08-17:
I also have the same issue with Windows 10, however not with the R200 (works fine) but with the SR300. I noticed that it's commit ff81ad9 (Merge pull request #535 from ev-mp/fourcc_fix) that breaks things for me. Commit f4834db (Merge pull request #526 from ffgiff/fix-for-latest-f25) ist the last commit on the master branch that does not crash with the error message from the title.

ev-mp, 2017-08-20:
Hi @VanessaLange , thanks for pointing this out.
The fix for unrecognized formats for SR300 on Windows [has been merged](https://github.com/IntelRealSense/librealsense/pull/571) into the master tree.

closed 2017-08-20 by ev-mp

VanessaLange, 2017-08-21:
Thank you very much. The SR300 works fine now. However, the fix broke the R200 for me:
no matching media type for  pixel format 494e565a