realsenseai/librealsense #383: Distortion in depth data?

floe, 2016-12-08:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.10.10.0  | 
| Operating System & Version |  Ubuntu 16.04  |
| Kernel Version (Linux Only)  |  4.4.0-53  |
| Build System |  Makefile | 

I'm running the SR300 with Ubuntu, and I have a question the depth data itself. In my setup, the camera is looking down at a flat desk surface from a distance of ~ 85 cm, but when I run `cpp-pointcloud`, the result looks something like this:

[image]

The curve is pretty obvious (and in reality, the desk is of course flat). Is this an inherent limitation of the depth imager, or is there anything I can do? In my application, I'm using `rs::stream::depth` and `rs::stream::color`, then I manually map from depth to color using the following code:

```
      rs::float2 pixel = { (float) x, (float) y };
      point = depth_intrinsics.deproject(pixel, raw_depth*scale);

      rs::float3 point2 = depth_to_color.transform(point);
      rs::float2 pixel2 = color_intrinsics.project(point2);

      const int color_x0 = static_cast<int>(pixel2.x + 0.5f);
      const int color_y0 = static_cast<int>(pixel2.y + 0.5f);
```

Any suggestions?

ddiakopoulos, 2017-03-21:
Hi @floe This means you received a prerelease SR300 with fundamental calibration issues. (See: https://github.com/IntelRealSense/librealsense/issues/46)

Intel would need to provide you with a new camera.

closed 2017-03-21 by ddiakopoulos

floe, 2017-03-21:
@ddiakopoulos thanks, very helpful! Who do I need to contact about this?