realsenseai/librealsense #384: Current firmware versions?

floe, 2016-12-08:
I have not been able to find any information (here or on the Intel website) what the _current_ firmware versions for the cameras are. Maybe add this to the README?

mdhorn, 2016-12-08:
It is published towards the top of the readme.
https://github.com/IntelRealSense/librealsense#compatible-devices

They are generally good about updating this when a newer firmware is required. Also new firmware in usually only released when required, so they are generally in-sync.

floe, 2016-12-10:
Hm. The readme states that required FW for the SR300 is 3.10.10.0, but my SR300 is already on 3.21.0.0, and this is from two months ago... so probably not quite in sync.

mgserafi, 2016-12-23:
The firmware list doesn't seem to be listed anymore. Is there another place to find the current versions?

dorodnic, 2017-01-03:
The list under [compatible-devices](https://github.com/IntelRealSense/librealsense#compatible-devices) was suggesting minimal required firmware versions. Since librealsense is not actually enforcing these limitations the list was removed for now. 
Until stated otherwise, librealsense remains an experimental and community project, meaning we can't physically test all possible combinations of device, firmware, OS, kernel version, USB controller etc.. 
We will make effort to enable any official Intel firmware, but each RealSense product is being distributed with a particular support matrix in mind. For example, while there is nothing stopping you from working with SR300 3.10.10 using librealsense, the only configuration that is officially supported by Intel for the SR300 is RealSense SDK on Windows 10 with the latest firmware as published through Windows Update.

closed 2017-01-03 by dorodnic