realsenseai/librealsense #385: rs.error: Couldn't query adapter board / motion module FW version!

mkhansen-intel, 2016-12-08:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version |  2.0.71.26 | 
| Operating System & Version |  Ubuntu 16.04 |
| Kernel Version (Linux Only)  |  4.4.0-53-generic |
| Build System |  Makefile | 

mkhansen@mkhansen-skull:~/catkin_ws/src/realsense_workspace/librealsense$ bin/cpp-motion-module
rs.error: Failed to get firmware version
rs.error: Couldn't query adapter board / motion module FW version!
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense ZR300
    Serial number: 3181801844
    Firmware version: 2.0.71.26

I've seen this frequently, sometimes re-plugging the camera resolves it, however, that's not a viable option for remote usage. 

I've tried rebooting and it persists across reboots.

Is this a known issue? Is there a way to resolve it without re-plugging the camera each time?

mkhansen-intel, 2016-12-08:
Note - I also see this issue when running cpp-enumerate. This was the simplest way I could find to reproduce it.

dorodnic, 2016-12-14:
When you re-plug the camera, what is the output from cpp-enumerate-devices (the topmost block)?
Might be a case of older firmware versions.
Does this reproduce consistently and how do you manage to reproduce this?

mkhansen-intel, 2016-12-16:
I discovered the cause of this was actually a boost::mutex problem, in my code. For some reason sometimes I was getting a boost::mutex error on shutdown, and after that, I would see this rs.error. I don't know why the mutex error caused this, but I was able to fix it by replacing the boost::mutex with a std::mutex. I'm closing this issue.

closed 2016-12-16 by mkhansen-intel