realsenseai/librealsense #386: Crippled image when using libuvc

oliverko, 2016-12-09:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.15.0.0  | 
| Operating System & Version | Linux SuSe 64-bit |
| Kernel Version (Linux Only)  |  4.4.27-2-default |
| Build System | Makefile | 

Hi everyone, 

I am experiencing image problem while running librealsense. I am not allowed to patch the kernel, so
I compiled the library with -DRS_USE_LIBUVC_BACKEND and removed uvcvideo module. 

I tested three SR300 cameras, all are working. However, in the case of the aforementioned configuration, I get a crippled image (both rgb and depth). Has anyone else experienced the same?

dmesg output is flooded with:
[251319.114124] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 29 ep 6 with no TDs queued?
[251319.136830] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 29 ep 4 with no TDs queued?

The resulting image is as such:
![sr300_error](https://cloud.githubusercontent.com/ass
[image]
ets/8171584/21050806/2d253870-be1d-11e6-8f01-a158bb2bc6d2.png)

ddiakopoulos, 2017-03-21:
Hi @oliverko See my comment here about changing the number of buffers in libuvc: https://github.com/IntelRealSense/librealsense/issues/443#issuecomment-286995090

seath49, 2017-03-24:
@ddiakopoulos
Thanks a lot for the suggestion, I'm on vacation so I'm not able to test it. I'll get to it in two weeks.

Thanks,
Oli

oliverko, 2017-04-18:
@ddiakopoulos
Thanks for your suggestion. I tried all range of buffer sizes without success. 
If there are no segfaults, the RGB image is filckering as mentioned in other issues. 
If there is a segfault it usually occurs at 
rsimpl::unpack_yuy2<(rs_format)5>
or  
src/libuvc/stream.c:482 memcpy(strmh->outbuf + strmh->got_bytes, payload + header_len, data_len); 

My guess is that the user space communication is not timely enough, which leads
to corruption of data buffer and subsequent "random" faults. Unfortunately, I don't 
have time to investigate this and be more helpful. 

Best, 
Oli

jlgarcia75, 2017-10-17:
Hello @oliverko ,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

seath49, 2017-10-18:
Hi @jlgarcia75, 

No, thanks for asking. We managed to switch to a patched 
kernel after all. 

Best,
Seath

closed 2017-10-26 by jlgarcia75