realsenseai/librealsense #388: Unplugging the camera crashes the application, cannot catch exception

lvretenar, 2016-12-09:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06  | 
| Operating System & Version | Ubuntu 16.04.1 LTS  |
| Kernel Version (Linux Only)  | 4.4.0-53-generic  |
| Build System |  Makefile | 

Hello,

I'm planning on using the R200 camera in a system where the camera can be disconnected from the system while the application is running. When I unplug the camera from the usb, an exception is thrown in one of the capture threads:

    terminate called after throwing an instance of 'std::runtime_error'
      what():  VIDIOC_DQBUF error 19, No such device
    Aborted (core dumped)

This exception terminates my application, since it is unhandled in a separate thread. I would like to catch that exception and recreate the context to enumerate the connected cameras again.

Is such functionality on the roadmap for librealsense?

dorodnic, 2017-01-03:
Yes, this should be addressed eventually. Exceptions from back-end thread are usually logged and should not escape the library, so it is a bug. However, responding to camera disconnect events is relatively low in priority in project road-map, so if you'd like to look into it and propose a fix it would be appreciated.

souch55, 2017-06-21:
Hi lvretenar there are errors in uvc-v4l2.cpp source,in the poll method we should handle the error ,this error when VIDIOC_DQBUF returns EAGAIN error with no buffer available ,we should check for device connectivity and after freeing buffers if device not found, should call for stop() and destructor for subdevice and re-enumerate for connected devices.

jlgarcia75, 2017-10-07:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

maciekd, 2017-10-07:
Hi,

Yes, we would love this bug to be fixed! It is a real problem for us in our system.

jlgarcia75, 2017-10-16:
@maciekd, 

Development on the Master branch of librealsense has stopped and the R200 camera has been discontinued for some time now. All focus is now on librealsense 2.0 (a.k.a. SDK 2.0, development branch of librealsense) which will support the D400 series of cameras and the SR300.  

Have you tried using the method recommended by souch55 above?

Regards,
Jesus
Intel Customer Support