realsenseai/librealsense #394: Unable to access synthetic data streams and intrinsic/extrinsic data in Ubuntu

mgserafi, 2016-12-23:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |  1.0.72.10  | 
| Operating System & Version |  Ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.4  |
| Build System |  Makefile | 

I'm currently using `ctypes` to use the librealsense library in python. I can access the native RGB and depth streams but get segmentation faults when trying to access the synthetic streams (such as `RS_STREAM_DEPTH_ALIGNED_TO_COLOR`). So I tried to perform the image alignment myself by rewriting the image alignment code in `image.cpp` in python. However, I also receive segmentation faults when trying to access the intrinsic and extrinsic data from the sensors.  

The alignment code I've written does work with the same camera in Windows 10. (I'm also able to access the synthetic streams in windows). So I think that it's an issue with by ubuntu setup.

I think this might be related; I get the [bad magic number issue](https://github.com/IntelRealSense/librealsense/issues/49) when running examples such as `cpp-capture` and `cpp-multicam`. Both of these will show the RGB images, but the IR and depth images do not come up (examples work fine in windows). I'm concerned it might be a hardware issue. The usb controller i'm using is: Intel Corporation Sunrise Point-H USB 3.0 xHCI Controller (rev 31)
Is this one of the known incompatible hardware?

dorodnic, 2016-12-24:
Hi @mgserafi 
This might be a kernel-patch issue. Ubuntu 14.04 still requires manual patching (described the in [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md).
To verify this theory, after connecting the camera, check the output of dmesg, unrecognized pixel formats are reported there when the kernel was not properly patched. You can either follow the instructions in [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) or switch to Ubuntu 16.04 (includes R200 support out of the box).

mgserafi, 2017-01-03:
I am getting unrecognized pixel format errors. I'm also getting:
```failed to execute '/usr/local/bin/usb-R200-in_udev' '/usr/local/bin/usb-R200-in_udev': No such file or directory```
When is this directory made? I followed the installation guide the first time and in step 3, I followed the instructions for Ubuntu 14.04 with the 4.4 Stable Kernel.\

mgserafi, 2017-01-03:
I see the `usb-R200-in_udev` file in the `librealsense/config` folder.

mgserafi, 2017-01-03:
So I was able to get the files in the right folder by using the fix mentioned for R200 with 3.19.xx Kernel.
However, I still get unkonwn video format errors:
```
[ 1851.172324] uvcvideo: Unknown video format 49323159-0000-0010-8000-00aa00389b71
[ 1851.172333] uvcvideo: Unknown video format 20493859-0000-0010-8000-00aa00389b71
[ 1851.172344] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1851.176690] uvcvideo: Unknown video format 2036315a-0000-0010-8000-00aa00389b71
[ 1851.176692] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1851.178121] uvcvideo: Unable to create debugfs 2-4 directory.
[ 1851.178205] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:14.0/usb2/2-9/2-9.2/2-9.2:1.2/input/input21
[ 1851.180517] uvcvideo: Unknown video format 30315752-0000-0010-8000-00aa00389b71
[ 1851.180519] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1851.181989] uvcvideo: Unable to create debugfs 2-4 directory.
```

mgserafi, 2017-01-03:
I just tried re-patching, as suggested in the installation guide. After running:
 
```./scripts/patch-uvcvideo-4.4.sh v4.4-wily```

I get the following

```
cp: will not overwrite just-created ‘./.config’ with ‘/usr/src/linux-headers-4.4.0-31-generic/.config’
cp: will not overwrite just-created ‘./.config’ with ‘/usr/src/linux-headers-4.4.0-53-generic/.config’
cp: will not overwrite just-created ‘./.config’ with ‘/usr/src/linux-headers-4.4.0-57-generic/.config’
```

Is there something else other than `linux-v4.4-wily` I should remove before trying to re-patch?

mgserafi, 2017-01-03:
Okay, just saw [this issue](https://github.com/intel-ros/realsense/issues/12#issuecomment-198079546) with the same error.

mgserafi, 2017-01-03:
Even after fixing the problems with the patching, I still run into errors when trying to access intrinsic/extrinsic data. I still get a segmentation fault.

jlgarcia75, 2017-10-03:
Hello,

If the issue is software related, we are no longer updating this software as we have moved to our latest generation, the D400 series. If you are in your warranty, a replacement can be provided.

Jesus G.
Intel Customer Support

closed 2017-10-03 by jlgarcia75