realsenseai/librealsense #398: Merging two SR300 to increase the field of view

dacula, 2016-12-31:
| Camera Models | SR300 | 
| Firmware Version |  3.3.27.5718 | 
| Operating System & Version | Windows 10  |
| Build System |  VS2015 | 

Hello,
 
I have two Intel RealSense SR300 Cameras which i want to merge into one Program.
My idea was to increase the field of view by positioning the cameras in a degree of about 90°.
Can you please suggest an approach on how this could be accomplished?
My first step would be trying to get them into a single window and align them so that they overlap by a view pixels and merge them together but i'm not sure how to get this done. Then i would alter the image streams to transform the pixels.
Is this possible ?

Many thanks in advance,
dacula

dbolkensteyn, 2017-01-23:
Using two SR300 cameras to get an extended field of view does not seem to be possible.

> This pattern causes destructive interference with other F200 and SR300 cameras and performance is dramatically degraded when multiple cameras are pointed at the same subject.

Reference: https://github.com/IntelRealSense/librealsense/blob/master/doc/camera_specs.md

However it seems that you might be able to do what you want using the R200.

drhalftone, 2017-02-21:
Assuming he aims the cameras at a 90 degree angle, shouldn't be a problem.  The SR's use a structured light pattern of binary bar codes.  So overlapping fields of view obviously don't work, but that's not to say you can't still align the cameras with a gap between.  All you need to do is generate a video of the two cameras sitting on a tripod, and then just pan the tripod around 360 degrees.  Use the objects in the room as fiducials from which to derive a 4x4 transform matrix from one camera to the other.  And then embed this transform into your rendering code.

closed 2017-03-21 by ddiakopoulos