realsenseai/librealsense #399: Multiple Realsense Cameras no longer working.

hinmanj, 2017-01-01:
| Required Info  |   |
|---|---|
| Camera Model | R200 (and S300) | 
| Firmware Version | 1.0.72.10  | 
| Operating System & Version | Windows 10 v1607  |
| Build System |  VS2015 | 

I've been working with multiple R200's for a while now, and recently (the past week or so) was testing with only 1 connected to my machine though 2 or 3 or more had been working as long as I remember. However, recently switching back to 2+, Librealsense can only find one at a time. My Windows 10 devices shows all the cameras as their own entities, but Librealsense will only find 1 when enumerating no matter what. If I plug 2 in, and then unplug the first one I had plugged in, it will find none.

I'm assuming this must have been some weird Windows 10 update that happened out of the blue (though I don't recall it ever auto-updating in the past week).

Trying to plug in an SR300 with an R200 results in the same problem.

dorodnic, 2017-01-02:
Hi,
I have never encountered this before, but we are also limited in validation resources and might have missed it. I'll try to reproduce and update. In the meantime you can take a look inside wmf backend around the call to MFEnumerate, this might help pin-point the problem.

hinmanj, 2017-01-03:
Hm, this was on my end. Today *no* cameras were being recognized no matter how many were plugged in. Turns out I needed to do a full clean and rebuild of librealsense for the cameras to start being recognized again. Not sure why though.

closed 2017-01-03 by hinmanj