realsenseai/librealsense #400: "Unknown pixelformat" error on ubuntu 16.04 leading to only RGB camera working.

didrocks, 2017-01-03:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version | Didn't find any official Intel tool to get firmware on the ZR300 (received in dec though)  | 
| Operating System & Version | Ubuntu 16.04 LTS |
| Kernel Version (Linux Only)  | 4.4.0-57-generic |
| Build System |  Makefile | 

Pretty similar than other report on Ubuntu 16.04 LTS here, but reporting separately as having a different camera model.

I did run scripts/patch-uvcvideo-16.04.simple.sh, checked that `echo $?` was 0 and that the patch was applied in every refered file.

`/lib/modules/`uname -r`/kernel/drivers/media/usb/uvc/uvcvideo.ko` is the new compiled file copy and `modprobe uvcvideo` was unloaded and reloaded with the new version.

However, I'm getting the *Unknown pixelformat* error when pluging the camera to my usb port in my syslogs (see below) and running `cpp-capture` example only display the RGB camera, no infrared1/2 or depth. I'm getting a ton of:
```
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 frame status 0x1
```

Could that be due to my kernel version? Seeing the number of similar report with other cameras but on same ubuntu version, is there a definitive ubuntu 16.04 kernel known to work with librealsense patch?

Here is the syslog output:
```
janv. 03 12:12:32 tidus kernel: usb 4-1: new SuperSpeed USB device number 20 using xhci_hcd
janv. 03 12:12:33 tidus kernel: usb 4-1: New USB device found, idVendor=0451, idProduct=8025
janv. 03 12:12:33 tidus kernel: usb 4-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
janv. 03 12:12:33 tidus kernel: hub 4-1:1.0: USB hub found
janv. 03 12:12:33 tidus kernel: hub 4-1:1.0: 2 ports detected
janv. 03 12:12:33 tidus kernel: usb 3-1: new high-speed USB device number 14 using xhci_hcd
janv. 03 12:12:33 tidus kernel: usb 3-1: New USB device found, idVendor=0451, idProduct=8027
janv. 03 12:12:33 tidus kernel: usb 3-1: New USB device strings: Mfr=0, Product=0, SerialNumber=1
janv. 03 12:12:33 tidus kernel: usb 3-1: SerialNumber: 9302107160B7
janv. 03 12:12:33 tidus kernel: hub 3-1:1.0: USB hub found
janv. 03 12:12:33 tidus kernel: hub 3-1:1.0: 2 ports detected
janv. 03 12:12:33 tidus kernel: usb 4-1.2: new SuperSpeed USB device number 21 using xhci_hcd
janv. 03 12:12:33 tidus kernel: usb 4-1.2: LPM exit latency is zeroed, disabling LPM.
janv. 03 12:12:33 tidus kernel: usb 4-1.2: New USB device found, idVendor=8086, idProduct=0ad0
janv. 03 12:12:33 tidus kernel: usb 4-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
janv. 03 12:12:33 tidus kernel: usb 4-1.2: Product: CX3-UVC
janv. 03 12:12:33 tidus kernel: usb 4-1.2: Manufacturer: Cypress
janv. 03 12:12:33 tidus kernel: uvcvideo: Found UVC 1.10 device CX3-UVC (8086:0ad0)
janv. 03 12:12:33 tidus kernel: usb 4-1.1: new SuperSpeed USB device number 22 using xhci_hcd
janv. 03 12:12:33 tidus kernel: usb 4-1.1: New USB device found, idVendor=8086, idProduct=0acb
janv. 03 12:12:33 tidus kernel: usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
janv. 03 12:12:33 tidus kernel: usb 4-1.1: Product: Intel RealSense 3D Camera ZR300
janv. 03 12:12:33 tidus kernel: usb 4-1.1: Manufacturer: Intel Corp
janv. 03 12:12:33 tidus kernel: usb 4-1.1: SerialNumber: SN_3391801151
janv. 03 12:12:33 tidus kernel: uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
janv. 03 12:12:33 tidus kernel: uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
janv. 03 12:12:33 tidus kernel: uvcvideo: Unable to create debugfs 4-22 directory.
janv. 03 12:12:33 tidus kernel: uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera ZR300 (8086:0acb)
janv. 03 12:12:33 tidus kernel: uvcvideo: Unable to create debugfs 4-22 directory.
janv. 03 12:12:33 tidus kernel: input: Intel RealSense 3D Camera ZR300 as /devices/pci0000:00/0000:00:1c.6/0000:0e:00.0/usb4/4-1/4-1.1/4-1.1:1.4/input/input29
janv. 03 12:12:33 tidus mtp-probe[9554]: checking bus 4, device 22: "/sys/devices/pci0000:00/0000:00:1c.6/0000:0e:00.0/usb4/4-1/4-1.1"
janv. 03 12:12:33 tidus mtp-probe[9552]: checking bus 4, device 21: "/sys/devices/pci0000:00/0000:00:1c.6/0000:0e:00.0/usb4/4-1/4-1.2"
janv. 03 12:12:33 tidus mtp-probe[9554]: bus: 4, device: 22 was not an MTP device
janv. 03 12:12:33 tidus mtp-probe[9552]: bus: 4, device: 21 was not an MTP device
janv. 03 12:12:33 tidus kernel: ------------[ cut here ]------------
janv. 03 12:12:33 tidus kernel: WARNING: CPU: 2 PID: 5806 at /build/linux-7x12eW/linux-4.4.0/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd86/0x10f0 [videodev]()
janv. 03 12:12:33 tidus kernel: Unknown pixelformat 0x2036315a
janv. 03 12:12:33 tidus kernel: Modules linked in: uvcvideo(OE) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media ufs qnx4 hfsplus hfs minix ntfs msdos jfs xfs nls_utf8 isofs uas usb_storage pci_stub vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) nls_iso8859_1 drbg ansi_cprng ctr ccm rfcomm xfrm_user xfrm_algo xt_addrtype br_netfilter xt_CHECKSUM iptable_mangle ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 nf_nat nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack nf_conntrack ipt_REJECT nf_reject_ipv4 xt_tcpudp aufs bridge stp llc ebtable_filter ebtables ip6table_filter ip6_tables xt_multiport iptable_filter ip_tables x_tables bnep binfmt_misc zfs(PO) zunicode(PO) zcommon(PO) znvpair(PO) spl(O) zavl(PO) btusb btrtl btbcm btintel bluetooth intel_rapl x86_pkg_temp_thermal
janv. 03 12:12:33 tidus kernel:  arc4 intel_powerclamp coretemp iwldvm input_leds snd_hda_codec_hdmi mac80211 joydev snd_hda_codec_conexant snd_hda_codec_generic serio_raw snd_hda_intel iwlwifi snd_hda_codec snd_hda_core cfg80211 snd_hwdep lpc_ich snd_pcm thinkpad_acpi nvram snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device snd_timer mei_me snd mei soundcore shpchp mac_hid kvm_intel kvm irqbypass cuse parport_pc ppdev lp parport autofs4 btrfs raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear mmc_block crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel aes_x86_64 lrw i915 gf128mul glue_helper ablk_helper cryptd psmouse ahci libahci i2c_algo_bit drm_kms_helper sdhci_pci e1000e sdhci syscopyarea sysfillrect sysimgblt
janv. 03 12:12:33 tidus kernel:  ptp fb_sys_fops pps_core drm wmi fjes video [last unloaded: videobuf2_memops]
janv. 03 12:12:33 tidus kernel: CPU: 2 PID: 5806 Comm: AudioThread Tainted: P        W  OE   4.4.0-57-generic #78-Ubuntu
janv. 03 12:12:33 tidus kernel: Hardware name: LENOVO 4287CTO/4287CTO, BIOS 8DET58WW (1.28 ) 02/14/2012
janv. 03 12:12:33 tidus kernel:  0000000000000286 0000000014082052 ffff8800b1d5bbf0 ffffffff813f6f33
janv. 03 12:12:33 tidus kernel:  ffff8800b1d5bc38 ffffffffc0887a78 ffff8800b1d5bc28 ffffffff810812d2
janv. 03 12:12:33 tidus kernel:  ffff8800b1d5bda0 0000000000000000 0000000000000000 ffff8801d302e240
janv. 03 12:12:33 tidus kernel: Call Trace:
[... 111 lines cut ...]
janv. 03 12:12:33 tidus kernel: CPU: 1 PID: 5806 Comm: AudioThread Tainted: P        W  OE   4.4.0-57-generic #78-Ubuntu
janv. 03 12:12:33 tidus kernel: Hardware name: LENOVO 4287CTO/4287CTO, BIOS 8DET58WW (1.28 ) 02/14/2012
janv. 03 12:12:33 tidus kernel:  0000000000000286 0000000014082052 ffff8800b1d5bbf0 ffffffff813f6f33
janv. 03 12:12:33 tidus kernel:  ffff8800b1d5bc38 ffffffffc0887a78 ffff8800b1d5bc28 ffffffff810812d2
janv. 03 12:12:33 tidus kernel:  ffff8800b1d5bda0 0000000000000000 ffff880018b77200 ffff880068b390c0
janv. 03 12:12:33 tidus kernel: Call Trace:
janv. 03 12:12:33 tidus kernel:  [<ffffffff813f6f33>] dump_stack+0x63/0x90
janv. 03 12:12:33 tidus kernel:  [<ffffffff810812d2>] warn_slowpath_common+0x82/0xc0
janv. 03 12:12:33 tidus kernel:  [<ffffffff8108136c>] warn_slowpath_fmt+0x5c/0x80
janv. 03 12:12:33 tidus kernel:  [<ffffffffc108472d>] ? uvc_ioctl_enum_fmt+0x9d/0xc0 [uvcvideo]
janv. 03 12:12:33 tidus kernel:  [<ffffffffc08726f6>] v4l_enum_fmt+0xd86/0x10f0 [videodev]
janv. 03 12:12:33 tidus kernel:  [<ffffffffc08732c1>] __video_do_ioctl+0x291/0x310 [videodev]
janv. 03 12:12:33 tidus kernel:  [<ffffffff81218d04>] ? terminate_walk+0x64/0xd0
janv. 03 12:12:33 tidus kernel:  [<ffffffff811ed69b>] ? __slab_free+0xcb/0x2c0
janv. 03 12:12:33 tidus kernel:  [<ffffffffc0872d96>] video_usercopy+0x336/0x5b0 [videodev]
janv. 03 12:12:33 tidus kernel:  [<ffffffffc0873030>] ? video_ioctl2+0x20/0x20 [videodev]
[... cut ...]
janv. 03 12:12:33 tidus kernel:  [<ffffffffc086e6a3>] v4l2_ioctl+0xd3/0xe0 [videodev]
janv. 03 12:12:33 tidus kernel:  [<ffffffff8122224f>] do_vfs_ioctl+0x29f/0x490
janv. 03 12:12:33 tidus kernel:  [<ffffffff8121df84>] ? putname+0x54/0x60
janv. 03 12:12:33 tidus kernel:  [<ffffffff8120d8df>] ? do_sys_open+0x1bf/0x2a0
janv. 03 12:12:33 tidus kernel:  [<ffffffff812224b9>] SyS_ioctl+0x79/0x90
janv. 03 12:12:33 tidus kernel:  [<ffffffff818374f2>] entry_SYSCALL_64_fastpath+0x16/0x71
janv. 03 12:12:33 tidus kernel: ---[ end trace 21088cf1a1899da7 ]---
```

dorodnic, 2017-01-03:
Hi @didrocks 
The log is indicative of a non-patched 4.4 kernel so there might have been some problems. Could you please run [gt_setup_verification.sh](https://github.com/IntelRealSense/librealsense/blob/master/scripts/gt_setup_verification.sh) to make sure the setup is Ok?
However, on Ubuntu 16.04 ZR300 Depth and IR should play out of the box, even without any patches.  Depth frames that arrive to librealsense seem to be corrupted. Can you please check the firmware version(s) through cpp-enumerate-devices example?

didrocks, 2017-01-03:
@dorodnic Thanks for the quick answer!
So, running gt_setup_verification.sh confirms that I have a patched kernel:
```
$ ./gt_setup_verification.sh 

Host Info:

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.1 LTS
Release:	16.04
Codename:	xenial

Kernel Version:

Linux 4.4.0-57-generic x86_64

Connected UVC Devices: 

UVC streaming device was found...

ZR300 Camera was found - Ok

Verify formats required by ZR300 Camera: 

Y8I	-	Ok
Y12I	-	Ok
Y16	-	Ok
Z16	-	Ok
GREY	-	Ok
YUYV	-	Ok
RW16	-	Ok

The pixel formats are set properly! 

Done...
```

However, `cpp-enumerates-devices` confirms that some communication with the Camera isn't working properly:
```$ ./cpp-enumerate-devices 
rs.error: Failed to get firmware version
rs.error: Couldn't query adapter board / motion module FW version!
rs.warn: Motion module capabilities were disabled due to failure to acquire intrinsic
Device 0 - Intel RealSense ZR300:
 Serial number: 3391801151
 Firmware version: 2.0.71.26
 USB Port ID: 4-1-2
RealSense error calling rs_get_device_info(device:0x747e00, info:ADAPTER_BOARD_FIRMWARE_VERSION):
    selected camera info is not supported for this camera!
```
(or maybe the first errors is due to internal laptop usb camera?)

dorodnic, 2017-01-03:
Couple of extra questions: What is the platform (Up board / Intel Joule / PC)? How is the camera  connected? Are you using a USB Hub, and if yes did you try to connect the camera directly?

didrocks, 2017-01-03:
The platform is my PC for now. Lenovo x220, and the camera is plugged directly to an usb3 port (I tried to avoid any kind of USB Hub and so on on purpose to avoid those kind of issues.

dorodnic, 2017-01-03:
Are you running in a VM?

didrocks, 2017-01-03:
I'm not running in a VM, but natively.

I just did try multiple times to reload the modules, here are some updates:
Now cpp-enumerate-devices gives:
```
$ ./cpp-enumerate-devices 
Device 0 - Intel RealSense ZR300:
 Serial number: 3391801151
 Firmware version: 2.0.71.26
 USB Port ID: 4-1-2
 Adapter Board Firmware version: 1.29.0.0
 Motion Module Firmware version: 1.25.0.0
 Camera info: 
    DEVICE_NAME         : 	Intel RealSense ZR300
    DEVICE_SERIAL_NUMBER: 	3391801151
    CAMERA_FIRMWARE_VERSION: 	2.0.71.26
    ADAPTER_BOARD_FIRMWARE_VERSION: 	1.29.0.0
    MOTION_MODULE_FIRMWARE_VERSION: 	1.25.0.0
    CAMERA_TYPE         : 	PRQ-Ready
    OEM_ID              : 	OEM None
    ISP_FW_VERSION      : 	0xd942
    CONTENT_VERSION     : 	12
    MODULE_VERSION      : 	4.4.0.111
    IMAGER_MODEL_NUMBER : 	33
    CALIBRATION_DATE    : 	2014-07-04 08:18:35 UTC
    EMITTER_TYPE        : 	Laser Driver 4.1
    FOCUS_VALUE         : 	0
    LENS_TYPE           : 	Newmax 58.9 x 45.9 degs in VGA
    3RD_LENS_TYPE       : 	AEOT
    LENS_COATING__TYPE  : 	Visible-light block / IR pass 43 nm width
    3RD_LENS_COATING_TYPE: 	IR coating
    NOMINAL_BASELINE    : 	70 mm
    3RD_NOMINAL_BASELINE: 	58 mm
 Supported options:                                    min        max       step  default  
    COLOR_BACKLIGHT_COMPENSATION                       : 0    ... 1           1     0         
    COLOR_BRIGHTNESS                                   : -64  ... 64          1     0         
    COLOR_CONTRAST                                     : 0    ... 100         1     50        
    COLOR_EXPOSURE                                     : 50   ... 666         1     166       
    COLOR_GAIN                                         : 0    ... 128         1     64        
    COLOR_GAMMA                                        : 100  ... 500         1     300       
    COLOR_HUE                                          : -180 ... 180         1     0         
    COLOR_SATURATION                                   : 0    ... 100         1     64        
    COLOR_SHARPNESS                                    : 0    ... 100         1     50        
    COLOR_WHITE_BALANCE                                : 2800 ... 6500        10    4600      
    COLOR_ENABLE_AUTO_EXPOSURE                         : 0    ... 1           1     1         
    COLOR_ENABLE_AUTO_WHITE_BALANCE                    : 0    ... 1           1     1         
    R200_LR_AUTO_EXPOSURE_ENABLED                      : 0    ... 1           1     0         
    R200_LR_GAIN                                       : 100  ... 6399        1     400       
    R200_LR_EXPOSURE                                   : 1    ... 330         1     164       
    R200_EMITTER_ENABLED                               : 0    ... 1           1     0         
    R200_DEPTH_UNITS                                   : 0    ... 2.14748e+09 1     1000      
    R200_DEPTH_CLAMP_MIN                               : 0    ... 65535       1     0         
    R200_DEPTH_CLAMP_MAX                               : 0    ... 65535       1     65535     
    R200_DEPTH_CONTROL_ESTIMATE_MEDIAN_DECREMENT       : 0    ... 255         1     5         
    R200_DEPTH_CONTROL_ESTIMATE_MEDIAN_INCREMENT       : 0    ... 255         1     5         
    R200_DEPTH_CONTROL_MEDIAN_THRESHOLD                : 0    ... 1023        1     192       
    R200_DEPTH_CONTROL_SCORE_MINIMUM_THRESHOLD         : 0    ... 1023        1     1         
    R200_DEPTH_CONTROL_SCORE_MAXIMUM_THRESHOLD         : 0    ... 1023        1     512       
    R200_DEPTH_CONTROL_TEXTURE_COUNT_THRESHOLD         : 0    ... 31          1     6         
    R200_DEPTH_CONTROL_TEXTURE_DIFFERENCE_THRESHOLD    : 0    ... 1023        1     24        
    R200_DEPTH_CONTROL_SECOND_PEAK_THRESHOLD           : 0    ... 1023        1     27        
    R200_DEPTH_CONTROL_NEIGHBOR_THRESHOLD              : 0    ... 1023        1     7         
    R200_DEPTH_CONTROL_LR_THRESHOLD                    : 0    ... 2047        1     24        
    FISHEYE_EXPOSURE                                   : 2    ... 320         1     4         
    FISHEYE_GAIN                                       : 0    ... 255         1     0         
    FISHEYE_STROBE                                     : 0    ... 1           1     0         
    FISHEYE_EXTERNAL_TRIGGER                           : 0    ... 1           1     0         
    FISHEYE_ENABLE_AUTO_EXPOSURE                       : 0    ... 1           1     1         
    FISHEYE_AUTO_EXPOSURE_MODE                         : 0    ... 2           1     0         
    FISHEYE_AUTO_EXPOSURE_ANTIFLICKER_RATE             : 50   ... 60          10    60        
[... 182 lines cut ...]
  320	x 240	@ 90Hz	Y8	55.2 x 42.8 degrees, distortion = NONE
  320	x 240	@ 90Hz	Y16	55.2 x 42.8 degrees, distortion = NONE
  320	x 240	@ 60Hz	Y8	55.2 x 42.8 degrees, distortion = NONE
  320	x 240	@ 60Hz	Y16	55.2 x 42.8 degrees, distortion = NONE
  320	x 240	@ 30Hz	Y8	55.2 x 42.8 degrees, distortion = NONE
  320	x 240	@ 30Hz	Y16	55.2 x 42.8 degrees, distortion = NONE
 Stream FISHEYE - 2 modes:
  640	x 480	@ 60Hz	RAW8	102 x 85.9 degrees, distortion = FTHETA
  640	x 480	@ 30Hz	RAW8	102 x 85.9 degrees, distortion = FTHETA
```

which is more inlined with what is expected I guess. However, I still have the same stacktrace in syslogs and cpp-capture only show frames incrementing in RGB camera (and Fisheye, but fisheye doesn't show any image), with:
```
[... cut ...]
Capturing DEPTH at 480 x 360, fov = 55.9 x 43.4, distortion = NONE
Capturing COLOR at 640 x 480, fov = 55.1 x 42.5, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 55.9 x 43.4, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 55.9 x 43.4, distortion = NONE
Capturing FISHEYE at 640 x 480, fov = 101.9 x 85.9, distortion = FTHETA
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 frame status 0x1
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 frame status 0x1
[…]
```

dorodnic, 2017-01-03:
So at the very least we know that:
a. Linux version and kernel version are fine, patch was successfully applied 
b. The system correctly recognizes the device
c. Firmware versions are fine
From librealsense point of view everything should be working. I can try to look-up "frame status 0x1" with the hardware team. In addition, in the unfortunate case that there is some hardware problem with the device, I'll also check who can help with hardware support. 
In the meanwhile,
1. If you are running on a laptop on battery, please try connecting it to power
2. If you have access to a powered USB3 Hub, this is also worth a try
3. Could you please run cpp-config-ui and check if depth alone, IR alone, depth + IR, depth + color are working. Also, in config-ui you can see the FPS, so you can check if fish-eye is not streaming or if it is just producing dark frames.

didrocks, 2017-01-03:
Ok, thanks for keeping me posted on the hardware side :)
on your questions:
1. I'm not running on battery
2. I don't have access to any power USB3 Hub unfortunately.
3. I did try this. And indeed, cpp-config-ui works: I have images for depth+color+IR1&2 all together. (I still have the same warnings about bad magic numbers though). So, could it be that cpp-capture example is broken?
The only one not working is the fishey one, even if enabled alone. The number of FPS is ramping up, but the image stays black, despite moving it, putting some lights on every side and such.

ev-mp, 2017-01-03:
Hi, If the FPS number for FishEye monotonically increases, then the frames arrive.  You can do a rough FPS approximation by dividing the frame number by timestamp (in millisec) - For FishEye the number is either 30 or 60.
In addition there are two GUI controls for FishEye that you should also try : "Exposure" and "Gain" that appear at the bottom of the sliders panel, next to "STROBE" and "External Trigger" check boxes.
And last thing, and I know that this sounds silly, but please verify that you have the FishEye lens cover plastic cap removed.

didrocks, 2017-01-04:
Thanks for your feedback!
So, after some rebooting and more experimenting, this morning, it does work quite ok I must say, on both the Joule and my laptop, including the fisheye one.

I'm still having the same stacktrace in syslog, however, I can just get frames and visual data via `cpp-config-ui` now. It just seems that the connexion is quite flacky and sometimes, I have some run without any. Relaunching cpp-config-ui works though.

So, it seems:
- the connexion can be flacky and some run will fail
- plug in the camera produce a kernel module stacktrace (as shown above), reliably
- `cpp-capture` is sample code is broken with the ZR300, and only show fisheye and RGB camera data (printing the error avobe)
- same with other sample code like `c-tutorial-1-depth` which does render only very few ascii character and doesn't seem to represent any particular shape, like hands and such, despite any distance of side of the camera you are using (can it be that the value given by this camera is really different than others and it's not configured for it?)
- `cpp-config-ui` is the most reliable sample to test if the camera is working

dorodnic, 2017-01-04:
There are couple of factors that have significant effect on RealSense performance:
a. Cables length and quality
b. Power (ZR300 in particular is quite power-hungry)
c. USB controller

I don't think there is anything wrong with cpp-capture software: The difference between cpp-capture and cpp-config-ui is that the former is trying to synchronize the streams based on timestamp (using wait_for_frame API), while the later does not (it uses direct callbacks with no synchronization). 
My theory is that your device produce corrupted frame headers, causing loss of frame timestamps. 

Regarding kernel warnings, there seem to be some conflicting information. Assuming you apply the latest patch from the master branch, you can see that the reported pixel formats are included in the headers. 
However, even if the patch was not applied for some reason, most of ZR300 functionality should not be affected on Ubuntu 16. You should be able to get Depth, Color, IR, Fish-eye and motion out-of-the-box, without any patches at all.

alexge233, 2017-05-16:
Sorry to revive this issue, seeing that it is still open I've encountered the same problem.
My system is Ubuntu 16.04 with kernel 4.4.0-66.
When I run `roslaunch realsense_camera zr300_nodelet_default.launch` I get:

```
process[camera/nodelet_manager-1]: started with pid [7436]
process[camera/driver-2]: started with pid [7437]
[ INFO] [1494931645.061626656]: Initializing nodelet with 2 worker threads.
[ERROR] [1494931645.393627308]: /camera/driver - Error calling rs_get_device_info ( device:0xf707d0, info:ADAPTER_BOARD_FIRMWARE_VERSION ): 
selected camera info is not supported for this camera! 

[FATAL] [1494931645.581627677]: Failed to load nodelet '/camera/driver` of type `realsense_camera/ZR300Nodelet` to manager `nodelet_manager'
[camera/nodelet_manager-1] process has died [pid 7436, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=nodelet_manager __log:=/home/mario/mario_ws/log/25a1727c-3a22-11e7-ba87-408d5ca7bf3c/camera-nodelet_manager-1.log].
log file: /home/mario/mario_ws/log/25a1727c-3a22-11e7-ba87-408d5ca7bf3c/camera-nodelet_manager-1*.log
[camera/driver-2] process has died [pid 7437, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense_camera/ZR300Nodelet nodelet_manager depth:=depth color:=color ir:=ir ir2:=ir2 fisheye:=fisheye imu:=imu __name:=driver __log:=/home/mario/mario_ws/log/25a1727c-3a22-11e7-ba87-408d5ca7bf3c/camera-driver-2.log].
log file: /home/mario/mario_ws/log/25a1727c-3a22-11e7-ba87-408d5ca7bf3c/camera-driver-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

My camera is a ZR300 received April 2017.

supertz, 2017-06-28:
My camera is a ZR300.
I did run gt_setup_verification.sh and the output was:
`Host Info:

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.2 LTS
Release:	16.04
Codename:	xenial

Kernel Version:

Linux 4.4.74-040474-generic x86_64

Connected UVC Devices: 

UVC streaming device was found...

ZR300 Camera was found - Ok

Verify formats required by ZR300 Camera: 

Y8I	-	Ok
Y12I	-	Ok
Y16	-	Ok
Z16	-	Ok
GREY	-	Ok
YUYV	-	Ok
RW16	-	Ok

The pixel formats are set properly! 
`
I did run cpp-capture and the output was:
`No device detected. Is it plugged in?
`

Can you help me with this problem? I have wasted several days for it.  :-(

ev-mp, 2017-08-20:
Hi @supertz , please provide the following info:
 - Kernel log ("dmesg -wT") when the camera is being plugged in
- A list of USB devices with ZR300 connected to USB3 port (generated with "lsusb")
Additionally, please recompile and rerun  cpp-enumerate-devices demo with the following line enabled 
rs::log_to_file(rs::log_severity::debug, "librealsense.log");, and add the resulted log file to the list

jlgarcia75, 2017-10-03:
Hello,

If the issue is software related, we are no longer updating this software as we have moved to our latest generation, the D400 series. If you are in your warranty, a replacement can be provided.

Jesus G.
Intel Customer Support

closed 2017-10-03 by jlgarcia75