realsenseai/librealsense #401: Libglfw3.dylib is deprecated in OSX homebrew

martinthomas, 2017-01-04:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 | 
| Firmware Version | N/A | 
| Operating System & Version | OSX  |
| Kernel Version (Linux Only)  |   |
| Build System |  XCode | 

The OSX installation instructions say to install libglfw3 but brew indicates it is deprecated and instead installs libglfw.

Either the Xcode projects need to be changed to link against libglfw.dylib (preferable) or the user needs to symlink from libglfw.dylib to libglfw3.dylib.

dorodnic, 2017-01-08:
Thank you for your feedback, would you like to contribute a fix to the XCode projects or document the work around procedure?

martinthomas, 2017-01-09:
I'll work on a fix.

martinthomas, 2017-01-09:
Pull request #408

OrganicIrradiation, 2017-03-25:
@martinthomas, are you able to install librealsense via homebrew now? (see: https://github.com/IntelRealSense/librealsense/pull/452)

jlgarcia75, 2017-10-06:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

closed 2017-10-16 by jlgarcia75