realsenseai/librealsense #403: How to run multiple realsense R200 devices on a single Ubuntu system？

XiaoHu1994, 2017-01-05:
When I only connected to a realsenser 200 device, run the./cpp-capture program, display normal, and there are four image output.

hwk@hwk-turtlebot:~/librealsense/build/examples$ ./cpp-capture 
Capturing DEPTH at 480 x 360, fov = 55.6 x 43.1, distortion = NONE
Capturing COLOR at 640 x 480, fov = 53.7 x 41.2, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 55.6 x 43.1, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 55.6 x 43.1, distortion = NONE

But when I open a new Ubuntu terminal, then access another realsenser R200 device, and run the./cpp-capture program, it will be wrong.

Capturing DEPTH at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
Capturing COLOR at 640 x 480, fov = 53.8 x 41.2, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 55.3 x 42.9, distortion = NONE
RealSense error calling rs_start_source(device:0x18d4ec8, source:VIDEO):
    UVCIOC_CTRL_QUERY:UVC_SET_CUR error 5, Input/output error

The image of the first device is normal。
So，How to run multiple realsense R200 devices on a single Ubuntu system？

Thank you for your attention and suggestions！

dorodnic, 2017-01-05:
Hi, did you try to run the cpp-multicam example?

XiaoHu1994, 2017-01-08:
@dorodnic  Thank you very much!

dorodnic, 2017-01-08:
Excellent. Feel free to reopen the issue if you have more questions. 
Please take into account that multi-camera setups are extremely sensitive to hardware setup (Optimally, each device should be connected through an independent USB3 controller to avoid USB overflows and power issues)

closed 2017-01-08 by dorodnic