realsenseai/librealsense #406: R200 always do not working on linux

BlazingForests, 2017-01-06:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.10  | 
| Operating System & Version | Ubuntu 14.04.5 LTS |
| Kernel Version (Linux Only)  | 4.4.0-58-generic #79~14.04.1-Ubuntu  |
| Build System |  Makefile | 

I am using `apt-get install ros-indigo-realsense-camera` to install RealSense camera.

When i power on my computer, plug-in R200, the `dmesg` output is

```
[ 1027.491773] usb 4-3: new SuperSpeed USB device number 2 using xhci_hcd
[ 1027.510455] usb 4-3: New USB device found, idVendor=8086, idProduct=0a80
[ 1027.510461] usb 4-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1027.510465] usb 4-3: Product: Intel RealSense 3D Camera R200
[ 1027.510481] usb 4-3: Manufacturer: Intel Corp
[ 1027.510484] usb 4-3: SerialNumber: SN_2361000097
[ 1027.552433] media: Linux media interface: v0.10
[ 1027.566863] Linux video capture interface: v2.00
[ 1027.598452] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1027.600217] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1027.601842] uvcvideo: Unable to create debugfs 4-2 directory.
[ 1027.601944] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:1c.6/0000:07:00.0/usb4/4-3/4-3:1.2/input/input16
[ 1027.602037] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[ 1027.603758] uvcvideo: Unable to create debugfs 4-2 directory.
[ 1027.603886] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:1c.6/0000:07:00.0/usb4/4-3/4-3:1.4/input/input17
[ 1027.603960] usbcore: registered new interface driver uvcvideo
[ 1027.603962] USB Video Class driver (1.1.1-3-realsense)
[ 1027.620506] ------------[ cut here ]------------
[ 1027.620525] WARNING: CPU: 6 PID: 3614 at /build/linux-lts-xenial-OvqN2x/linux-lts-xenial-4.4.0/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd3e/0x10c0 [videodev]()
[ 1027.620534] Unknown pixelformat 0x2036315a
[ 1027.620535] Modules linked in: uvcvideo(OE) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 nf_nat nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack nf_conntrack ipt_REJECT nf_reject_ipv4 xt_CHECKSUM iptable_mangle xt_tcpudp bridge stp llc xt_multiport ip6table_filter ip6_tables iptable_filter ip_tables ebtable_nat ebtables x_tables vmw_vsock_vmci_transport vsock vmw_vmci nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) input_leds snd_hda_codec_hdmi snd_hda_codec_realtek bnep rfcomm snd_hda_codec_generic bluetooth snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm drm_kms_helper drm snd_pcm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device snd_timer snd aes_x86_64 lrw gf128mul glue_helper ablk_helper cryptd serio_raw sb_edac edac_core fb_sys_fops mei_me syscopyarea sysfillrect mei sysimgblt wmi soundcore shpchp 8250_fintek lpc_ich mac_hid parport_pc ppdev lp parport binfmt_misc nls_iso8859_1 hid_generic usbhid hid uas usb_storage isci psmouse e1000e libsas ahci ptp libahci scsi_transport_sas pps_core pata_acpi fjes
[ 1027.620615] CPU: 6 PID: 3614 Comm: AudioThread Tainted: P           OE   4.4.0-58-generic #79~14.04.1-Ubuntu
[ 1027.620617] Hardware name: LENOVO 98002825/LENOVO, BIOS A2KT44AUS 03/13/2014
[ 1027.620619]  0000000000000000 ffff88042279bbe8 ffffffff813dbd6c ffff88042279bc30
[ 1027.620623]  ffffffffc060bab0 ffff88042279bc20 ffffffff8107e9d6 ffff88042279bd98
[ 1027.620626]  0000000000000000 0000000000000000 ffff8800702cba00 ffffffffc060e180
[ 1027.620629] Call Trace:
[ 1027.620634]  [<ffffffff813dbd6c>] dump_stack+0x63/0x87
[ 1027.620638]  [<ffffffff8107e9d6>] warn_slowpath_common+0x86/0xc0
[ 1027.620640]  [<ffffffff8107ea5c>] warn_slowpath_fmt+0x4c/0x50
[ 1027.620645]  [<ffffffffc0646197>] ? uvc_ioctl_enum_fmt+0xb7/0x140 [uvcvideo]
[ 1027.620650]  [<ffffffffc05f580e>] v4l_enum_fmt+0xd3e/0x10c0 [videodev]
[ 1027.620655]  [<ffffffffc05f5dea>] __video_do_ioctl+0x25a/0x2e0 [videodev]
[ 1027.620658]  [<ffffffff812202e4>] ? mntput+0x24/0x40
[ 1027.620661]  [<ffffffff8120e539>] ? path_openat+0x2e9/0x12d0
[ 1027.620666]  [<ffffffffc05f83fb>] video_usercopy+0x21b/0x590 [videodev]
[ 1027.620670]  [<ffffffffc05f5b90>] ? v4l_enum_fmt+0x10c0/0x10c0 [videodev]
[ 1027.620673]  [<ffffffff812108a2>] ? do_filp_open+0x92/0xd0
[ 1027.620678]  [<ffffffffc05f8785>] video_ioctl2+0x15/0x20 [videodev]
[ 1027.620682]  [<ffffffffc05f1644>] v4l2_ioctl+0xa4/0xc0 [videodev]
[ 1027.620685]  [<ffffffff8121396d>] do_vfs_ioctl+0x2dd/0x4c0
[ 1027.620687]  [<ffffffff8120f983>] ? putname+0x53/0x60
[ 1027.620689]  [<ffffffff811ffbb8>] ? do_sys_open+0x1b8/0x270
[ 1027.620692]  [<ffffffff81213bc9>] SyS_ioctl+0x79/0x90
[ 1027.620695]  [<ffffffff81802c76>] entry_SYSCALL_64_fastpath+0x16/0x75
[ 1027.620697] ---[ end trace 5c33dafd59dc85f9 ]---
[ 1027.621064] ------------[ cut here ]------------
[ 1027.621071] WARNING: CPU: 6 PID: 3614 at /build/linux-lts-xenial-OvqN2x/linux-lts-xenial-4.4.0/drivers/media/v4l2-core/v4l2-ioctl.c:1272 v4l_enum_fmt+0xd3e/0x10c0 [videodev]()
[ 1027.621072] Unknown pixelformat 0x49323159
[ 1027.621073] Modules linked in: uvcvideo(OE) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common videodev media ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 nf_nat nf_conntrack_ipv4 nf_defrag_ipv4 xt_conntrack nf_conntrack ipt_REJECT nf_reject_ipv4 xt_CHECKSUM iptable_mangle xt_tcpudp bridge stp llc xt_multiport ip6table_filter ip6_tables iptable_filter ip_tables ebtable_nat ebtables x_tables vmw_vsock_vmci_transport vsock vmw_vmci nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) input_leds snd_hda_codec_hdmi snd_hda_codec_realtek bnep rfcomm snd_hda_codec_generic bluetooth snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm drm_kms_helper drm snd_pcm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device snd_timer snd aes_x86_64 lrw gf128mul glue_helper ablk_helper cryptd serio_raw sb_edac edac_core fb_sys_fops mei_me syscopyarea sysfillrect mei sysimgblt wmi soundcore shpchp 8250_fintek lpc_ich mac_hid parport_pc ppdev lp parport binfmt_misc nls_iso8859_1 hid_generic usbhid hid uas usb_storage isci psmouse e1000e libsas ahci ptp libahci scsi_transport_sas pps_core pata_acpi fjes
[ 1027.621120] CPU: 6 PID: 3614 Comm: AudioThread Tainted: P        W  OE   4.4.0-58-generic #79~14.04.1-Ubuntu
[ 1027.621122] Hardware name: LENOVO 98002825/LENOVO, BIOS A2KT44AUS 03/13/2014
[ 1027.621123]  0000000000000000 ffff88042279bbe8 ffffffff813dbd6c ffff88042279bc30
[ 1027.621125]  ffffffffc060bab0 ffff88042279bc20 ffffffff8107e9d6 ffff88042279bd98
[ 1027.621126]  0000000000000000 0000000000000000 ffff8803c90ff200 ffffffffc060e180
[ 1027.621128] Call Trace:
[ 1027.621131]  [<ffffffff813dbd6c>] dump_stack+0x63/0x87
[... 1615 lines cut ...]
[ 1732.480419] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1732.828377] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1733.176338] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1733.524339] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1733.872334] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1734.220308] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1734.568282] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1734.916284] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1735.264181] uvcvideo: Failed to query (SET_CUR) UVC control 8 on unit 7: -110 (exp. 2).
[ 1735.564151] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1735.912193] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1736.260106] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1736.608078] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1736.956083] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1737.304021] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[... cut ...]
[ 1738.695943] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1739.043910] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).
[ 1739.391877] uvcvideo: Failed to query (SET_CUR) UVC control 11 on unit 7: -110 (exp. 1).

[ 1849.821214] systemd-hostnamed[9833]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
[ 1864.474071] usb 4-3: Failed to disable LTM before suspend
```

dorodnic, 2017-01-08:
ROS related questions are better to be discussed at [github.com/intel-ros/realsense](https://github.com/intel-ros/realsense/issues)

closed 2017-01-08 by dorodnic

BlazingForests, 2017-01-08:
Hi, @dorodnic 

OK, i will move this problem to intel-ros. But i really think this problem irrelevant to ROS.
If i can't get answer in intel-ros, I will come back.

THX

teknotus, 2017-01-08:
It might be a permissions issue. Librealsense uses the extended camera
control ioctl instead of the camera control ioctl. The kernel seems to
treat them differently as far as permissions. One needs nearly raw io
permissions whereas the other only needs camera permissions.

BlazingForests, 2017-01-09:
Hi, @teknotus Daniel, Long time no see :)

But why when i restart the computer, it will working a few minute? then ERROR!

THX
DD

teknotus, 2017-01-10:
Hi, @teknotus <https://github.com/teknotus> Daniel, Long time no see :)

Yeah! I saw a video of your robot on YouTube. Great work.

But why when i restart the computer, it will working a few minute? then
ERROR!

Could be a kernel bug. When I used epoll to have the kernel notify that a
new frame is available it works the first time I start the camera but not
the second. Unplugging the camera and plugging it back in makes it work
again. Detaching, and reattaching via sysfs, or reinitializing the kernel
moduel via rmmod insmod are also methods that work. Oh and a USB port reset
via ioctl on usb.

I don't think the bug is specific to RealSense cameras, but I don't have
any other UVC cameras that trigger it. I think most UVC cameras use
isochronous transfers and the RealSense cameras use bulk transfers so that
might be part of why it hasn't shown up much.

I should note that I have only seen the bug when using mmap and epoll. I
don't think librealsense uses epoll, but the symptoms are the same so I
think it could be the same kernel bug.

BlazingForests, 2017-01-10:
Hi, @teknotus 

OK, But even i replug-in, still not working...

teknotus, 2017-01-10:
OK, But even i replug-in, still not working...

Do you get any new /dev/videoX nodes after reattaching the camera? If you
do then something is preventing cleanup of the camera in kernel space.
Possibly an open file handle in user space. Normally the kernel will reuse
the old numbers as the video4linux core just looks for the lowest unused
number.

If that is the case try lsof on the /dev/video nodes to find out what is
locking it up.

BlazingForests, 2017-01-20:
@teknotus 

Thanks, I will try it.