realsenseai/librealsense #407: Video4Linux backend install error

richardgedwards, 2017-01-08:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu 16.04.1 LTS  |
| Kernel Version (Linux Only)  | 4.4.0-57-generic  |
| Build System |  VS2013, XCode or Makefile | 

After running the following script 
```
 scripts/patch-uvcvideo-16.04.simple.sh  
```
All goes well except for loading the driver with
```
$ sudo modprobe uvcvideo
```

which generates the following error

```
modprobe: ERROR: could not insert 'uvcvideo': Required key not available
```

Here is the last few lines of 
```
sudo dmesg | tail -n
```

```
[  225.153204] usbcore: deregistering interface driver uvcvideo
[  225.436838] media: Linux media interface: v0.10
[  225.441827] Linux video capture interface: v2.00
[ 8383.147599] [drm:intel_dp_link_training_channel_equalization [i915]] *ERROR* 5.4 Gbps link rate without HBR2/TPS3 support
[24491.664566] [drm:intel_dp_link_training_channel_equalization [i915]] *ERROR* 5.4 Gbps link rate without HBR2/TPS3 support
[25507.682479] [drm:intel_dp_link_training_channel_equalization [i915]] *ERROR* 5.4 Gbps link rate without HBR2/TPS3 support
```

ev-mp, 2017-01-09:
This is probably a case of a security feature enabled in BIOS on your PC.
This feature, EFI_SECURE_BOOT_SIG_ENFORCE,  will prevent loading of non-signed kernel modules.

The following link will provide an explanatory and also specify several ways to overcome it -
http://askubuntu.com/questions/762254/why-do-i-get-required-key-not-available-when-install-3rd-party-kernel-modules

boulabiar, 2017-01-13:
I followed the install process, then after modprobbing the module I have a uvc problem

$ ./cpp-alignimages 
RealSense error calling rs_create_context(api_version:11201):
    UVCIOC_CTRL_QUERY:UVC_GET_CUR error 56, Invalid request code

ev-mp, 2017-08-20:
Hi @boulabiar , I'm closing this ticket as the installation issue for the patched module was resolved.
You're welcomed to open additional tickets if you're experiencing run-time issues with the device.

closed 2017-08-20 by ev-mp