realsenseai/librealsense #409: Unable to get depth and IR image

kazoo-kmt, 2017-01-09:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 16.04 LTS  |
| Kernel Version (Linux Only)  | 4.4.0-57-generic  |
| Build System |  Makefile | 

Hi, I have the exactly same issue on the same spec machine as #396 . Even though I applied "patch-uvcvideo-16.04.simple.sh", it didn't solve the issue (no depth and ir image when I ran cpp-capture, for example).
When I ran the patch script (`~/Projects/librealsense$ ./scripts/patch-uvcvideo-16.04.simple.sh`), I had the following messages. After running this, I reloaded `sudo modprobe uvcvideo`.

Are there anything I can try?

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libusb-1.0-0-dev is already the newest version (2:1.0.20-1).
The following packages were automatically installed and are no longer required:
  libpango1.0-0 libpangox-1.0-0
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.1ubuntu2).
linux-headers-generic is already the newest version (4.4.0.57.60).
The following packages were automatically installed and are no longer required:
  libpango1.0-0 libpangox-1.0-0
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 5 not upgraded.
patching file drivers/media/usb/uvc/Makefile
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/media/usb/uvc/Makefile.rej
patching file drivers/media/usb/uvc/uvc_driver.c
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/media/usb/uvc/uvc_driver.c.rej
patching file drivers/media/usb/uvc/uvcvideo.h
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/media/usb/uvc/uvcvideo.h.rej
patching file include/uapi/linux/videodev2.h
Reversed (or previously applied) patch detected!  Assume -R? [n] n
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file include/uapi/linux/videodev2.h.rej
```

ev-mp, 2017-01-09:
The patch was executed at least twice. For whatever reason it didn't work on the first shot, you'll need to run again.
But before that you also have to to revert all the kernel changes applied:
      - browse to "ubuntu-xenial" folder and reset all changes with "$git reset --hard".
Or  -  delete/rename "ubuntu-xenial" folder. (the former will save you ~5 min time)

After that rerun the patch and collect script/kernel logs for verification.

kazoo-kmt, 2017-01-09:
Thank you. I executed `./scripts/patch-uvcvideo-16.04.simple.sh` once, and because it didn't work, I tried again. 

"ubuntu-xenial" folder is under librealsense folder. So the process is
- reset all changes with `git reset --hard`
- run `./scripts/patch-uvcvideo-16.04.simple.sh` again
Is this correct?

ev-mp, 2017-01-09:
A dependency package is missing - openssl:

> scripts/sign-file.c:23:30: fatal error: openssl/opensslv.h: N

You can install it directly, or run an additional Librealsense scripts:
`./scripts/install_dependencies-4.4.sh`

kazoo-kmt, 2017-01-10:
Thank you. After running `git reset --hard` and  `./scripts/install_dependencies-4.4.sh` then rebooting, I ran `./scripts/patch-uvcvideo-16.04.simple.sh`. Then I tried to run `sudo modprobe uvcvideo` but the terminal stopped to run. 

The message I got when running `./scripts/patch-uvcvideo-16.04.simple.sh` is as follows.
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libusb-1.0-0-dev is already the newest version (2:1.0.20-1).
The following packages were automatically installed and are no longer required:
  libpango1.0-0 libpangox-1.0-0 linux-headers-4.4.0-21
  linux-headers-4.4.0-21-generic linux-image-4.4.0-21-generic
  linux-image-extra-4.4.0-21-generic
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 30 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.1ubuntu2).
linux-headers-generic is already the newest version (4.4.0.57.60).
The following packages were automatically installed and are no longer required:
  libpango1.0-0 libpangox-1.0-0 linux-headers-4.4.0-21
  linux-headers-4.4.0-21-generic linux-image-4.4.0-21-generic
  linux-image-extra-4.4.0-21-generic
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 30 not upgraded.
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
patching file include/uapi/linux/videodev2.h
scripts/kconfig/conf  --silentoldconfig Kconfig
*
* Restart config...
*
*
* General setup
*
Cross-compiler tool prefix (CROSS_COMPILE) [] 
Compile also drivers which will not load (COMPILE_TEST) [N/y/?] n
Local version - append to kernel release (LOCALVERSION) [] 
Automatically append version information to the version string (LOCALVERSION_AUTO) [N/y/?] n
Kernel compression mode
> 1. Gzip (KERNEL_GZIP)
  2. Bzip2 (KERNEL_BZIP2)
  3. LZMA (KERNEL_LZMA)
  4. XZ (KERNEL_XZ)
  5. LZO (KERNEL_LZO)
  6. LZ4 (KERNEL_LZ4)
choice[1-6?]: 1
Default hostname (DEFAULT_HOSTNAME) [(none)] (none)
Arbitrary version signature (VERSION_SIGNATURE) [] (NEW) 
Support for paging of anonymous memory (swap) (SWAP) [Y/n/?] y
System V IPC (SYSVIPC) [Y/n/?] y
POSIX Message Queues (POSIX_MQUEUE) [Y/n/?] y
Enable process_vm_readv/writev syscalls (CROSS_MEMORY_ATTACH) [Y/n/?] y
open by fhandle syscalls (FHANDLE) [Y/n/?] y
uselib syscall (USELIB) [Y/n/?] y
Auditing support (AUDIT) [Y/?] y
Enable system-call auditing support (AUDITSYSCALL) [Y/n/?] y
*
* IRQ subsystem
*
Expose hardware/virtual IRQ mapping via debugfs (IRQ_DOMAIN_DEBUG) [N/y/?] n
Use IRQ threading by default (IRQ_FORCED_THREADING_DEFAULT) [N/y] (NEW) 
Kernel .config support (IKCONFIG) [N/m/y/?] n
Kernel log buffer size (16 => 64KB, 17 => 128KB) (LOG_BUF_SHIFT) [18] 18
CPU kernel log buffer size contribution (13 => 8 KB, 17 => 128KB) (LOG_CPU_MAX_BUF_SHIFT) [12] 12
Memory placement aware NUMA scheduler (NUMA_BALANCING) [Y/n/?] y
  Automatically enable NUMA aware memory/task placement (NUMA_BALANCING_DEFAULT_ENABLED) [Y/n/?] y
Checkpoint/restore support (CHECKPOINT_RESTORE) [Y/n/?] y
Automatic process group scheduling (SCHED_AUTOGROUP) [Y/n/?] y
[... 1064 lines cut ...]
  CHK     include/generated/timeconst.h
  UPD     include/generated/timeconst.h
  CC      arch/x86/kernel/asm-offsets.s
  CHK     include/generated/asm-offsets.h
  UPD     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
make: Entering directory '/home/username/Projects/librealsense/ubuntu-xenial'
  CC [M]  /home/username/Projects/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_driver.o
  CC [M]  /home/username/Projects/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_queue.o
  CC [M]  /home/username/Projects/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_v4l2.o
  CC [M]  /home/username/Projects/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_video.o
  CC [M]  /home/username/Projects/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_ctrl.o
  CC [M]  /home/username/Projects/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_status.o
  CC [M]  /home/username/Projects/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_isight.o
  CC [M]  /home/username/Projects/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_debugfs.o
  CC [M]  /home/username/Projects/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_entity.o
  LD [M]  /home/username/Projects/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/username/Projects/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.mod.o
  LD [M]  /home/username/Projects/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko
make: Leaving directory '/home/username/Projects/librealsense/ubuntu-xenial'
Unloading existing uvcvideo driver...
./scripts/patch-uvcvideo-16.04.simple.sh: line 42:  7949 Killed                  sudo modprobe uvcvideo
```

ev-mp, 2017-01-10:
The patched module has been built correctly but it failed to load back to kernel.
Please run "dmesg -wT", then execute "$sudo modprobe uvcvideo"  in another terminal to get kernel logs.

kazoo-kmt, 2017-01-10:
Thank you so much. I ran `dmesg -wT` after above, and opened another terminal to run `sudo modprobe uvcvideo`. All the logs are as follows:
```
~/Projects/librealsense$ dmesg -wT
[Mon Jan  9 17:10:18 2017] Initializing cgroup subsys cpuset
[Mon Jan  9 17:10:18 2017] Initializing cgroup subsys cpu
[Mon Jan  9 17:10:18 2017] Initializing cgroup subsys cpuacct
[Mon Jan  9 17:10:18 2017] Linux version 4.4.0-040400-generic (kernel@gomeisa) (gcc version 5.2.1 20151010 (Ubuntu 5.2.1-22ubuntu2) ) #201601101930 SMP Mon Jan 11 00:32:41 UTC 2016
[Mon Jan  9 17:10:18 2017] Command line: BOOT_IMAGE=/boot/vmlinuz-4.4.0-040400-generic root=UUID=a2f06444-c812-4f1b-b21b-26fc40d21fd6 ro quiet splash
[Mon Jan  9 17:10:18 2017] KERNEL supported cpus:
[Mon Jan  9 17:10:18 2017]   Intel GenuineIntel
[Mon Jan  9 17:10:18 2017]   AMD AuthenticAMD
[Mon Jan  9 17:10:18 2017]   Centaur CentaurHauls
[Mon Jan  9 17:10:18 2017] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[Mon Jan  9 17:10:18 2017] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[Mon Jan  9 17:10:18 2017] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[Mon Jan  9 17:10:18 2017] x86/fpu: Supporting XSAVE feature 0x04: 'AVX registers'
[Mon Jan  9 17:10:18 2017] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[Mon Jan  9 17:10:18 2017] x86/fpu: Using 'lazy' FPU context switches.
[Mon Jan  9 17:10:18 2017] e820: BIOS-provided physical RAM map:
[Mon Jan  9 17:10:18 2017] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[Mon Jan  9 17:10:18 2017] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[Mon Jan  9 17:10:18 2017] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[Mon Jan  9 17:10:18 2017] BIOS-e820: [mem 0x0000000000100000-0x00000000dffeffff] usable
[Mon Jan  9 17:10:18 2017] BIOS-e820: [mem 0x00000000dfff0000-0x00000000dfffffff] ACPI data
[Mon Jan  9 17:10:18 2017] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[Mon Jan  9 17:10:18 2017] BIOS-e820: [mem 0x0000000100000000-0x000000021fffffff] usable
[Mon Jan  9 17:10:18 2017] NX (Execute Disable) protection: active
[Mon Jan  9 17:10:18 2017] SMBIOS 2.5 present.
[Mon Jan  9 17:10:18 2017] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[Mon Jan  9 17:10:18 2017] Hypervisor detected: KVM
[Mon Jan  9 17:10:18 2017] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[Mon Jan  9 17:10:18 2017] e820: remove [mem 0x000a0000-0x000fffff] usable
[Mon Jan  9 17:10:18 2017] e820: last_pfn = 0x220000 max_arch_pfn = 0x400000000
[Mon Jan  9 17:10:18 2017] MTRR default type: uncachable
[Mon Jan  9 17:10:18 2017] MTRR variable ranges disabled:
[Mon Jan  9 17:10:18 2017] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[Mon Jan  9 17:10:18 2017] MTRR: Disabled
[Mon Jan  9 17:10:18 2017] CPU MTRRs all blank - virtualized system.
[Mon Jan  9 17:10:18 2017] e820: last_pfn = 0xdfff0 max_arch_pfn = 0x400000000
[Mon Jan  9 17:10:18 2017] found SMP MP-table at [mem 0x0009fff0-0x0009ffff] mapped at [ffff88000009fff0]
[Mon Jan  9 17:10:18 2017] Scanning 1 areas for low memory corruption
[Mon Jan  9 17:10:18 2017] Base memory trampoline at [ffff880000099000] 99000 size 24576
[Mon Jan  9 17:10:18 2017] BRK [0x021f6000, 0x021f6fff] PGTABLE
[Mon Jan  9 17:10:18 2017] BRK [0x021f7000, 0x021f7fff] PGTABLE
[Mon Jan  9 17:10:18 2017] BRK [0x021f8000, 0x021f8fff] PGTABLE
[Mon Jan  9 17:10:18 2017] BRK [0x021f9000, 0x021f9fff] PGTABLE
[Mon Jan  9 17:10:18 2017] BRK [0x021fa000, 0x021fafff] PGTABLE
[Mon Jan  9 17:10:18 2017] BRK [0x021fb000, 0x021fbfff] PGTABLE
[Mon Jan  9 17:10:18 2017] RAMDISK: [mem 0x33c62000-0x35e28fff]
[Mon Jan  9 17:10:18 2017] ACPI: Early table checksum verification disabled
[Mon Jan  9 17:10:18 2017] ACPI: RSDP 0x00000000000E0000 000024 (v02 VBOX  )
[Mon Jan  9 17:10:18 2017] ACPI: XSDT 0x00000000DFFF0030 00003C (v01 VBOX   VBOXXSDT 00000001 ASL  00000061)
[Mon Jan  9 17:10:18 2017] ACPI: FACP 0x00000000DFFF00F0 0000F4 (v04 VBOX   VBOXFACP 00000001 ASL  00000061)
[Mon Jan  9 17:10:18 2017] ACPI: DSDT 0x00000000DFFF0480 002118 (v01 VBOX   VBOXBIOS 00000002 INTL 20100528)
[Mon Jan  9 17:10:18 2017] ACPI: FACS 0x00000000DFFF0200 000040
[Mon Jan  9 17:10:18 2017] ACPI: FACS 0x00000000DFFF0200 000040
[Mon Jan  9 17:10:18 2017] ACPI: APIC 0x00000000DFFF0240 00006C (v02 VBOX   VBOXAPIC 00000001 ASL  00000061)
[Mon Jan  9 17:10:18 2017] ACPI: SSDT 0x00000000DFFF02B0 0001CC (v01 VBOX   VBOXCPUT 00000002 INTL 20100528)
[Mon Jan  9 17:10:18 2017] ACPI: Local APIC address 0xfee00000
[Mon Jan  9 17:10:18 2017] No NUMA configuration found
[Mon Jan  9 17:10:18 2017] Faking a node at [mem 0x0000000000000000-0x000000021fffffff]
[Mon Jan  9 17:10:18 2017] NODE_DATA(0) allocated [mem 0x21fff6000-0x21fff9fff]
[Mon Jan  9 17:10:18 2017] kvm-clock: Using msrs 4b564d01 and 4b564d00
[Mon Jan  9 17:10:18 2017] kvm-clock: cpu 0, msr 2:1fff2001, primary cpu clock
[Mon Jan  9 17:10:18 2017] kvm-clock: using sched offset of 1909020543855 cycles
[Mon Jan  9 17:10:18 2017] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[Mon Jan  9 17:10:18 2017] Zone ranges:
[Mon Jan  9 17:10:18 2017]   DMA32    [mem 0x0000000000001000-0x00000000ffffffff]
[Mon Jan  9 17:10:18 2017]   Normal   [mem 0x0000000100000000-0x000000021fffffff]
[Mon Jan  9 17:10:18 2017]   Device   empty
[... 605 lines cut ...]
[... cut ...]
[Mon Jan  9 23:57:52 2017] usb 1-2: Product: BlasterX Senz3D VF0810
[Mon Jan  9 23:57:52 2017] usb 1-2: Manufacturer: Creative Technology Ltd
[Mon Jan  9 23:57:52 2017] usb 1-2: SerialNumber: 160603170234
[Mon Jan  9 23:57:52 2017] input: Creative Technology Ltd BlasterX Senz3D VF0810 as /devices/pci0000:00/0000:00:0c.0/usb1/1-2/1-2:1.3/0003:041E:409B.0003/input/input10
[Mon Jan  9 23:57:52 2017] hid-generic 0003:041E:409B.0003: input,hiddev0,hidraw1: USB HID v1.11 Device [Creative Technology Ltd BlasterX Senz3D VF0810] on usb-0000:00:0c.0-2/input3
```

ev-mp, 2017-01-11:
Please inspect the following segment where modprobe occures:
> [Mon Jan  9 17:13:19 2017] usbcore: registered new interface driver uvcvideo
> [Mon Jan  9 17:13:19 2017] USB Video Class driver (1.1.1)
> [Mon Jan  9 17:22:27 2017] usbcore: deregistering interface [...]

Are you sure you're running with kernel 4.4.0-57  and not 4.4.0.-04.. ? You can check it with "$uname -r".
The patched module should be applied to a PC with the same Kernel version.
In case it is not - then you'll have to switch it to 4.4.0-57. This can be done during PC boot phase -> Ubuntu Boot Options->"Ubuntu Advanced Options"->Select Kernel version.

kazoo-kmt, 2017-01-11:
I check the kernel version again, it shows "4.4.0-040400-generic" now.

ev-mp, 2017-01-11:
The "ubuntu-xenial" folder is just a sandbox for building patched modules, and it doesn't affect the active kernel choice.
You need to restart the PC and select Kernel 4.4.0.-57 from GRUB2 boot menu-> "Advanced Options for Ubuntu".
After that you'll be able to insert the patched kernel module with "modprobe"

kazoo-kmt, 2017-01-11:
Thank you. Now everything seems fine and working. After changing the kernel, I `git reset --hard` again and ran `./scripts/patch-uvcvideo-16.04.simple.sh`, then executed `sudo modprobe uvcvideo`.

closed 2017-03-21 by ddiakopoulos

mambawockeez, 2018-08-16:
Hi i have a really simple question, for your ubuntu 16.04 LTS, are you running it on virtual machine? I used the virtual box but ubuntu even didnt detect my sr300.