realsenseai/librealsense #410: library not found for -lglfw3

kazoo-kmt, 2017-01-11:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version | macOS 10.12.2  |
| Kernel Version (Linux Only)  |   |
| Build System | XCode | 

Continued from #152. Since then, I changed some setting on Xcode and now I'm getting a build error as follows (e.g. when I ran cpp-pointcloud):
```
ld: warning: directory not found for option '-L/Users/username/Projects/librealsense/librealsense.xc/../librealsense/build/Debug'
ld: warning: directory not found for option '-L/Users/username/Projects/librealsense/librealsense.xc/../bin'
ld: warning: directory not found for option '-L/Users/username/Library/Developer/Xcode/DerivedData/bin/Debug'
ld: warning: directory not found for option '-L/Users/username/Projects/librealsense/librealsense.xc/build/Debug'
ld: library not found for -lglfw3
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

As @martinthomas suggested, this error might be because of the deprecated glfw3 from brew, but 
when I ran `brew install homebrew/versions/glfw3`, the output said the followings.
```
Warning: Use glfw instead of deprecated homebrew/versions/glfw3
Warning: glfw-3.2.1 already installed
```
I don't have libglfw3.dylib, but I have libglfw.3.2.dylib, libglfw.3.dylib and libglfw.dylib under /usr/local/lib/.

martinthomas, 2017-01-11:
See pull request #408

martinthomas, 2017-01-11:
This is because glfw3 is deprecated.  If you read the result of the brew install command, it says it installed glfw-3.2.1 - which is not glfw3.  The ld command fails to find "-lglfw3" because brew did not install it.

You can try changing the library target in the Xcode project from glfw3 to glfw and that will fix it or you can use the pull request that I submitted to fix the dependency.

kazoo-kmt, 2017-01-11:
Thank you very much. Now everything works fine.

martinthomas, 2017-01-12:
Close?

kazoo-kmt, 2017-01-12:
Yes, thanks.

closed 2017-01-12 by kazoo-kmt