realsenseai/librealsense #414: Question: Reading Motion data (rs_motion_data)

ohadlights, 2017-01-15:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013, XCode or Makefile | 

It seems that I can successfully read the motion data from the ZR300 motion controller.
But I'm not sure how to interpret it.

This is the documentation:
float               axes[3];    /**< Three [x,y,z] axes; 16 bit data for Gyro [rad/sec], 12 bit for Accelerometer; 2's complement [m/sec^2]*/

So when I extract the first 16 bits - do I build a float out of it? same of the next 12 bits.

jlgarcia75, 2017-10-17:
Hello @ohadlights,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

Liuyangdong, 2017-10-21:
Hi @jlgarcia75,
I still want to know how to interpret "float axes[3]" to get Gyro and Accelerometer data!
Thanks!

Liuyangdong, 2017-10-21:
Hi @jlgarcia75 ,

I still need help with how to interpret "float axes[3]" to get the Gyro and Accelerometer data?

Thanks!

jlgarcia75, 2017-10-23:
@ohadlights,

Each axis [x,y,z] is a float. The bit mapping of each float is: 16 bit data for Gyro [rad/sec], 12 bit for Accelerometer; 2's complement [m/sec^2]*/. This means that you get gyro and accelerometer data for each axis.

Does this help?

Jesus G.
Intel Customer Support

closed 2017-10-26 by jlgarcia75