realsenseai/librealsense #415: UVCIOC_CTRL_QUERY:UVC_GET_CUR error 56, Invalid request code

boulabiar, 2017-01-16:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version |  16.04 64bit | 
| Kernel Version (Linux Only)  |  4.4.0-59-generic |
| Build System |  Makefile | 

I followed the instructions for the install process, then after inserting the new module I get a uvc error

$ ./cpp-alignimages
RealSense error calling rs_create_context(api_version:11201):
UVCIOC_CTRL_QUERY:UVC_GET_CUR error 56, Invalid request code

ev-mp, 2017-01-16:
This is a very general UVC protocol error. Could be HW/FW/USB issue.
Please perform several technical steps to gather data for analysis:
- Did you try to run other samples, such as cpp-enumerate ? What were the results - was it always the same error ? 
- What is the kernel log output when this error occurs ?
- Did try to switching USB ports and connecting via a self-powered USB hub ?
-  Did you check that the camera is operational by running LibRealSense demos on Windows/Another Linux PC ? If it does - share the output of "cpp-enumerate" to obtain the Camera FW info.

boulabiar, 2017-01-16:
1.
Any example I try to open gives me the same error, even cpp-enumerate

2.
kernel log output when this error occurs:
http://pastie.org/10990375

3.
I have 3 usb ports on my laptop, I tried them all.

4.
I haven't checked it on windows, but the first time I purchased the camera, it worked fine, then I haven't tried it until another month.

boulabiar, 2017-01-16:
1.
any sample I try to open gives me the same error even cpp-enumerate
2.
this the kernel log:
http://pastie.org/10990375
3.
I tried the camera on the 3 usb of my laptop, same problem
4. I haven't tried it on another computer/system. 
The first time I got it it worked fine (8 December), then when trying after a month it won't anymore.

ev-mp, 2017-01-16:
Are you using the camera in the same environment (Laptop/Cables) that worked last month ?

Is it possible that you're trying to run demos without activating X server (boot into console only mode) ?

According to the log the patch was not applied correctly and there are still unresolved pixel formats.
Check kernel version - '$uname -r'. to make sure it is 4.4.0-59.
Delete "ubuntu-xenial" folder found in librealsense source, then rerun the the patch script and post  kernel log (~50 lines) when the process completes.

boulabiar, 2017-01-17:
Yes, same configuration,
I have of course X server !

uname -a
Linux ASUS-G550JK 4.4.0-59-generic #80-Ubuntu SMP Fri Jan 6 17:47:47 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

The patch is set and I reopened the system with kernel 4.4.0-51 which is the kernel of the first working run.
I've done all these basic verifications and tests.

The camera may be used meanwhile by someone else on windows, and the firmware may got updated by a new intel SDK. 
How to know the version of the firmware inside the RealSense?

boulabiar, 2017-01-17:
As I finally expected, it was a firmware problem as someone updated it while I'm not present.
I got it working now.

Here are the steps:
I used a windows computer to install the last driver: intel_rs_dcm_r200_2.1.27.2853
I rerun the same driver to uninstall it (the only way to uninstall the driver)
I downloaded an old driver: intel_rs_dcm_r200_2.0.3.53109
I installed it and I've chosen to "update" the firmware (downgrade?)
I took a lot of time just in the step 1of8 (about 10 minutes) then showed me an error that it can't continue.

I removed the RS from the windows pc and connected it to linux and now everything work as in the past.

closed 2017-08-01 by oordan