realsenseai/librealsense #418: VIDIOC_DQBUF error 19, No such device

rubyhsu, 2017-01-19:
| Required Info  |   |
|---|---|
| Camera Model | LR200 | 
| Firmware Version |   | 
| Operating System & Version | Linux 14.04.1  |
| Kernel Version (Linux Only)  |  3.10.40 |
| Build System |  Makefile | 

I use librealsense v1.9.7 and update uvc patch to my linux image.
I can use VLC player to show video 0/1/2 image, but it'll fail to use cpp-capture
error message as below
$ ./cpp-capture 
Capturing DEPTH at 480 x 360, fov = 56.5 x 43.9, distortion = NONE
Capturing COLOR at 640 x 480, fov = 55.3 x 42.9, distortion = MODIFIED_BROWN_CONRADY
Capturing INFRARED at 480 x 360, fov = 56.5 x 43.9, distortion = NONE
Capturing INFRARED2 at 480 x 360, fov = 56.5 x 43.9, distortion = NONE
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
terminate called after throwing an instance of 'std::runtime_error'
  what():  VIDIOC_DQBUF error 19, No such device
Aborted

rubyhsu, 2017-01-19:
I also got an error after execute cpp-motion-module
$./cpp-motion-module
There are 1 connected Realsense devices.
Using device 0, an Intel RealSense LR200
    Serial number : 3311802643
    Firmware version: 2.0.71.18

Iteration 1 started

rs::error was thrown when calling rs_enable_stream_ex(device:0x1d570, stream:FISHEYE, width:640, height:480, format:RAW8, framerate:60):
     unsupported stream

rubyhsu, 2017-01-19:
$ v4l2-ctl --list-formats-ext -d /dev/video0
[ 1718.060930] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
[ 1718.083483] uvcvideo 2-1.1:1.1: resume error -5
[ 1718.110660] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
[ 1718.119380] uvcvideo 2-1.1:1.3: resume error -5
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'YUYV'
        Name        : YUV 4:2:2 (YUYV)
                Size: Discrete 320x962
                        Interval: Discrete 0.017s (60.000 fps)

        Index       : 1
        Type        : Video Capture
        Pixel Format: 'Y12I'
        Name        : Greyscale 12 L/R (Y12I)
                Size: Discrete 640x481
                        Interval: Discrete 0.011s (90.000 fps)
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 640x373
                        Interval: Discrete 0.011s (90.000 fps)
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 640x254
                        Interval: Discrete 0.011s (90.000 fps)
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)

        Index       : 2
        Type        : Video Capture
        Pixel Format: 'Y16 '
        Name        : Greyscale 16-bit (Y16 )
                Size: Discrete 640x481
                        Interval: Discrete 0.011s (90.000 fps)
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 640x373
                        Interval: Discrete 0.011s (90.000 fps)
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 640x254
                        Interval: Discrete 0.011s (90.000 fps)
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)

        Index       : 3
        Type        : Video Capture
        Pixel Format: 'Y8I '
        Name        : Greyscale 8 L/R (Y8I)
                Size: Discrete 640x481
                        Interval: Discrete 0.011s (90.000 fps)
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 640x373
                        Interval: Discrete 0.011s (90.000 fps)
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 640x254
                        Interval: Discrete 0.011s (90.000 fps)
                        Interval: Discrete 0.017s (60.000 fps)
[... 87 lines cut ...]
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 424x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.017s (60.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)

        Index       : 1
        Type        : Video Capture
        Pixel Format: 'RW16'
        Name        : Raw data 16-bit (RW16)
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)

ev-mp, 2017-01-19:
Hi, 
1.Is there any particular reason that you work with  LibRealSense 1.9.7?  It was dated back by  September and since then we have published additional versions, to the latest stable release 1.12.1.

2. VIDIOC_DQBUF error 19 - means that the USB connection was dropped. It could be due USB controller or USB PHY error
In case that the error occurs when you try to stream all sensors simultaneously, but streaming one-by-one works correctly then I'd suggest to switch USB port, and if still reproduced - connect via a self-powered USB hub.

3.  cpp-motion-module -  the demo is intended for cameras with Motion sensors. LR200 lacks those, so there is no reason to run it.

4. We provide support for Kernels v4.4+ only. The dmesg error 

> Failed to set UVC commit control : -32 (exp. 34)

 was observed on older Kernels as well as on some VMs. Therefore I would suggest to consider upgrading the Ubuntu to a stable Kernel version 4.4.

rubyhsu, 2017-01-20:
1.I use  librealsense,master version, can't execure ./cpp-capture on R200 & LR200 , so I decide to use v1.9.7.

2. I use the same device and librealsense(v1.9.7) then execute ./cpp-capture on R200 device, it is work.
Is there any different HW SPEC between LR200 and R200?

3.OK, I ran this demo just for getting device's Firmware version

rubyhsu, 2017-01-20:
Hi ,
I update to v1.12.1,but my device will Kernel panic !!!
dmesg 
```
[   82.841751] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
[   82.848693] uvcvideo 2-1.1:1.1: resume error -5
[   82.855649] uvcvideo: Failed to set UVC commit control : -32 (exp. 34).
[   82.862464] uvcvideo 2-1.1:1.3: resume error -5
[   82.868881] uvcvideo: Failed to query (GET_LEN) UVC control 1 on unit 2: -32 (exp. 2).
[   90.650943] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[   90.659028] pgd = c0004000
[   90.661727] [00000000] *pgd=00000000
[   90.665307] Internal error: Oops: 805 [#1] PREEMPT SMP ARM
[   90.670779] Modules linked in: joydev 8821au(O) rtk_btusb(O) dm_crypt dm_mod inv_ak8975 inv_mpu rfcomm bnep bluetooth rfkill
[   90.682083] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W  O 3.10.40+ #16
[   90.689203] task: c0bae800 ti: c0b70000 task.ti: c0b70000
[   90.694598] PC is at uvc_video_decode_start+0x278/0x710
[   90.699812] LR is at 0x15f5
[   90.702599] pc : [<c055ce88>]    lr : [<000015f5>]    psr: 600f0193
[   90.702599] sp : c0b71ca0  ip : 200f0193  fp : f065200c
[   90.714053] r10: 00000000  r9 : f0652000  r8 : 00000001
[   90.719266] r7 : 00008000  r6 : eb056000  r5 : f0652000  r4 : ec54c800
[   90.725779] r3 : 00000000  r2 : ec54cbcc  r1 : 000005a5  r0 : 00000000
[   90.732293] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
[   90.739673] Control: 10c5387d  Table: aafa006a  DAC: 00000015
[   90.745406]
[   90.745406] PC: 0xc055ce08:
[   90.749664] ce08  e1a0a001 0affff9e e5940008 e28d3048 e18410b2 e58d3038 e5900000 ebfcdc86
[   90.757902] ce28  e6ff0070 e58d003c e28d0048 ebed4e89 e30023ca e19410b2 e30f0fff e1510000
[   90.766138] ce48  0a000125 e2842ff3 e58d1034 e58d2030 e1a00002 eb0a05d2 e59d1034 e51be006
[   90.774374] ce68  e081100a e594a3b8 e7ea1051 e1a0c000 e59403bc e0800100 e1a00100 e08a3000
[   90.782608] ce88  e78ae000 e283e008 e1c310b4 e59d103c e1c311b0 e59d3038 e8930003 e88e0003
[   90.790844] cea8  e594a3c4 e59403bc e58dc034 e1a0100a e2800001 ebf500ac e59433c0 e15a0003
[   90.799080] cec8  82833001 e58413bc e59d2030 e59dc034 858433c0 e1a00002 e1a0100c eb0a065f
[   90.807316] cee8  e5943380 e3530000 1affff6b eaffff67 e5d53000 e1520003 8affff76 e3510000
[   90.815552]
[   90.815552] SP: 0xc0b71c20:
[   90.819810] 1c20  c0cb3700 c01228bc 000003c0 c0b70000 c1ca6ac0 c0020bf0 00000000 c001a520
[   90.828046] 1c40  c055ce88 600f0193 ffffffff c0b71c8c 00000001 c000e858 00000000 000005a5
[   90.836282] 1c60  ec54cbcc 00000000 ec54c800 f0652000 eb056000 00008000 00000001 f0652000
[   90.844517] 1c80  00000000 f065200c 200f0193 c0b71ca0 000015f5 c055ce88 600f0193 ffffffff
[   90.852754] 1ca0  04b789a0 c0b71cc8 00000000 c1df3b50 eb7cf240 c006a0ec 00000000 2aafa087
[   90.860988] 1cc0  c0b71cf0 c0073460 c1df56f0 c007299c ec54cbcc 0000007c c0b71ce8 000005a5
[   90.869224] 1ce0  00000000 a00f0193 0000005a 255408f6 ffffffff ec54c800 eb056000 ec54c800
[   90.877459] 1d00  00008000 eb368e40 f0652000 ec54c888 eb368e40 c055ddb4 0012ad40 c049cc9c
[   90.885697]
[   90.885697] FP: 0xf0651f8c:
[   90.889955] 1f8c  ******** ******** ******** ******** ******** ******** ******** ********
[   90.898198] 1fac  ******** ******** ******** ******** ******** ******** ******** ********
[   90.906438] 1fcc  ******** ******** ******** ******** ******** ******** ******** ********
[   90.914679] 1fec  ******** ******** ******** ******** ******** 15f18d0c 15f50000 05290000
[   90.922918] 200c  06030504 05040504 05040404 05040504 05040504 05040604 05040503 05040504
[   90.931154] 202c  04040504 04040504 05040504 05050405 05060505 05050504 04050605 04050405
[   90.939392] 204c  04050505 05050505 05050304 05050505 04040505 05050405 05050506 06050506
[   90.947628] 206c  04050505 05040605 05050504 05040505 05040405 05040505 05040604 05050405
[   90.955867]
[   90.955867] R2: 0xec54cb4c:
[   90.960125] cb4c  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   90.968362] cb6c  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   90.976597] cb8c  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   90.984833] cbac  00000000 00000800 00000000 00000000 00000000 00000000 00000020 007c0529
[   90.993068] cbcc  00010000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   91.001303] cbec  00000000 00000000 00000000 00000000 00000000 00011124 01d50274 10da8580
[   91.009539] cc0c  382dbd00 00028b0b 00000004 01b20700 028b0b00 05161500 0a2c2b00 11242a00
[   91.017774] cc2c  02740002 6f000169 bd0010da 8b0b382d 00040002 07000000 0b0001b2 1500028b
[   91.026011]
[   91.026011] R4: 0xec54c780:
[   91.030269] c780  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   91.038505] c7a0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   91.046741] c7c0  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[... 66 lines cut ...]
[   91.557474] [<c055cbac>] (uvc_video_complete+0x70/0xd4) from [<c049ce48>] (usb_hcd_giveback_urb+0x54/0xc4)
[   91.567115] [<c049ce48>] (usb_hcd_giveback_urb+0x54/0xc4) from [<c04caf98>] (handle_tx_event+0x20c/0xea8)
[   91.576666] [<c04caf98>] (handle_tx_event+0x20c/0xea8) from [<c04cbe50>] (xhci_irq+0x21c/0x694)
[   91.585350] [<c04cbe50>] (xhci_irq+0x21c/0x694) from [<c04ba768>] (tegra_xhci_irq+0xac/0xd4)
[... cut ...]
[   91.719256] [<c058abf8>] (cpuidle_idle_call+0xf0/0x294) from [<c000faf8>] (arch_cpu_idle+0x10/0x40)
[   91.728288] [<c000faf8>] (arch_cpu_idle+0x10/0x40) from [<c00af2e0>] (cpu_startup_entry+0x130/0x258)
[   91.737408] [<c00af2e0>] (cpu_startup_entry+0x130/0x258) from [<c0aefa84>] (start_kernel+0x30c/0x318)
[   91.746610] Code: e59403bc e0800100 e1a00100 e08a3000 (e78ae000)
[   91.752691] ---[ end trace b1563f70c4d516ab ]---
[   91.770283] Kernel panic - not syncing: Fatal exception in interrupt

ev-mp, 2017-01-21:
Hi, in order to advance you need to get a recommended stable LTS kernel version 4.4+. Support of earlier kernels is discontinued.
You can follow a link suggested in another [thread](https://github.com/IntelRealSense/librealsense/issues/413), or use other resources to obtain the appropriate kernel version .
After that please reapply kernel patch and rerun the demos.

Note: after installing a new kernel make it active with "sudo update-grub" or selecting it manually from grub boot menu.

RoboEvangelist, 2017-05-09:
I get this error when I try to run multiple cameras in parallel with callbacks.

When I run the cameras in a while loop (polling frames device after device), then this does not happen. But in this way, if I want to, for example, save every new frame to a file .jpg file, then the FPS for all cameras drops to 20 FPS (from 30 FPS originally). The hardware/software is very delicate.

papr, 2017-07-18:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 1.0.72.06 | 
| Operating System & Version | Ubuntu 16.04 LTS |
| Kernel version |  4.8.0-58-generic |
| librealsense version | 1.12.1 |

I use the C librealsense interface and get the aboce C++ exception (`VIDIOC_DQBUF error 19, No such device`) when the USB connection fails, e.g. device disconnects, bad USB extension cable.

I do not know how the interaction between C calls and C++ code works, but my intuition tells me that this C++ exception should be caught by the C level-library and converted to a `rs_error` pointer.

papr, 2017-07-18:
After looking through the code, it looks like the exception should be actually caught by `HANDLE_EXCEPTIONS_AND_RETURN`. Could it be that the exception is thrown in a different thread?

Looks like the exception is thrown here: [uvc-v4l2.cpp#L341](https://github.com/IntelRealSense/librealsense/blob/master/src/uvc-v4l2.cpp#L341).

papr, 2017-07-19:
There is no such crash on macOS. On macOS the following is printed:
```bash
$ ./cpp-pointcloud 
RealSense error calling rs_wait_for_frames(device:0x7fbab3803420):
    Timeout waiting for frames.
```

ev-mp, 2017-08-22:
Duplicate of [558](https://github.com/IntelRealSense/librealsense/issues/558).
Additional info can be found in  [515](https://github.com/IntelRealSense/librealsense/issues/515), [437](https://github.com/IntelRealSense/librealsense/issues/437), [418](https://github.com/IntelRealSense/librealsense/issues/418) and 
[108](https://github.com/IntelRealSense/librealsense/issues/108).

closed 2017-08-22 by ev-mp

atmadeep, 2018-12-26:
Ok, so I'm dealing with this error from time immemorial (3 months). I'm getting this error on the raspberry pi with raspbian stretch. As some mentioned, this is the trouble when you run all the sensors (in my case ALL OF MY 2 cameras), The pi is not able to deliver the power required. So it dumps them. Will using a y-cable for power solve this issue?