realsenseai/librealsense #420: I want to ask which release linux kernel version will include the kernel patch?

BlazingForests, 2017-01-20:
I want to ask which release linux kernel version will include the kernel patch?

dorodnic, 2017-01-23:
Patches required for R200 and ZR300 devices are already included in the 4.4 Kernel. 
At the moment, we are working on preparing F200 and SR300 patches for upstream (4.10 at the moment)
Future RealSense devices are likely to require more modifications to the Kernel.

bedenko, 2017-01-28:
Will support for F200 in the future also be ported to 4.4 kernel, for LTE distros?

2Reeko, 2017-02-01:
@dorodnic Does that mean it is unnecessary to run the `patch-uvcvideo-16.04.simple.sh` script during install if I'm only going to use the R200 camera?

dorodnic, 2017-02-05:
@Reeko, that is correct. We provide single script per distribution that enables all supported devices. R200 will work out of the box on Ubuntu 16.04 LTS (in terms of kernel patching. you do still have to install udev rules)

fnoop, 2017-03-04:
@dorodnic Aha!  This would be a very important note to add to the install documentation :)

closed 2017-08-20 by oordan