realsenseai/librealsense #421: Point Cloud example image freezes for couple of seconds at random time instances

christow, 2017-01-20:
| Camera Model | R200 |
| Firmware Version | 1.0.72.10  | 
| Operating System & Version | Ubuntu 16.04 LTS  |
| Kernel Version (Linux Only)  | 4.4.0-59-generic #80-Ubuntu SMP Fri Jan 6 17:47:47 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux  |
| Build System |  Makefile |

ddiakopoulos, 2017-03-16:
@christow does this happen in other examples as well?

christow, 2017-03-17:
As far as I remember it was freezing for all the c++ tutorial demos. In the further, investigation it turned out that it is freezing when I'm feeding too many image sources.

jlgarcia75, 2017-10-07:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

closed 2017-10-16 by jlgarcia75