realsenseai/librealsense #426: What are the differences between the two r200 Models

bloyl, 2017-01-27:
Can anyone tell me the difference between the two r200 developer kits

- Model # :946432 - $99 USD
- Model # :VF0830 - Price: $169.00  

My use case is to use multiple cameras for object scanning in relatively close range 1-2 meters. Assuming both will work with librealsense, what differentiates these 2 models?

Thanks

poqudrof, 2017-01-27:
I had the same question. The first obvious reason was 70$, however there might be something more. I doubt I recently bought the 99$USD one.

bloyl, 2017-01-30:
I found this https://software.intel.com/en-us/forums/realsense/topic/698512
which suggests that the VF0830 has abetter RGB camera, but I suspect that $70 buys you more then just a better color camera?

Does anyone have any specs or experience using these cameras?

ddiakopoulos, 2017-03-16:
Intel is exhausting the supply of the older R200 cameras (#946432) at the existing old price. The improved LR200 (#VF0830) has newer sensors, but the same functionality and form factor. There was a decision to increase the price of new devkits, and this is reflected in the price of the LR200.

closed 2017-03-16 by ddiakopoulos

nouei, 2017-04-13:
I saved the point clouds with two models Model # :946432 and Model # :VF0830. I compare them in Meshlab, I see Model # VF0830 has about 2k points more and in my opinion it is better.

ddiakopoulos, 2017-04-13:
@nouei yes, as I noted above, the LR200 has slightly newer sensors (with marginally improved performance). Although, I wouldn't say 2k additional points is a particularly scientific metric.

nouei, 2017-06-07:
Now intel buys Model #: 951798 for $169