realsenseai/librealsense #428: exception thrown on global rs::context construction using statically linked libs

melax, 2017-01-31:
An error occurs in a simple 10 line program when using statically linked librealsense, on windows, and seems to happen when rs context is constructed in a global outside of main.

Using statically linked versions of the realsense libraries, an error occurs in the following short program:

```c++
#define NOMINMAX
#include "../../include/librealsense/rs.hpp"
#include "../../include/librealsense/rsutil.h"

#ifdef _WIN64
#ifdef _DEBUG
#pragma comment(lib, "../../librealsense.vc14/realsense-s/obj/Debug-x64/realsense-sd.lib") 
#else  // release x64
#pragma comment(lib, "..../../librealsense.vc14/realsense-s/obj/Release-x64/realsense-s.lib")   
#endif // _DEBUG vs release for x64
#else  // we have _WIN32
#ifdef _DEBUG
#pragma comment(lib, "../../librealsense.vc14/realsense-s/obj/Debug-Win32/realsense-sd.lib")  
#else  // release 32 
#pragma comment(lib, "../../librealsense.vc14/realsense-s/obj/Release-Win32/realsense-s.lib")   // pragma easier than using settings within project properties
#endif // _DEBUG vs release for Win32 
#endif // _WIN64 vs _WIN32

rs::context ctx;  // in global scope this causes 'access violation' and throws exception on program startup
int main(int, const char**)
{
	return 0;
}
```
But putting the context on the stack, constructing after main is invoked, then the bug does not happen:
```c++
int main(int, const char**)
{
	rs::context ctx;  // fine when on the stack here 
	return 0;
}
```
So this indicates something isn't handled properly during the construction of the rs::context within the global scope of a program.   

The cause of the error appears to happen during a mutex/lock operation:
```c++
    Exception thrown at 0x536039E3 (msvcp140d.dll) in unit_test_g_ctx_construction.exe: 0xC0000005: Access violation reading location 0x00000000.
```

The call stack at runtime:
```c++
          testrs.exe!rs_context::acquire_instance() Line 38	C++
          testrs.exe!rs_create_context(int api_version, rs_error * * error) Line 48	C++
          testrs.exe!rs::context::context() Line 173	C++
          testrs.exe!`dynamic initializer for 'ctx''() Line 21	C++
```          

The project file used for this test was a newly created project file from vs2015 with all default settings.  hence the include pathing and library pragmas at the beginning of the source code shown above.  
I can push this unit test into a github branch if you are not able to reproduce.

| Required Info  |   |
|---|---|
| Camera Model | all cameras R200, F200 and SR300 | 
| Firmware Version | all  | 
| Operating System & Version |  PC  |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2015 |

dorodnic, 2017-02-05:
Hi, what you are reporting makes sense, since the rs::context is managed like a singleton but not really. Originally, librealsense included a limitation allowing only a single instance of context at any given time. Later down the line, my team has somewhat relaxed this restriction based on feedback from ROS community, allowing context to be ref-counted. We intentionally avoided something like Mayers singleton, in order to provide deterministic destruction (once last instance of context is destroyed, we would like all RealSense resources to be deallocated). This might have caused issues when linking statically. This is a purely software issue and hence can be investigated and fixed.

jlgarcia75, 2017-10-06:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

melax, 2017-10-06:
You can close this issue.
Thanks

closed 2017-10-06 by jlgarcia75

melax, 2017-10-06:
We're ok now.  Thanks